code-iai/iai_kinect2 #158: Make Target Errors

rahulraw, 2015-09-30:
When using catkin_make to build the iai_kinect2 package I am receiving the following error messages:
```
[ 42%] Built target kinect2_registration
Scanning dependencies of target kinect2_calibration
Scanning dependencies of target kinect2_bridge
Scanning dependencies of target kinect2_bridge_nodelet
Scanning dependencies of target kinect2_viewer
make[2]: *** No rule to make target `/usr/lib/x86_64-linux-gnu/libopencv_ts.so.2.4.8', needed by `/home/administrator/catkin_ws/devel/lib/kinect2_calibration/kinect2_calibration'.  Stop.
make[2]: *** Waiting for unfinished jobs....
[ 57%] Building CXX object iai_kinect2/kinect2_calibration/CMakeFiles/kinect2_calibration.dir/src/kinect2_calibration.cpp.o
make[2]: *** make[2]: *** No rule to make target `/usr/lib/x86_64-linux-gnu/libopencv_ts.so.2.4.8', needed by `/home/administrator/catkin_ws/devel/lib/libkinect2_bridge_nodelet.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....
[ 71%] [ 85%] No rule to make target `/usr/lib/x86_64-linux-gnu/libopencv_ts.so.2.4.8', needed by `/home/administrator/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge'.  Stop.
make[2]: *** Waiting for unfinished jobs....
Building CXX object iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o
Building CXX object iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/src/kinect2_bridge.cpp.o
[100%] make[2]: *** No rule to make target `/usr/lib/x86_64-linux-gnu/libopencv_ts.so.2.4.8', needed by `/home/administrator/catkin_ws/devel/lib/kinect2_viewer/kinect2_viewer'.  Stop.
make[2]: *** Waiting for unfinished jobs....
Building CXX object iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/src/viewer.cpp.o
In file included from /usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:52:0,
                 from /usr/include/pcl-1.7/pcl/visualization/cloud_viewer.h:39,
                 from /home/administrator/catkin_ws/src/iai_kinect2/kinect2_viewer/src/viewer.cpp:32:
/usr/include/pcl-1.7/pcl/visualization/area_picking_event.h: In member function ‘bool pcl::visualization::AreaPickingEvent::getPointsIndices(std::vector<int>&) const’:
/usr/include/pcl-1.7/pcl/visualization/area_picking_event.h:64:30: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
           if (nb_points_ == -1)
                              ^
make[1]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/all] Error 2
make[1]: *** [iai_kinect2/kinect2_calibration/CMakeFiles/kinect2_calibration.dir/all] Error 2
make[1]: *** [iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```

It seems as if it is looking for opencv 2.4.8 ? However, according to the instruction any 2.4.x system should be okay.

wiedemeyer, 2015-10-01:
It should work with an 2.4.x version, but I only test with the default one from the official repositories. It seems like your installation/configuration is wrong. iai_kinect2 will use the OpenCV that is found by cmake. it looks like cmake found some configuration which are wrong. So please check your OpenCV installation.

rahulraw, 2015-10-06:
How can I safely remove my opencv and download it again?

wiedemeyer, 2015-10-07:
I don't know. Maybe you should look in the OpenCV manual.

For installing it you could probably do `sudo apt-get install libopencv-dev`

closed 2015-10-26 by wiedemeyer