code-iai/iai_kinect2 #160: SD cloud is shifted

JimmyDaSilva, 2015-10-01:
As I was calibrating the extrinsics for the Kinect2 I realized that there is a shift between the hd (or qhd) cloud and the sd cloud.
I would estimate it to be a shift of 4~5cm to the right so it's probably just because the sd pointcloud is not published in the right frame (it's about the shift between camera_rgb_optical_frame and camera_ir_optical_frame)
Or it could also be due to a problem with the calib_pose.yaml as mentionned in https://github.com/code-iai/iai_kinect2/issues/117

See for yourselves:

HD
[image]
QHD
[image]
SD
[image]

wiedemeyer, 2015-10-01:
Hi,
the sd cloud is published in the right frame. The difference between (q)hd and sd is the registration method. The (q)hd topics are using a depth to color frame registration, so the depth image is transformed to the color frame using the calibrated extrinsic parameters. For the sd topic a color to depth registration is used, so for each depth pixel the corresponding color pixel is looked up. The sd registration is using parameters from the sensor and not the calibrated ones. It is the registration method that is also part of the official Microsoft SDK.

JimmyDaSilva, 2015-10-01:
Thanks @wiedemeyer !
Is there any way to use the calibration file instead?
I would like to use the sd cloud to speed my post processing but if the points are shifted it's going to be a problem.

wiedemeyer, 2015-10-01:
For that case there is the qhd topic which has just a quater of the HD size, but also allows easy mapping to the HD topics.
But if you want to transformiert the SD depth image using the calibration you might want to take a look at depth_image_proc.

JimmyDaSilva, 2015-10-01:
I managed to create the sd points to be on the right spot using depth_image_proc.
<b>Be careful, the color is still shifted on the cloud, but the points are well placed.</b>

Original QHD points
[image]
New SD points
[image]

I am just using the qhd depth image and projecting the image to be sd. Then I create the pointcloud for the depth image.

Posting my launch file in case this could be helpful to someone.
```
<launch>

  <arg name="kinect_name" default="kinectV2" />  

  <!-- Nodelet manager for this pipeline -->
  <node pkg="nodelet" type="nodelet" args="manager"
        name="$(arg kinect_name)_record_player_manager" output="screen"/>

  <!-- Get the scene depth image transformed to the rgb camera frame -->
  <node pkg="nodelet" type="nodelet" name="$(arg kinect_name)_depth_registration_nodelet"
  args="load depth_image_proc/register $(arg kinect_name)_record_player_manager --no-bond">

      <!-- Input: RGB camera calibration and metadata. (sensor_msgs/CameraInfo) -->
      <remap from="rgb/camera_info" to="$(arg kinect_name)/sd/camera_info"/> 
      <!-- Input: Depth camera calibration and metadata. (sensor_msgs/CameraInfo) -->
      <remap from="depth/camera_info" to="$(arg kinect_name)/qhd/camera_info"/> 
      <!-- Input: Rectified depth image. (sensor_msgs/Image) -->
      <remap from="depth/image_rect" to="$(arg kinect_name)/qhd/image_depth_rect"/>

      <!-- Output: Camera calibration and metadata. Same as rgb/camera_info but time-synced 
      to depth_registered/image_rect. (sensor_msgs/CameraInfo)   -->
      <remap from="depth_registered/camera_info" to="$(arg kinect_name)/sd/depth_registered/camera_info"/>
      <!-- Output: Reprojected depth image in the RGB camera frame. -->
      <remap from="depth_registered/image_rect" to="$(arg kinect_name)/sd/depth_registered/image_rect"/> 
  </node>     

  <!-- Convert it into a point cloud -->
  <node pkg="nodelet" type="nodelet" name="$(arg kinect_name)_cloudify_nodelet"
        args="load depth_image_proc/point_cloud_xyzrgb $(arg kinect_name)_record_player_manager --no-bond">
    <remap from="depth_registered/image_rect" to="$(arg kinect_name)/sd/depth_registered/image_rect"/>
    <remap from="depth_registered/points" to="$(arg kinect_name)/sd/depth_registered/points"/>
    <remap from="rgb/image_rect_color" to="$(arg kinect_name)/sd/image_color_rect"/>
    <remap from="rgb/camera_info" to="$(arg kinect_name)/sd/camera_info"/>
  </node>
  
</launch>
```

Still this is kind of a hack and there should be a proper way to do this.

wiedemeyer, 2015-10-05:
I would need to write some code to get the small depth image together with a small color image in the color frame. Right now I don't have the time for it, but I will look in it later.

JimmyDaSilva, 2015-10-05:
Thanks @wiedemeyer! Good job on this amazing package ;)

closed 2015-10-14 by wiedemeyer