code-iai/iai_kinect2 #161: kinect2_viewer catkin_make error: VTK?

NukeVol22, 2015-10-01:
I am having a linker error that I can't seem to shake while building the kinect2_viewer. I added extra language to the CMake file to link to VTK 5.8, but there was no change. The other three executables built just fine. Using Ubuntu 14.04 and Indigo. libfreenect2 also installed okay.

```terminal
Linking CXX executable /home/matt/catkin_ws/devel/lib/kinect2_viewer/kinect2_viewer
/usr/bin/ld: cannot find -lvtkRenderingFreeTypeFontConfig
/usr/bin/ld: cannot find -lvtkRenderingFreeType
/usr/bin/ld: cannot find -lvtkRenderingCore
/usr/bin/ld: cannot find -lvtkCommonExecutionModel
/usr/bin/ld: cannot find -lvtkCommonDataModel
/usr/bin/ld: cannot find -lvtkCommonMath
/usr/bin/ld: cannot find -lvtkCommonCore
/usr/bin/ld: cannot find -lvtkCommonMisc
/usr/bin/ld: cannot find -lvtkCommonSystem
/usr/bin/ld: cannot find -lvtkCommonTransforms
/usr/bin/ld: cannot find -lvtkFiltersExtraction
/usr/bin/ld: cannot find -lvtkFiltersCore
/usr/bin/ld: cannot find -lvtkFiltersGeneral
/usr/bin/ld: cannot find -lvtkCommonComputationalGeometry
/usr/bin/ld: cannot find -lvtkFiltersStatistics
/usr/bin/ld: cannot find -lvtkImagingFourier
/usr/bin/ld: cannot find -lvtkImagingCore
/usr/bin/ld: cannot find -lvtkFiltersGeometry
/usr/bin/ld: cannot find -lvtkFiltersSources
/usr/bin/ld: cannot find -lvtkIOImage
/usr/bin/ld: cannot find -lvtkIOCore
/usr/bin/ld: cannot find -lvtkIOXMLParser
/usr/bin/ld: cannot find -lvtkIOXML
/usr/bin/ld: cannot find -lvtkIOGeometry
/usr/bin/ld: cannot find -lvtkRenderingParallel
/usr/bin/ld: cannot find -lvtkFiltersParallel
/usr/bin/ld: cannot find -lvtkFiltersModeling
/usr/bin/ld: cannot find -lvtkParallelCore
/usr/bin/ld: cannot find -lvtkIOLegacy
/usr/bin/ld: cannot find -lvtkRenderingOpenGL
/usr/bin/ld: cannot find -lvtkImagingHybrid
/usr/bin/ld: cannot find -lvtkRenderingAnnotation
/usr/bin/ld: cannot find -lvtkImagingColor
/usr/bin/ld: cannot find -lvtkIOLSDyna
/usr/bin/ld: cannot find -lvtkIOAMR
/usr/bin/ld: cannot find -lvtkFiltersAMR
/usr/bin/ld: cannot find -lvtkIOMINC
/usr/bin/ld: cannot find -lvtkFiltersHybrid
/usr/bin/ld: cannot find -lvtkImagingSources
/usr/bin/ld: cannot find -lvtkNetCDF
/usr/bin/ld: cannot find -lvtkNetCDF_cxx
/usr/bin/ld: cannot find -lvtkInfovisCore
/usr/bin/ld: cannot find -lvtkIOImport
/usr/bin/ld: cannot find -lvtkFiltersParallelStatistics
/usr/bin/ld: cannot find -lvtkIOODBC
/usr/bin/ld: cannot find -lvtkIOSQL
/usr/bin/ld: cannot find -lvtksqlite
/usr/bin/ld: cannot find -lvtkPythonInterpreter
/usr/bin/ld: cannot find -lvtkFiltersFlowPaths
/usr/bin/ld: cannot find -lvtkInfovisBoostGraphAlgorithms
/usr/bin/ld: cannot find -lvtkFiltersProgrammable
/usr/bin/ld: cannot find -lvtkRenderingVolumeOpenGL
/usr/bin/ld: cannot find -lvtkRenderingVolume
/usr/bin/ld: cannot find -lvtkIOPLY
/usr/bin/ld: cannot find -lvtkInteractionStyle
/usr/bin/ld: cannot find -lvtkViewsCore
/usr/bin/ld: cannot find -lvtkInteractionWidgets
/usr/bin/ld: cannot find -lvtkImagingGeneral
/usr/bin/ld: cannot find -lvtkRenderingGL2PS
/usr/bin/ld: cannot find -lvtkRenderingContext2D
/usr/bin/ld: cannot find -lvtkIOInfovis
/usr/bin/ld: cannot find -lvtkFiltersReebGraph
/usr/bin/ld: cannot find -lvtkCommonColor
/usr/bin/ld: cannot find -lvtkDomainsChemistry
/usr/bin/ld: cannot find -lvtkIOExport
/usr/bin/ld: cannot find -lvtkIOFFMPEG
/usr/bin/ld: cannot find -lvtkIOMovie
/usr/bin/ld: cannot find -lvtkIOMPIImage
/usr/bin/ld: cannot find -lvtkParallelMPI
/usr/bin/ld: cannot find -lvtkViewsGeovis
/usr/bin/ld: cannot find -lvtkGeovisCore
/usr/bin/ld: cannot find -lvtkInfovisLayout
/usr/bin/ld: cannot find -lvtkViewsInfovis
/usr/bin/ld: cannot find -lvtkFiltersImaging
/usr/bin/ld: cannot find -lvtkRenderingLabel
/usr/bin/ld: cannot find -lvtkTestingIOSQL
/usr/bin/ld: cannot find -lvtkFiltersParallelGeometry
/usr/bin/ld: cannot find -lvtkFiltersSelection
/usr/bin/ld: cannot find -lvtkGUISupportQt
/usr/bin/ld: cannot find -lvtkLocalExample
/usr/bin/ld: cannot find -lvtkInteractionImage
/usr/bin/ld: cannot find -lvtkWrappingTools
/usr/bin/ld: cannot find -lvtkRenderingLOD
/usr/bin/ld: cannot find -lvtkImagingStencil
/usr/bin/ld: cannot find -lvtkIOGDAL
/usr/bin/ld: cannot find -lvtkRenderingQt
/usr/bin/ld: cannot find -lvtkFiltersTexture
/usr/bin/ld: cannot find -lvtkIOMySQL
/usr/bin/ld: cannot find -lvtkGUISupportQtWebkit
/usr/bin/ld: cannot find -lvtkViewsQt
/usr/bin/ld: cannot find -lvtkImagingMath
/usr/bin/ld: cannot find -lvtkChartsCore
/usr/bin/ld: cannot find -lvtkIOParallelNetCDF
/usr/bin/ld: cannot find -lvtkFiltersCosmo
/usr/bin/ld: cannot find -lvtkWrappingJava
/usr/bin/ld: cannot find -lvtkFiltersVerdict
/usr/bin/ld: cannot find -lverdict
/usr/bin/ld: cannot find -lvtkIOXdmf2
/usr/bin/ld: cannot find -lvtkxdmf2
/usr/bin/ld: cannot find -lvtkViewsContext2D
/usr/bin/ld: cannot find -lvtkIOParallel
/usr/bin/ld: cannot find -lvtkIONetCDF
/usr/bin/ld: cannot find -lvtkIOParallelLSDyna
/usr/bin/ld: cannot find -lvtkIOVideo
/usr/bin/ld: cannot find -lvtkIOEnSight
/usr/bin/ld: cannot find -lvtkRenderingFreeTypeOpenGL
/usr/bin/ld: cannot find -lvtkjsoncpp
/usr/bin/ld: cannot find -lvtkFiltersParallelFlowPaths
/usr/bin/ld: cannot find -lvtkRenderingHybridOpenGL
/usr/bin/ld: cannot find -lvtkWrappingPythonCore
/usr/bin/ld: cannot find -lvtkRenderingMatplotlib
/usr/bin/ld: cannot find -lvtkIOParallelExodus
/usr/bin/ld: cannot find -lvtkIOExodus
/usr/bin/ld: cannot find -lvtkImagingMorphological
/usr/bin/ld: cannot find -lvtkGUISupportQtSQL
/usr/bin/ld: cannot find -lvtkImagingStatistics
/usr/bin/ld: cannot find -lvtkTestingGenericBridge
/usr/bin/ld: cannot find -lvtkTestingRendering
/usr/bin/ld: cannot find -lvtkIOPostgreSQL
/usr/bin/ld: cannot find -lvtkFiltersGeneric
/usr/bin/ld: cannot find -lvtkFiltersHyperTree
/usr/bin/ld: cannot find -lvtkFiltersParallelMPI
/usr/bin/ld: cannot find -lvtkIOMPIParallel
/usr/bin/ld: cannot find -lvtkRenderingVolumeAMR
/usr/bin/ld: cannot find -lvtkRenderingImage
/usr/bin/ld: cannot find -lvtkFiltersParallelImaging
/usr/bin/ld: cannot find -lvtkGUISupportQtOpenGL
/usr/bin/ld: cannot find -lvtkIOVPIC
collect2: error: ld returned 1 exit status
make[2]: *** [/home/matt/catkin_ws/devel/lib/kinect2_viewer/kinect2_viewer] Error 1
make[1]: *** [iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```
Thank you for the assist!

wiedemeyer, 2015-10-02:
kinect2_viewer does not depend on VTK, PCL is using VTK. Do you use packages from the official repositories or do you have some custom installations?

NukeVol22, 2015-10-07:
The short answer is both; I had an official repository and a build from trunk, but seeing where your question was going, I removed the trunk build and wrestled with the VTK and ROS packages installed on my computer until I had a successful compile.

Now I'm experiencing a different error (segmentation fault) with respect to the viewer, but I'll close this issue and start a new one after taking a few cracks at it.  Thanks.

closed 2015-10-07 by NukeVol22