code-iai/iai_kinect2 #162: calibrate Errors in steps "4,6,8,9"

kerolos, 2015-10-04:
I have some problem in calibration.

4- Calibrate the intrinsics: rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 calibrate color :

kerolos@kerolos:~/catkin_ws$ rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 calibrate color

Start settings:
       Mode: calibrate
     Source: color
      Board: chess
 Dimensions: 5 x 7
 Field size: 0.03
Dist. model: 5 coefficients
Topic color: /kinect2/hd/image_mono
   Topic ir: /kinect2/sd/image_ir
Topic depth: /kinect2/sd/image_depth
       Path: ./

restoring files...
starting calibration...
calibrating intrinsics...
OpenCV Error: Assertion failed (nimages > 0) in calibrateCamera, file /home/kerolos/build/opencv-2.4.9/modules/calib3d/src/calibration.cpp, line 3415
terminate called after throwing an instance of 'cv::Exception'
  what():  /home/kerolos/build/opencv-2.4.9/modules/calib3d/src/calibration.cpp:3415: error: (-215) nimages > 0 in function calibrateCamera

Aborted (core dumped)

6- Calibrate the intrinsics of the ir camera: rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 calibrate ir :

kerolos@kerolos:~/catkin_ws$ rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 calibrate ir

Start settings:
       Mode: calibrate
     Source: ir
      Board: chess
 Dimensions: 5 x 7
 Field size: 0.03
Dist. model: 5 coefficients
Topic color: /kinect2/hd/image_mono
   Topic ir: /kinect2/sd/image_ir
Topic depth: /kinect2/sd/image_depth
       Path: ./

restoring files...
starting calibration...
calibrating intrinsics...
OpenCV Error: Assertion failed (nimages > 0) in calibrateCamera, file /home/kerolos/build/opencv-2.4.9/modules/calib3d/src/calibration.cpp, line 3415
terminate called after throwing an instance of 'cv::Exception'
  what():  /home/kerolos/build/opencv-2.4.9/modules/calib3d/src/calibration.cpp:3415: error: (-215) nimages > 0 in function calibrateCamera

Aborted (core dumped)

8-Calibrate the extrinsics: rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 calibrate sync:

kerolos@kerolos:~/catkin_ws$ rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 calibrate sync

Start settings:
       Mode: calibrate
     Source: sync
      Board: chess
 Dimensions: 5 x 7
 Field size: 0.03
Dist. model: 5 coefficients
Topic color: /kinect2/hd/image_mono
   Topic ir: /kinect2/sd/image_ir
Topic depth: /kinect2/sd/image_depth
       Path: ./

restoring files...
starting calibration...
Camera Matrix Color:
[]
Distortion Coeeficients Color:
[]

Camera Matrix Ir:
[]
Distortion Coeeficients Ir:
[]

calibrating Color and Ir extrinsics...
OpenCV Error: Assertion failed (nimages > 0 && nimages == (int)imagePoints1.total() && (!imgPtMat2 || nimages == (int)imagePoints2.total())) in collectCalibrationData, file /home/kerolos/build/opencv-2.4.9/modules/calib3d/src/calibration.cpp, line 3164
terminate called after throwing an instance of 'cv::Exception'
  what():  /home/kerolos/build/opencv-2.4.9/modules/calib3d/src/calibration.cpp:3164: error: (-215) nimages > 0 && nimages == (int)imagePoints1.total() && (!imgPtMat2 || nimages == (int)imagePoints2.total()) in function collectCalibrationData

Aborted (core dumped)

9-Calibrate the depth measurements: rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 calibrate depth :

kerolos@kerolos:~/catkin_ws$ rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 calibrate depth

Start settings:
       Mode: calibrate
     Source: depth
      Board: chess
 Dimensions: 5 x 7
 Field size: 0.03
Dist. model: 5 coefficients
Topic color: /kinect2/hd/image_mono
   Topic ir: /kinect2/sd/image_ir
Topic depth: /kinect2/sd/image_depth
       Path: ./

restoring files...
Segmentation fault (core dumped)

wiedemeyer, 2015-10-05:
Did you recorded images before? See here: https://github.com/code-iai/iai_kinect2/blob/master/kinect2_calibration/README.md for detailed information.

kerolos, 2015-10-05:
yes, i recorded images for color, IR and both in steps(3,5 and 6). i followed all those procedures in the link that you mentioned in your comment.
unfortunately i faced problems when i run those steps "4,6,8 and 9" that declare in my previous comment. Do you know how can i fix those errors ?
thanks in advance

wiedemeyer, 2015-10-06:
After `restoring files...` it should print the names of the files that are read, but since there is no output it means that zero images are loaded.
You have to run the calibration in the same folder where the recorded images are stored (or you could provide the path to the images as a parameter).

kerolos, 2015-10-06:
Everything went well when i calibrated the kinect2 except  "Calibrate the depth measurements", i got this error:
kerolos@kerolos:~/kinect_cal_data$ rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 calibrate depth
Start settings:
       Mode: calibrate
     Source: depth
      Board: chess
 Dimensions: 5 x 7
 Field size: 0.03
Dist. model: 5 coefficients
Topic color: /kinect2/hd/image_mono
   Topic ir: /kinect2/sd/image_ir
Topic depth: /kinect2/sd/image_depth
       Path: ./

restoring files...
starting calibration...
frame: ./0000_depth.png
frame: ./0000_sync_depth.png
frame: ./0001_depth.png
frame: ./0001_sync_depth.png
frame: ./0002_depth.png
frame: ./0002_sync_depth.png
Segmentation fault (core dumped)

wiedemeyer, 2015-10-07:
Compile iai_kinect2 in debug and start the calibration with gdb and paste a stack trace here.

kerolos, 2015-10-20:
Start settings:
       Mode: calibrate
     Source: depth
      Board: chess
 Dimensions: 5 x 7
 Field size: 0.03
Dist. model: 5 coefficients
Topic color: /kinect2/hd/image_mono
   Topic ir: /kinect2/sd/image_ir
Topic depth: /kinect2/sd/image_depth
       Path: ./

restoring files...
[New Thread 0x7fffdf49b700 (LWP 5366)]
[New Thread 0x7fffdec9a700 (LWP 5367)]
[New Thread 0x7fffde499700 (LWP 5368)]
[New Thread 0x7fffddc98700 (LWP 5369)]
[New Thread 0x7fffdd497700 (LWP 5370)]
[New Thread 0x7fffdcc96700 (LWP 5371)]
[New Thread 0x7fffdc495700 (LWP 5372)]
starting calibration...
frame: ./0000_depth.png
[New Thread 0x7fffdb71c700 (LWP 5373)]
[New Thread 0x7fffdb31b700 (LWP 5374)]
[New Thread 0x7fffdaf1a700 (LWP 5375)]
[New Thread 0x7fffdab19700 (LWP 5376)]
[New Thread 0x7fffda317700 (LWP 5378)]
[New Thread 0x7fffda718700 (LWP 5377)]
[New Thread 0x7fffd9f16700 (LWP 5379)]
frame: ./0000_sync_depth.png
frame: ./0001_depth.png
frame: ./0001_sync_depth.png
frame: ./0002_depth.png
frame: ./0002_sync_depth.png

Program received signal SIGSEGV, Segmentation fault.
0x0000000000427605 in DepthCalibration::compareDists(std::vector<double, std::allocator<double> > const&, std::vector<double, std::allocator<double> > const&) const ()
(gdb) backtrace 
#0  0x0000000000427605 in DepthCalibration::compareDists(std::vector<double, std::allocator<double> > const&, std::vector<double, std::allocator<double> > const&) const ()
#1  0x0000000000427dc5 in DepthCalibration::calibrate() ()
#2  0x0000000000413571 in main ()

wiedemeyer, 2015-10-20:
Something seems to be wrong. Maybe the input vectors are empty. Why does the calibration only load 6 frames? Shouldn't there be more? How did you managed to calibrate the intrinsic and extrinsic parameters with just 3 images each?

wiedemeyer, 2015-10-26:
I added some more checks to the calibration tool. It should provide some more meaningful error messages. Please update and check again.

closed 2015-11-05 by wiedemeyer