code-iai/iai_kinect2 #163: Flashing black screen after calibration

renxinhe, 2015-10-08:
Hello,
I've recently calibrated the Kinect sensor with the checker board pattern. When I look at the output from kinect2_viewer, however, the screen is constantly flashing black. I've attached some screenshots. I've also noticed that when the viewer is black, the RGB topic outputs a bunch of zeros, and the viewer fps shots to above 100, compared to 20 fps when there is a crisp image. Is it possible that the amount of data is too high for the computer to handle?

Thanks,
Jim

[image]
[image]

wiedemeyer, 2015-10-14:
I don't know. I never experienced anything like that before. Since it's an OpenCL error, maybe it is related to the GPU. What GPU are you using. What is the CPU usage while running kinect2_bridge and the viewer?
Try to run kinect2_bridge and then subscribe to a topic with `rostopic hz /kinect2/hd/image_depth_rect`. Do the errors show up too?

renxinhe, 2015-10-14:
The laptop is running on NVIDIA® GTX 860M. It has built-in Intel HD graphics.

I've ran the kinect2_bridge, and here is a screenshot of the output:
[image]

It doesn't seem to generate any error.

wiedemeyer, 2015-10-15:
It's probably related to the opencl errors, if it is not running correctly, the depth image is wrong to.
The `-5` means `CL_OUT_OF_RESOURCES`, maybe running the cloud viewer and the kinect2_bridge is using to much VRAM, or there are also some other applications using the GPU. I don't know how to solve this issue, maybe it helps to use a newer nvidia driver, or by closing windows that are not needed, or don't use point cloud viewer and use `rosrun kinect2_viewer kinect2_viewer ... -image` instead.
One other thing that looks wrong is `... avg. time: 0ms` So there is something wrong with the measurement in libfreenect2.

wiedemeyer, 2015-10-26:
Hi, I updated the OpenCL code. Could you pull the lastest changes and check again?

closed 2015-11-05 by wiedemeyer