code-iai/iai_kinect2 #164: GL_MAX_RECTANGLE_TEXTURE_SIZE is too small: 0

mwswartwout, 2015-10-08:
I've seen the previous issue raised with this and the response that it is caused by OpenGL and a switch to OpenCL should solve it. I'm on a Nvidia GPU and have the latest drivers and have opencl-headers installed. When starting kinect2_bridge, after the calibration data is displayed, I get this output:
> [DepthRegistration::New] Using OpenCL registration method!
> [DepthRegistration::New] Using OpenCL registration method!
>[DepthRegistrationOpenCL::listDevice]  devices: 0: GeForce GTX 645 (GPU)[NVIDIA Corporation]
>[DepthRegistrationOpenCL::init]  selected device: GeForce GTX 645 (GPU)[NVIDIA Corporation]
>[DepthRegistrationOpenCL::listDevice]  devices: 0: GeForce GTX 645 (GPU)[NVIDIA Corporation]
>[DepthRegistrationOpenCL::init]  selected device: GeForce GTX 645 (GPU)[NVIDIA Corporation]
>[kinect2_bridge] waiting for clients to connect

Then, when I connect a client, in this case Rviz, and attempt to view the point clouds I get this:
>[kinect2_bridge] client connected. starting device...
>[Info] [Freenect2DeviceImpl] starting...
>[Error] [Texture<FormatT>] GL_MAX_RECTANGLE_TEXTURE_SIZE is too small: 0 terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >' what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument

Is there another step to setting up OpenCL correctly?

wiedemeyer, 2015-10-09:
It uses OpenCL for the registration, but the depth processor from libfreenect2 probably still uses OpenGL. Try using the OpenCL depth processor with the parameter `_depth_method:=opencl`. If that is not working, then libfreenect2 is compiled without OpenCL support. Make sure you compiled libfreenect2 with OpenCL support and recompile kinect2_bridge. Then try again.

mwswartwout, 2015-10-13:
I can run `./Protonect cl` with no issues, however when I do `rosrun kinect2_bridge kinect2_bridge _depth_method:=opencl` I get this:
> parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/mattswartwout/ros_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

>OpenCL depth processing is not available!
>Initialization failed!

wiedemeyer, 2015-10-14:
Please make sure you ran `sudo make install` in libfreenect2, after you enabled OpenCL there. Than you have to recompile kinect2_bridge, so that the changes in libfreenect2 are propagated to kinect2_bridge.

mwswartwout, 2015-10-14:
I'm unclear on this, is there a step to enabling OpenCL in libfreenect2? All I have done is installed `opencl-headers` and done run `sudo make install` in libfreenect2. I am able to run `./Protonect cl` with no issues, does that indicate that OpenCL is enabled in libfreenect2? I rebuilt kinect2_bridge and I still receive the same error.

wiedemeyer, 2015-10-14:
Check the `include/libfreenect2/config.h` for `#define LIBFREENECT2_WITH_OPENCL_SUPPORT` (in the install location, probably `/usr/local`). If it is there everything is fine.
Force to rebuild kinect2_bridge by removing the devel and build folders in your catkin workspace `rm -rf <catkin_ws>/devel <catkin_ws>/build` and then `catkin_make` it again.

mwswartwout, 2015-10-14:
That did it, thanks!

closed 2015-10-14 by mwswartwout