code-iai/iai_kinect2 #165: [Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!

wkentaro, 2015-10-09:
```
$ roslaunch kinect2_bridge kinect2_bridge.launch publish_tf:=true depth_method:=cpu
..

[kinect2_bridge] client connected. starting device...
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[kinect2_bridge] depth processing: ~7291.91Hz (0.137138ms) publishing rate: ~3.3309Hz
[kinect2_bridge] color processing: ~90.3696Hz (11.0657ms) publishing rate: ~22.9832Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.809ms -> ~59.492Hz
[Info] [DepthPacketStreamParser] 97 of 30 packets were lost
[kinect2_bridge] depth processing: ~10244.6Hz (0.0976125ms) publishing rate: ~3.98773Hz
[kinect2_bridge] color processing: ~92.3814Hz (10.8247ms) publishing rate: ~29.908Hz
[Info] [DepthPacketStreamParser] 52 of 30 packets were lost
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.9469ms -> ~59.0077Hz
[kinecany idea to fix this?t2_bridge] depth processing: ~14020Hz (0.0713269ms) publishing rate: ~3.98662Hz
[kinect2_bridge] color processing: ~90.711Hz (11.024ms) publishing rate: ~28.903Hz
[kinect2_bridge] depth processing: ~14074.8Hz (0.0710487ms) publishing rate: ~0.333159Hz
[kinect2_bridge] color processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~-nanHz (-nanms) publishing rate: ~0Hz
```

I installed libfreenect2 and iai_kinect2 following instruction at README, but I have error above.
At the first subscription, I can see the depth points and color image, but it stops soon.
any idea to fix this?
my environment is

* ubuntu 14.04
* ros indigo

wiedemeyer, 2015-10-09:
Either nothing is received from libfreenect2 or something freezes. I think there might be a problem with libfreenect2. Please test if Protonect is running without any issues. If it works, you could add debug output to `receiveFrames` to see if new frames are received.

wkentaro, 2015-10-09:
I see. thanks for your advice.
I created issue at libfreenect2:
https://github.com/OpenKinect/libfreenect2/issues/421

wkentaro, 2015-10-10:
That works. thanks!

closed 2015-10-10 by wkentaro