code-iai/iai_kinect2 #166: Failed to open X display at launch kinect2_bridge.launch with machine tag

wkentaro, 2015-10-10:
```
$ ssh doura

$ roslaunch kinect2_bridge kinect2_bridge.launch base_name:=kinect2_torso
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65542 X11: Failed to open X display
[Error] [OpenGLDepthPacketProcessor] Failed to initialize GLFW.
process[kinect2_torso_points_xyzrgb_sd-3]: started with pid [19591]
process[kinect2_torso_points_xyzrgb_qhd-4]: started with pid [19596]
[FATAL] [1444463086.167830097]: Service call failed!
[FATAL] [1444463086.167946042]: Service call failed!
[FATAL] [1444463086.170023792]: Service call failed!
[kinect2_torso-1] process has died [pid 19567, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=kinect2_torso __log:=/home/baxter/.ros/log/783b6388-6e88-11e5-9fb2-b8ca3ab5d80e/kinect2_torso-1.log].
log file: /home/baxter/.ros/log/783b6388-6e88-11e5-9fb2-b8ca3ab5d80e/kinect2_torso-1*.log
process[kinect2_torso_points_xyzrgb_hd-5]: started with pid [19601]
[kinect2_torso_points_xyzrgb_qhd-4] process has died [pid 19596, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2_torso rgb/camera_info:=kinect2_torso/qhd/camera_info rgb/image_rect_color:=kinect2_torso/qhd/image_color_rect depth_registered/image_rect:=kinect2_torso/qhd/image_depth_rect depth_registered/points:=kinect2_torso/qhd/points __name:=kinect2_torso_points_xyzrgb_qhd __log:=/home/baxter/.ros/log/783b6388-6e88-11e5-9fb2-b8ca3ab5d80e/kinect2_torso_points_xyzrgb_qhd-4.log].
log file: /home/baxter/.ros/log/783b6388-6e88-11e5-9fb2-b8ca3ab5d80e/kinect2_torso_points_xyzrgb_qhd-4*.log
[kinect2_torso_points_xyzrgb_qhd-4] restarting process
process[kinect2_torso_points_xyzrgb_qhd-4]: started with pid [19602]
[kinect2_torso_bridge-2] process has died [pid 19582, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2_torso __name:=kinect2_torso_bridge __log:=/home/baxter/.ros/log/783b6388-6e88-11e5-9fb2-b8ca3ab5d80e/kinect2_torso_bridge-2.log].
log file: /home/baxter/.ros/log/783b6388-6e88-11e5-9fb2-b8ca3ab5d80e/kinect2_torso_bridge-2*.log
[kinect2_torso_points_xyzrgb_sd-3] process has died [pid 19591, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2_torso rgb/camera_info:=kinect2_torso/sd/camera_info rgb/image_rect_color:=kinect2_torso/sd/image_color_rect depth_registered/image_rect:=kinect2_torso/sd/image_depth_rect depth_registered/points:=kinect2_torso/sd/points __name:=kinect2_torso_points_xyzrgb_sd __log:=/home/baxter/.ros/log/783b6388-6e88-11e5-9fb2-b8ca3ab5d80e/kinect2_torso_points_xyzrgb_sd-3.log].
log file: /home/baxter/.ros/log/783b6388-6e88-11e5-9fb2-b8ca3ab5d80e/kinect2_torso_points_xyzrgb_sd-3*.log
[kinect2_torso_bridge-2] restarting process
process[kinect2_torso_bridge-2]: started with pid [19611]
[kinect2_torso_points_xyzrgb_sd-3] restarting process
process[kinect2_torso_points_xyzrgb_sd-3]: started with pid [19612]
[ INFO] [1444463086.418958273]: Loading nodelet /kinect2_torso_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2_torso with the following remappings:
[ INFO] [1444463086.423116316]: waitForService: Service [/kinect2_torso/load_nodelet] has not been advertised, waiting...
```

Is that possible to use machine tag to launch kinect2_bridge?

wiedemeyer, 2015-10-10:
Yes, just give the machine as parameter `roslaunch kinect2_bridge kinect2_bridge.launch machine:=<somemachine>`. But you need to configure the X server on the remote machine correctly, so that the OpenGL code has access to the GPU. I don't remember what to do exactly, but maybe `xhost +` on the remote machine helps.

wkentaro, 2015-10-14:
I tried `xhost +` but the error output does not change.
```
% xhost +
access control disabled, clients can connect from any host
```

vishu2287, 2015-10-15:
ssh -X doura@192.xxx.x.xxx

wkentaro, 2015-10-15:
> ssh -X doura@192.xxx.x.xxx

I also tried ssh with `-X` option, but it does not work actually.

wiedemeyer, 2015-10-15:
Try to use OpenCL if you want to run it on a remote machine.
For OpenGL you need to setup [virtualgl](http://www.virtualgl.org/) and then you have to use `vglconnect` and `vglrun`.

wkentaro, 2015-10-15:
I installed

* opencl-headers
* nvidia-modprobe

and re-compiled libfreenect and iai_kinect2.

Now I have error below:
```
[ INFO] [1444896892.170559084]: Loading nodelet /kinect2_torso_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2_torso with the following remappings:
process[kinect2_torso_points_xyzrgb_sd-4]: started with pid [21276]
[ INFO] [1444896892.173749651]: waitForService: Service [/kinect2_torso/load_nodelet] has not been advertised, waiting...
process[kinect2_torso_points_xyzrgb_qhd-5]: started with pid [21283]
[ INFO] [1444896892.199195468]: Initializing nodelet with 8 worker threads.
process[kinect2_torso_points_xyzrgb_hd-6]: started with pid [21323]
[ INFO] [1444896892.215391866]: waitForService: Service [/kinect2_torso/load_nodelet] is now available.
process[doura_static_tf_publisher-7]: started with pid [21351]
modprobe: ERROR: could not insert 'fglrx': No such device
Error: Fail to load fglrx kernel module!
Error! Fail to load fglrx kernel module! Maybe you can switch to root user to load kernel module directly
modprobe: ERROR: could not insert 'fglrx': No such device
Error: Fail to load fglrx kernel module!
Error! Fail to load fglrx kernel module! Maybe you can switch to root user to load kernel module directly
modprobe: ERROR: could not insert 'fglrx': No such device
Error: Fail to load fglrx kernel module!
Error! Fail to load fglrx kernel module! Maybe you can switch to root user to load kernel module directly
modprobe: FATAL: Module nvidia-uvm not found.
parameter:
        base_name: kinect2_torso
           sensor: 
        fps_limit: -1
       calib_path: /home/baxter/ros/indigo/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: default
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2_torso
   worker_threads: 4

[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel HD Graphics Family (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl]   1: Intel(R) Core(TM) i7-3920XM CPU @ 2.90GHz (CPU)[GenuineIntel]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel HD Graphics Family (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
/tmp/fileNuJ2Rj.cl:16803:54: warning: double precision constant requires cl_khr_fp64, casting to single pr
```

If I try to install fglrx, nvidia gpu kernel will be removed..
```
% sudo aptitude install fglrx
[sudo] password for baxter: 
Sorry, try again.
[sudo] password for baxter: 
The following NEW packages will be installed:
  fglrx{b} fglrx-amdcccle{a} 
0 packages upgraded, 2 newly installed, 0 to remove and 5 not upgraded.
Need to get 5,121 kB/42.7 MB of archives. After unpacking 182 MB will be used.
The following packages have unmet dependencies:
 nvidia-346 : Conflicts: xorg-driver-binary which is a virtual package.
 fglrx : Conflicts: xorg-driver-binary which is a virtual package.
The following actions will resolve these dependencies:

     Remove the following packages:
1)     nvidia-346                  
2)     nvidia-346-dev              

Accept this solution? [Y/n/q/?] 
```

Do you know how to fix this?
Or should I create another issue?

wiedemeyer, 2015-10-15:
`fglrx` is the amd driver, but if you don't have an amd gpu you shouldn't install it. Maybe you need to install `nvidia-346-uvm`.

wkentaro, 2015-10-15:
I don't have amd gpu.
I installed `nvidia-346-uvm` but have same error. (I'm launching kinect2 on local, not with ssh)
I compiled libfreenect2 with `-DENABLE_OPENCL=ON` option.

wiedemeyer, 2015-10-15:
So you are running it local. Is Protonect from libfreenect working? And I just saw, your GPU is not showing up in the OpenCL devices list, but you Intel CPU and GPU. Do you have beignet installed? Maybe some opencl configurations are messed up.

wkentaro, 2015-10-15:
Protonect seems working.
[image]

and beignet is installed:

```
% dpkg --get-selections | grep beignet
beignet						install
beignet-dev					install
```

wiedemeyer, 2015-10-15:
did you run `Protonect cl` or just `Protonect`?

wkentaro, 2015-10-15:
It is just `$ ./Protonect`.

wkentaro, 2015-10-15:
I did not have window when `Protonect cl`
``
% ./Protonect cl
modprobe: ERROR: could not insert 'fglrx': No such device
Error: Fail to load fglrx kernel module!
Error! Fail to load fglrx kernel module! Maybe you can switch to root user to load kernel module directly
modprobe: ERROR: could not insert 'fglrx': No such device
Error: Fail to load fglrx kernel module!
Error! Fail to load fglrx kernel module! Maybe you can switch to root user to load kernel module directly
modprobe: ERROR: could not insert 'fglrx': No such device
Error: Fail to load fglrx kernel module!
Error! Fail to load fglrx kernel module! Maybe you can switch to root user to load kernel module directly
modprobe: FATAL: Module nvidia-uvm not found.
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:8 with serial 502489742542
[Info] [Freenect2Impl] found 1 devices
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel HD Graphics Family (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl]   1: Intel(R) Core(TM) i7-3920XM CPU @ 2.90GHz (CPU)[GenuineIntel]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel HD Graphics Family (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
/tmp/file96xGpF.cl:16803:54: warning: double precision constant requires cl_khr_fp64, casting to single precision
  float max_depth = phase_final * UNAMBIGIOUS_DIST * 2.0;
                                                     ^
/tmp/file96xGpF.cl:16807:32: warning: double precision constant requires cl_khr_fp64, casting to single precision
  xmultiplier = (xmultiplier * 90.0) / (max_depth * max_depth * 8192.0);
                               ^
/tmp/file96xGpF.cl:16807:65: warning: double precision constant requires cl_khr_fp64, casting to single precision
  xmultiplier = (xmultiplier * 90.0) / (max_depth * max_depth * 8192.0);
                                                                ^
3 warnings generated.
ASSERTION FAILED: TODO Boolean values cannot escape their definition basic block
  at file /build/buildd/beignet-0.3/backend/src/llvm/llvm_gen_backend.cpp, function void gbe::GenWriter::emitMovForPHI(llvm::BasicBlock*, llvm::BasicBlock*), line 1012
Stack dump:
0.	Running pass 'Function Pass Manager' on module '/tmp/fileZjm3Sn.ll'.
1.	Running pass 'Gen Back-End' on function '@processPixelStage2'
```

wiedemeyer, 2015-10-15:
There is something wrong with your OpenCL setup. You could try to uninstall beignet and use the nvidia card instead. Make sure `nvidia-opencl-icd-xxx` is also installed.

wkentaro, 2015-10-15:
There are lots of debug info, and warning, but the window appears now.

```
% sudo aptitude purge beignet
% dpkg --get-selections | grep nvidia
nvidia-346                                      install
nvidia-346-dev                                  install
nvidia-346-uvm                                  install
nvidia-modprobe                                 install
nvidia-opencl-icd-346                           install
nvidia-prime                                    install
nvidia-settings                                 install

```
[image]

wiedemeyer, 2015-10-15:
At least it works. The `skipping package` probably comes from a high CPU load.
Did it listed your GPU in the OpenCL devices list and did it stated it as selected device? 
```
...
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: GeForce GTX 970 (GPU)[NVIDIA Corporation]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: GeForce GTX 970 (GPU)[NVIDIA Corporation]
...
```

wkentaro, 2015-10-16:
```
% ./Protonect cl
modprobe: ERROR: could not insert 'fglrx': No such device
Error: Fail to load fglrx kernel module!
Error! Fail to load fglrx kernel module! Maybe you can switch to root user to load kernel module directly
modprobe: ERROR: could not insert 'fglrx': No such device
Error: Fail to load fglrx kernel module!
Error! Fail to load fglrx kernel module! Maybe you can switch to root user to load kernel module directly
modprobe: ERROR: could not insert 'fglrx': No such device
Error: Fail to load fglrx kernel module!
Error! Fail to load fglrx kernel module! Maybe you can switch to root user to load kernel module directly
modprobe: FATAL: Module nvidia-uvm not found.
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:8 with serial 502489742542
[Info] [Freenect2Impl] found 1 devices
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) Core(TM) i7-3920XM CPU @ 2.90GHz (CPU)[GenuineIntel]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) Core(TM) i7-3920XM CPU @ 2.90GHz (CPU)[GenuineIntel]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Warning] [OpenCLDepthPacketProcessorImpl] Your ocl-icd has deadlock bugs. Update to 2.2.4+ is recommended.
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] ReadData0x14 response
[Debug] [Freenect2DeviceImpl] 92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 02 04 0d 47 4d 46 31 33 33 2e 31    31332e2e2e2e2e2e474d463133332e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 502489742542
device firmware: 2.3.3912.0.7
```
cpu is selected

```
```

wiedemeyer, 2015-10-17:
If the CPU is used for OpenCL, then it's similar to the cpu depth processor. Somehow you GPUis not showing up. Check if there is a file which points to some nvidia lib in `/etc/OpenCL/vendors`. Check if the `nvidia-opencl-icd-xxx` package is installed.

wkentaro, 2015-10-18:
I have nvidia.icd in /etc/OpenCL/vendors, and nvidia-opencl-icd-346 installed, so now I will follow the opencl installation instruction here https://github.com/OpenKinect/libfreenect2#debianubuntu-1404-perhaps-earlier.

wkentaro, 2015-10-18:
I had I had intel.icd in /etc/OpenCL/vendors (not only nvidia.icd) and after moving that file, I have

[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Error] [OpenCLDepthPacketProcessor] could not initialize OpenCLDepthPacketProcessor
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...
[Error] [OpenCLDepthPacketProcessor] could not initialize OpenCLDepthPacketProcessor
[Error] [OpenCLDepthPacketProcessor] could not initialize OpenCLDepthPacketProcessor
[Info] [Freenect2DeviceImpl] stopped
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing
when ./bin/Protonect cl and no images and points come.

./bin/Protonect does work.

wkentaro, 2015-10-18:
`nvidia.icd` is still there.
```
% cat /etc/OpenCL/vendors/nvidia.icd
libnvidia-opencl.so.1
```

wiedemeyer, 2015-10-19:
I think there are some issues with the configuration of OpenCL on your system. It's not really related to iai_kinect nor libfreenect2. I don't know how to fix it. But you could try to run it on a different machine, maybe that helps to find out what configuration is wrong on the other one.

wiedemeyer, 2015-10-19:
Or you could uninstall all the nvidia drivers, beignet, and other opencl related drivers, like intel sdk etc. Then reinstall only nvidia drivers and opencl headers and test again.

wkentaro, 2015-10-19:
I see. I will try it when I have time.
Thanks.

vishu2287, 2015-10-22:
I have solved this by creating ros network in ~/.bashrc and specified the rosmaster where the kinect2 is connected.
its working fine.

wkentaro, 2015-10-22:
@vishu2287
What do you mean by 'ros network'?
Do u mean ROS_MASTER_URI?

vishu2287, 2015-10-22:
yes.  add it in your bashrc

in host computer
export ROS_HOSTNAME=(host)
export ROS_MASTER_URI=http://(master):11311

in master computer (where kinect2 is connected
export ROS_HOSTNAME=(master)
export ROS_MASTER_URI=http://(master):11311

wkentaro, 2015-10-22:
actually, those are configured already.

vishu2287, 2015-10-28:
hello @wkentaro 
Change the depth_method in host machine while launching
$ ssh blah@192.xxx.x.x
$ sudo -s
# roslaunch kinect2_bridge kinect2_bridge depth_method:=cpu

wkentaro, 2015-10-28:
@vishu2287 

That works! thanks :sparkles:

closed 2015-11-05 by wiedemeyer

GaiTech-Robotics, 2015-12-23:
The other way to overcome GLFW error during running kinect2_bridge in x server can be done by running in remote desktop method
In host pc
$ gvncviewer <ip of master  pc (which kinect2 is connected)>
Before doing this, in master pc you should change options for remote desktop sharing

Once gvncviewer is running, your master pc desktop will appears in host pc. You can run kinect2_bridge directly there.. remaining launch files like kinect2_viewer can be run directly in host pc in another terminal  (ros network need to be configured)

vishu2287, 2015-12-23:
The other way to overcome GLFW error during running kinect2_bridge in x server can be done by running in remote desktop method
In host pc
$ gvncviewer "ip of master pc (which kinect2 is connected)"
Before doing this, in master pc you should change options for remote desktop sharing

Once gvncviewer is running, your master pc desktop will appears in host pc. You can run kinect2_bridge directly there.. remaining launch files like kinect2_viewer can be run directly in host pc in another terminal (ros network need to be configured)