code-iai/iai_kinect2 #167: Trying to Run kinect2_bridge using Beignet on Intel HD

rahulraw, 2015-10-13:
How does one get kinect2_bridge running using OpenCL (Beignet) 
If I set Depth_Method and Reg_Method to opencl
I get the following error:
```
[ INFO] [1444760667.340297521]: Connected to master at [localhost:11311]
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/pr2kinect/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

OpenCL depth processing is not available!
Initialization failed!
```
Please note that I installed Beignet after installing iai_kinect2 and libfreenect2 
Is there something I must do during the build process to get this to work?

rahulraw, 2015-10-13:
So it looks like the right way to do this is getting beignet before installing iai_kinect2. (I took the precaution and did it before libfreenect2 as well). Can anyone confirm this?

wiedemeyer, 2015-10-14:
It's ok to install beignet afterwards, but you have the reconfigure and recompile libfreenect2 and kinect2_bridge after installing beignet.
Please make sure you ran `sudo make install` in libfreenect2, after you installed beignet. Than you have to recompile kinect2_bridge, so that the changes in libfreenect2 are propagated to kinect2_bridge.

wiedemeyer, 2015-10-26:
I updated the OpenCL code. Please pull and check again.

rahulraw, 2015-10-30:
I actually got this working on my system with OpenCL depth and registration method.
However, after pulling the latest commit on another system I get the following error
```
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65542 X11: Failed to open X display
[Error] [OpenGLDepthPacketProcessor] Failed to initialize GLFW.
[FATAL] [1446216332.337094919]: Service call failed!
[FATAL] [1446216332.337396245]: Service call failed!
[FATAL] [1446216332.337627642]: Service call failed!
[FATAL] [1446216332.339660598]: Service call failed!
No handlers could be found for logger "roslaunch"
[kinect2-2] process has died [pid 1199, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=kinect2 __log:=/tmp/ddf078b6-7f14-11e5-a1ab-8b0c174e4c71/kinect2-2.log].
log file: /tmp/ddf078b6-7f14-11e5-a1ab-8b0c174e4c71/kinect2-2*.log
[kinect2_bridge-3] process has died [pid 1220, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/tmp/ddf078b6-7f14-11e5-a1ab-8b0c174e4c71/kinect2_bridge-3.log].
log file: /tmp/ddf078b6-7f14-11e5-a1ab-8b0c174e4c71/kinect2_bridge-3*.log
[kinect2_points_xyzrgb_qhd-5] process has died [pid 1298, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/qhd/camera_info rgb/image_rect_color:=kinect2/qhd/image_color_rect depth_registered/image_rect:=kinect2/qhd/image_depth_rect depth_registered/points:=kinect2/qhd/points __name:=kinect2_points_xyzrgb_qhd __log:=/tmp/ddf078b6-7f14-11e5-a1ab-8b0c174e4c71/kinect2_points_xyzrgb_qhd-5.log].
log file: /tmp/ddf078b6-7f14-11e5-a1ab-8b0c174e4c71/kinect2_points_xyzrgb_qhd-5*.log
[kinect2_points_xyzrgb_hd-6] process has died [pid 1317, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/hd/camera_info rgb/image_rect_color:=kinect2/hd/image_color_rect depth_registered/image_rect:=kinect2/hd/image_depth_rect depth_registered/points:=kinect2/hd/points __name:=kinect2_points_xyzrgb_hd __log:=/tmp/ddf078b6-7f14-11e5-a1ab-8b0c174e4c71/kinect2_points_xyzrgb_hd-6.log].
log file: /tmp/ddf078b6-7f14-11e5-a1ab-8b0c174e4c71/kinect2_points_xyzrgb_hd-6*.log
[kinect2_bridge-3] restarting process
process[kinect2_bridge-3]: started with pid [1715]
[kinect2_points_xyzrgb_qhd-5] restarting process
process[kinect2_points_xyzrgb_qhd-5]: started with pid [1716]
[kinect2_points_xyzrgb_hd-6] restarting process
process[kinect2_points_xyzrgb_hd-6]: started with pid [1717]
[ INFO] [1446216332.634953049]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1446216332.642175427]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
[kinect2_points_xyzrgb_sd-4] process has died [pid 1274, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/sd/camera_info rgb/image_rect_color:=kinect2/sd/image_color_rect depth_registered/image_rect:=kinect2/sd/image_depth_rect depth_registered/points:=kinect2/sd/points __name:=kinect2_points_xyzrgb_sd __log:=/tmp/ddf078b6-7f14-11e5-a1ab-8b0c174e4c71/kinect2_points_xyzrgb_sd-4.log].
log file: /tmp/ddf078b6-7f14-11e5-a1ab-8b0c174e4c71/kinect2_points_xyzrgb_sd-4*.log
[kinect2_points_xyzrgb_sd-4] restarting process
process[kinect2_points_xyzrgb_sd-4]: started with pid [1763]
pr2-head@pr2head-desktop:/var/log/upstart$ 
```

I am running this process at upstart and this works on my other computer at upstart. What has been changed ?

wiedemeyer, 2015-10-31:
Looks like a problem with the x server. Maybe X is not running? Your could try OpenCL instead of OpenGL. But it's either a problem of the system or (less likely) libfreenect2.

rahulraw, 2015-11-02:
That's the other thing. I want to launch OpenCL actually. I have beignet installed but with robot upstart it seems to launch in OpenCl mode

rahulraw, 2015-11-04:
This issue was solved. It seems that installing Beignet by doing
```
sudo apt-get install beignet-dev
```

does not install beignet-opencl-icd which is needed to build libfreenect2. Interestingly, installing that package has solved my problem.

wiedemeyer, 2015-11-05:
good to know.

closed 2015-11-05 by wiedemeyer