code-iai/iai_kinect2 #168: cv::Exception when trying to calibrate camera

klajdi123, 2015-10-14:
I installed the whole package and checked if both libfreenect2 and iai_kinect2 works. They work both with opengl and opencl. My problem is when I try to calibrate the camera I get the following error:
```
Start settings:
       Mode: calibrate
     Source: color
      Board: chess
 Dimensions: 5 x 7
 Field size: 0.03
Dist. model: 5 coefficients
Topic color: /kinect2/hd/image_mono
   Topic ir: /kinect2/sd/image_ir
Topic depth: /kinect2/sd/image_depth
       Path: ./

restoring files...
starting calibration...
calibrating intrinsics...
OpenCV Error: Assertion failed (nimages > 0) in calibrateCamera, file /usr/share/OpenCV/opencv-2.4.11/modules/calib3d/src/calibration.cpp, line 3415
terminate called after throwing an instance of 'cv::Exception'
  what():  /usr/share/OpenCV/opencv-2.4.11/modules/calib3d/src/calibration.cpp:3415: error: (-215) nimages > 0 in function calibrateCamera
```
The order of the commands is exactly as mentioned in Kinect2 Calibration:
```
rosrun kinect2_bridge kinect2_bridge _fps_limit:=2
rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 record color
rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 calibrate color
```

Also, sometimes I get the error:
```
GL_MAX_RECTANGLE_TEXTURE_SIZE is too small: 0
```
but, if I run it again it works normally

wiedemeyer, 2015-10-14:
No images are restored and a calibration with no input data will fail. You need to run the `record color` first and record data. There should be a couple of files created in the current directory. Then execute `calibrate color` in the same directory.

klajdi123, 2015-10-14:
Thanks for the quick reply; I tried to run it but, it doesn't save any data in the folder.
Also, I noticed that when it prompts me to input a option it seems that it doesn't accept the input.

I run:
```
rosrun kinect2_bridge kinect2_bridge _fps_limit:=2
rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 record color
rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 record data
```

From ```record color``` I get the below message:
```
Start settings:
       Mode: record
     Source: color
      Board: chess
 Dimensions: 5 x 7
 Field size: 0.03
Dist. model: 5 coefficients
Topic color: /kinect2/hd/image_mono
   Topic ir: /kinect2/sd/image_ir
Topic depth: /kinect2/sd/image_depth
       Path: ./

starting recorder...
Controls:
   [ESC, q] - Exit
 [SPACE, s] - Save current frame
        [l] - decreas min and max value for IR value rage
        [h] - increas min and max value for IR value rage
        [1] - decreas min value for IR value rage
        [2] - increas min value for IR value rage
        [3] - decreas max value for IR value rage
        [4] - increas max value for IR value rage
init done 
opengl support available 
```
and from ```record data``` the below message:
```
Start settings:
       Mode: record
     Source: sync
      Board: chess
 Dimensions: 5 x 7
 Field size: 0.03
Dist. model: 5 coefficients
Topic color: /data/hd/image_mono
   Topic ir: /data/sd/image_ir
Topic depth: /data/sd/image_depth
       Path: ./

starting recorder...
Controls:
   [ESC, q] - Exit
 [SPACE, s] - Save current frame
        [l] - decreas min and max value for IR value rage
        [h] - increas min and max value for IR value rage
        [1] - decreas min value for IR value rage
        [2] - increas min value for IR value rage
        [3] - decreas max value for IR value rage
        [4] - increas max value for IR value rage
```

klajdi123, 2015-10-14:
I think I found the problem... The data that I should get from the calibration are: one .png image and xxxx_color_points.yaml?

if yes, then I was inserting the argument for saving the frame from the terminal rather from the window... I did not know that you used ```waitKey()``` to wait an input. 

I am sorry for the inconvenience and thanks for the help .

closed 2015-10-14 by wiedemeyer