code-iai/iai_kinect2 #169: Problem running kinect2_viewer

klajdi123, 2015-10-15:
The package worked with ```kinect2_calibration``` (I finished with calibration) but, when I try to run ```kinect2_viewer``` I get the following error from ```gdb```:
```
Starting program: /home/klajdo/catkin_ws/devel/lib/kinect2_viewer/kinect2_viewer 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Traceback (most recent call last):
  File "/usr/share/gdb/auto-load/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.19-gdb.py", line 63, in <module>
    from libstdcxx.v6.printers import register_libstdcxx_printers
ImportError: No module named 'libstdcxx'

Program received signal SIGSEGV, Segmentation fault.
0x00007fffe8379dfc in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
```

Any idea? 
Thanks in advance.

wiedemeyer, 2015-10-15:
Compile it in debug, run it with gdb and paste a stack trace here.
Are you using the opencv version from the ubuntu repositories or a different one? If you are not using the one from ubuntu, change to that version and try again.

klajdi123, 2015-10-15:
Thank you for the quick reply. The above is in debug mode, and it is the stack trace from gdb.

I have installed opencv 2.4.11 from the official site of opencv and followed the steps from the following site:
https://help.ubuntu.com/community/OpenCV

wiedemeyer, 2015-10-15:
use the one from the official ubuntu repositories. `sudo apt-get install opencv-dev`

klajdi123, 2015-10-15:
I am sorry but, do you mean ```libopencv-dev``` because I can't find any ```opencv-dev```?

wiedemeyer, 2015-10-15:
yes

klajdi123, 2015-10-15:
I have already installed it; probably that may be the problem because I have both libraries installed. I will try to uninstall opencv 2.4.11.

klajdi123, 2015-10-15:
Thanks man; it works perfectly. I just run ```make uninstall``` in the build folder of opencv-2.4.11 directory

wiedemeyer, 2015-10-15:
what problems do you have?

klajdi123, 2015-10-15:
Sorry... When I run ```rosrun kinect2_viewer kinect2_viewer hd cloud``` I get the following message:
```
topic color: /kinect2/hd/image_color_rect
topic depth: /kinect2/hd/image_depth_rect
starting receiver...
Segmentation fault (core dumped)
```
Then when I run gdb I get frames but, with the following message:
```[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Traceback (most recent call last):
  File "/usr/share/gdb/auto-load/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.19-gdb.py", line 63, in <module>
    from libstdcxx.v6.printers import register_libstdcxx_printers
ImportError: No module named 'libstdcxx'
[New Thread 0x7fffced21700 (LWP 14263)]
topic color: /kinect2/qhd/image_color_rect
topic depth: /kinect2/qhd/image_depth_rect
[New Thread 0x7fffcdac2700 (LWP 14265)]
[New Thread 0x7fffcd2c1700 (LWP 14266)]
[New Thread 0x7fffccac0700 (LWP 14267)]
[New Thread 0x7fffc7fff700 (LWP 14272)]
starting receiver...
[New Thread 0x7fffc71b8700 (LWP 14299)]
[New Thread 0x7fffc69b7700 (LWP 14300)]
[New Thread 0x7fffc61b6700 (LWP 14301)]
[New Thread 0x7fffc59b5700 (LWP 14302)]
[New Thread 0x7fffc51b4700 (LWP 14303)]
[New Thread 0x7fffc49b3700 (LWP 14304)]
[New Thread 0x7fffa7fff700 (LWP 14305)]
[New Thread 0x7fffa77fe700 (LWP 14306)]
[New Thread 0x7fffa5f2a700 (LWP 14308)]
[New Thread 0x7fffa5729700 (LWP 14309)]
[New Thread 0x7fffa4f28700 (LWP 14310)]
[New Thread 0x7fff97fff700 (LWP 14311)]
[New Thread 0x7fff977fe700 (LWP 14312)]
[New Thread 0x7fff96ffd700 (LWP 14313)]
[New Thread 0x7fff967fc700 (LWP 14314)]
[setPointCloudRenderingProperties] Could not find any PointCloud datasets with id <rendered>!
```

wiedemeyer, 2015-10-16:
so `rosrun kinect2_viewer kinect2_viewer hd image` is working fine? If so than the problem is probably related to the pcl visualizer. The cloud with the id `rendered` should be added first. I don't see why it shouldn't.
Remove the devel and build folder `cd <catkin_ws>`, `rm -rf devel build` and then `catkin_make -DCMAKE_BUILD_TYPE=Debug`. Then start it again with gdb and do a `bt` after it failed.

klajdi123, 2015-10-16:
Exactly ``` rosrun kinect2_viewer kinect2_viewer hd image ``` works fine.

Now I did ``` catkin_make -DCMAKE_BUILD_TYPE="Debug" ``` in the workspace after deleting ``` devel ``` and ``` build ```. Then when I run ``` rosrun kinect2_viewer kinect2_viewer hd cloud ``` and I get the following message but, also I get frames:
```
[setPointCloudRenderingProperties] Could not find any PointCloud datasets with id <rendered>!
```
both when I run simple rosrun and gdb. By the way ``` bt ``` has no stack.
The message from ``` gdb ```
```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Traceback (most recent call last):
  File "/usr/share/gdb/auto-load/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.19-gdb.py", line 63, in <module>
    from libstdcxx.v6.printers import register_libstdcxx_printers
ImportError: No module named 'libstdcxx'
[New Thread 0x7fffce60f700 (LWP 6549)]
topic color: /kinect2/qhd/image_color_rect
topic depth: /kinect2/qhd/image_depth_rect
[New Thread 0x7fffcd3b0700 (LWP 6551)]
[New Thread 0x7fffccbaf700 (LWP 6552)]
[New Thread 0x7fffc7fff700 (LWP 6553)]
[New Thread 0x7fffc77fe700 (LWP 6558)]
starting receiver...
[New Thread 0x7fffc69b7700 (LWP 6585)]
[New Thread 0x7fffc61b6700 (LWP 6586)]
[New Thread 0x7fffc59b5700 (LWP 6587)]
[New Thread 0x7fffc51b4700 (LWP 6588)]
[New Thread 0x7fffc49b3700 (LWP 6589)]
[New Thread 0x7fffaffff700 (LWP 6590)]
[New Thread 0x7fffaf7fe700 (LWP 6591)]
[New Thread 0x7fffaeffd700 (LWP 6592)]
[New Thread 0x7fffad829700 (LWP 6593)]
[New Thread 0x7fffad028700 (LWP 6594)]
[New Thread 0x7fffac827700 (LWP 6595)]
[New Thread 0x7fffa7fff700 (LWP 6596)]
[New Thread 0x7fffa77fe700 (LWP 6597)]
[New Thread 0x7fffa6ffd700 (LWP 6598)]
[New Thread 0x7fffa67fc700 (LWP 6599)]
[setPointCloudRenderingProperties] Could not find any PointCloud datasets with id <rendered>!
bt
[Thread 0x7fffc51b4700 (LWP 6588) exited]
[Thread 0x7fffaf7fe700 (LWP 6591) exited]
[Thread 0x7fffc59b5700 (LWP 6587) exited]
[Thread 0x7fffc49b3700 (LWP 6589) exited]
[Thread 0x7fffc69b7700 (LWP 6585) exited]
[Thread 0x7fffaeffd700 (LWP 6592) exited]
[Thread 0x7fffc61b6700 (LWP 6586) exited]
[Thread 0x7fffaffff700 (LWP 6590) exited]
[Thread 0x7fffc77fe700 (LWP 6558) exited]
[Thread 0x7fffcd3b0700 (LWP 6551) exited]
[Thread 0x7fffccbaf700 (LWP 6552) exited]
[Thread 0x7fffce60f700 (LWP 6549) exited]
[Thread 0x7fffa6ffd700 (LWP 6598) exited]
[Thread 0x7fffa77fe700 (LWP 6597) exited]
[Thread 0x7fffa7fff700 (LWP 6596) exited]
[Thread 0x7fffac827700 (LWP 6595) exited]
[Thread 0x7fffad028700 (LWP 6594) exited]
[Thread 0x7fffad829700 (LWP 6593) exited]
[Thread 0x7fffc7fff700 (LWP 6553) exited]
[Thread 0x7ffff7f66f00 (LWP 6547) exited]
[Inferior 1 (process 6547) exited normally]
(gdb) bt
No stack.
```

wiedemeyer, 2015-10-17:
As I said, there should be a cloud added with that name. Only thing you could check is whether the cloud that is added is valid.

klajdi123, 2015-10-19:
I am sorry for the late response. You are right. Thanks a lot for the help

closed 2015-10-19 by klajdi123