code-iai/iai_kinect2 #170: Simulink Matlab bridge with iai_kinect2 low frame rates

shamoorti, 2015-10-19:
I'm currently working on a project which we want to use Simulink Matlab as our image processor. With the help of this bridge we could actually get the frame with a simple function which converts the Ros msg to readable image in Simulink. But the problem is that it is two slow and we're experiencing 3 seconds delay from ROS to Simulink. Does anyone has any idea if it can be improved to real time?
Thanks in  advance...

wiedemeyer, 2015-10-20:
I don't know much about simulink, but do you mean each frame is 3 seconds delayed or the conversion of one frame to simulink takes 3 seconds?

shamoorti, 2015-10-20:
The problem is we can't run opencl since some problems occurs which I will put it at the end of this massage. But the bridge with cpu registration method is killing 50 percent of an Intel Hazwell i7 and that's the cause of delay i think. In simulink we just use a block to subscribe the ROS message and a s-function converter to read the image. Matlab can do this conversion in 0.007ms and it's very fast. We tried to limit the fps to 5 and it worked better but still we are limiting publishing to 5 which is not good. Here is the opencl problem:
[OpenCLDepthPacketProcessor::init] ERROR: clGetDeviceIDs(-1)
terminate called after throwing an instance of 'cl::Error'
what(): clGetDeviceIDs
Aborted (core dumped)
¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤4
By the way I my CLquery is like this: 
lDeviceQuery Starting...

3 OpenCL Platforms found

CL_PLATFORM_NAME: Experiment Intel Gen OCL Driver
CL_PLATFORM_VERSION: OpenCL 1.1 beignet 0.3
OpenCL Device Info:

No devices found supporting OpenCL (return code -1)

clDeviceQuery, Platform Name = Experiment Intel Gen OCL Driver, Platform Version = OpenCL 1.1 beignet 0.3, NumDevs = 0
CL_PLATFORM_NAME: Intel Gen OCL Driver
CL_PLATFORM_VERSION: OpenCL 1.2 beignet 1.1.1
OpenCL Device Info:

1 devices found supporting OpenCL on: Intel Gen OCL Driver

Device Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3

CL_DEVICE_NAME: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3
CL_DEVICE_VENDOR: Intel
CL_DRIVER_VERSION: 1.1.1
CL_DEVICE_TYPE: CL_DEVICE_TYPE_GPU
CL_DEVICE_MAX_COMPUTE_UNITS: 48
CL_DEVICE_MAX_WORK_ITEM_DIMENSIONS: 3
CL_DEVICE_MAX_WORK_ITEM_SIZES: 512 / 512 / 512 
CL_DEVICE_MAX_WORK_GROUP_SIZE: 512
CL_DEVICE_MAX_CLOCK_FREQUENCY: 1000 MHz
CL_DEVICE_ADDRESS_BITS: 32
CL_DEVICE_MAX_MEM_ALLOC_SIZE: 1024 MByte
CL_DEVICE_GLOBAL_MEM_SIZE: 2048 MByte
CL_DEVICE_ERROR_CORRECTION_SUPPORT: no
CL_DEVICE_LOCAL_MEM_TYPE: global
CL_DEVICE_LOCAL_MEM_SIZE: 64 KByte
CL_DEVICE_MAX_CONSTANT_BUFFER_SIZE: 131072 KByte
CL_DEVICE_QUEUE_PROPERTIES: CL_QUEUE_PROFILING_ENABLE
CL_DEVICE_IMAGE_SUPPORT: 1
CL_DEVICE_MAX_READ_IMAGE_ARGS: 128
CL_DEVICE_MAX_WRITE_IMAGE_ARGS: 8

CL_DEVICE_IMAGE 2D_MAX_WIDTH 8192
2D_MAX_HEIGHT 8192
3D_MAX_WIDTH 8192
3D_MAX_HEIGHT 8192
3D_MAX_DEPTH 2048
CL_DEVICE_PREFERRED_VECTOR_WIDTH_ CHAR 16, SHORT 8, INT 4, FLOAT 2, DOUBLE 4

clDeviceQuery, Platform Name = Experiment Intel Gen OCL Driver, Platform Version = OpenCL 1.1 beignet 0.3, NumDevs = 0
Intel Gen OCL Driver, Platform Version = OpenCL 1.2 beignet 1.1.1, NumDevs = 1, Device = Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3
*** Error in `./clDeviceQuery': double free or corruption (fasttop): 0x0000000001a267f0 ***
Aborted (core dumped)

Note that I don't know what is that beignet 0.3 since I use 1.1 version and I did the Utests and it was successfull with only two failed tests.

wiedemeyer, 2015-10-20:
CPU depth processing is really slow, try to avoid it.
But the Beignet issue is strange. Maybe you have both versions installed or didn't uninstalled 0.3 correctly? Where are the files in `/etc/OpenCL/vendors` pointing to?

shamoorti, 2015-10-20:
We fixed the opencl problem in protonect and we are haveing opengl and opencl both working. But in the bridge it give this error:

rosrun kinect2_bridge kinect2_bridge _reg_method:=opencl _depth_method:=opengl
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/buttler/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:7 with serial 025613444747
[Info] [Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 025613444747 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened

starting kinect2

[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started

device serial: 025613444747
device firmware: 4.3.3913.0.7
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...
[Info] [Freenect2DeviceImpl] stopped

default ir camera parameters: 
fx 366.289, fy 366.289, cx 258.596, cy 207.897
k1 0.0939456, k2 -0.272606, p1 0, p2 0, k3 0.0955068

default color camera parameters: 
fx 1081.37, fy 1081.37, cx 959.5, cy 539.5
using sensor defaults for color intrinsic parameters.
using sensor defaults for ir intrinsic parameters.
using defaults for rotation and translation.
using defaults for depth shift.

camera parameters used:
camera matrix color:
[1081.3720703125, 0, 959.5;
  0, 1081.3720703125, 539.5;
  0, 0, 1]
distortion coefficients color:
[0, 0, 0, 0, 0]
camera matrix ir:
[366.2894897460938, 0, 258.5958862304688;
  0, 366.2894897460938, 207.8966979980469;
  0, 0, 1]
distortion coefficients ir:
[0.09394556283950806, -0.2726058959960938, 0, 0, 0.09550677239894867]
camera matrix depth:
[366.2894897460938, 0, 258.5958862304688;
  0, 366.2894897460938, 207.8966979980469;
  0, 0, 1]
distortion coefficients depth:
[0.09394556283950806, -0.2726058959960938, 0, 0, 0.09550677239894867]
rotation:
[1, 0, 0;
  0, 1, 0;
  0, 0, 1]
translation:
[0; 0; 0]
depth shift:
0

[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistrationOpenCL::init] ERROR: clGetDeviceIDs(-1)
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
Initialization failed!
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing

shamoorti, 2015-10-20:
I should also mention that I am using beignet 1.1 and Floe latest Libfreenect2 and everything is working on protonect now but not in bridge.

wiedemeyer, 2015-10-21:
If libfreenect2 works, kinect2_bridge should work as well. I will check if something in the OpenCL part changed. And those two commands are working fine?
`rosrun kinect2_bridge kinect2_bridge _reg_method:=cpu _depth_method:=opengl`
`rosrun kinect2_bridge kinect2_bridge _reg_method:=cpu _depth_method:=opencl`

In the mean time, try to recompile kinect2_bridge, remove devel and build folders in the catkin workspace to make sure it gets recompiled.

shamoorti, 2015-10-21:
Yes these two commands are working perfectly fine. This is the compile output:
####
#### Running command: "cmake /home/buttler/catkin_ws/src -DCMAKE_BUILD_TYPE=Release -DCATKIN_DEVEL_PREFIX=/home/buttler/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/buttler/catkin_ws/install -G Unix Makefiles" in "/home/buttler/catkin_ws/build"
####
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Using CATKIN_DEVEL_PREFIX: /home/buttler/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/buttler/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/buttler/catkin_ws/devel;/opt/ros/indigo
-- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/buttler/catkin_ws/build/test_results
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.14
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 5 packages in topological order:
-- ~~  - iai_kinect2 (metapackage)
-- ~~  - kinect2_registration
-- ~~  - kinect2_bridge
-- ~~  - kinect2_calibration
-- ~~  - kinect2_viewer
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'iai_kinect2'
-- ==> add_subdirectory(iai_kinect2/iai_kinect2)
-- +++ processing catkin package: 'kinect2_registration'
-- ==> add_subdirectory(iai_kinect2/kinect2_registration)
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Try OpenMP C flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Try OpenMP CXX flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Found OpenMP: -fopenmp  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- checking for module 'eigen3'
--   found eigen3, version 3.2.0
-- Found Eigen: /usr/include/eigen3  
-- Eigen found (include: /usr/include/eigen3)
-- Found OpenCL: /usr/lib/libOpenCL.so  
-- CPU based depth registration enabled
-- OpenCL based depth registration enabled
-- +++ processing catkin package: 'kinect2_bridge'
-- ==> add_subdirectory(iai_kinect2/kinect2_bridge)
[... 81 lines cut ...]
Scanning dependencies of target kinect2_calibration
Scanning dependencies of target kinect2_bridge
Scanning dependencies of target kinect2_viewer
Scanning dependencies of target kinect2_bridge_nodelet
[ 57%] Building CXX object iai_kinect2/kinect2_calibration/CMakeFiles/kinect2_calibration.dir/src/kinect2_calibration.cpp.o
[ 71%] [ 85%] Building CXX object iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o
Building CXX object iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/src/kinect2_bridge.cpp.o
[100%] Building CXX object iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/src/viewer.cpp.o
Linking CXX shared library /home/buttler/catkin_ws/devel/lib/libkinect2_bridge_nodelet.so
Linking CXX executable /home/buttler/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge
[100%] Built target kinect2_bridge_nodelet
Linking CXX executable /home/buttler/catkin_ws/devel/lib/kinect2_calibration/kinect2_calibration
[100%] Built target kinect2_bridge
[100%] Built target kinect2_calibration
Linking CXX executable /home/buttler/catkin_ws/devel/lib/kinect2_viewer/kinect2_viewer
[100%] Built target kinect2_viewer
¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤
The only issue which I don't think is a matter is this:
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
Which I think is fine. I still get the same error after recompiling the bridge.

shamoorti, 2015-10-21:
The strange thing here is how the bridge is recognizing the Opencl for depth and not for reg? Look at this:
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl]   1: Intel(R) Core(TM) i7-5557U CPU @ 3.10GHz (CPU)[Intel(R) Corporation]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] ReadData0x14 response
[Debug] [Freenect2DeviceImpl] 92 bytes of raw data
0x0000:  03 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    2e213355202e2e2e2e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 07 01 05 47 4d 46 31 36 35 2e 31    31332e2e2e2e2e2e474d463136352e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 025613444747
device firmware: 4.3.3913.0.7
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Info] [OpenCLDepthPacketProcessor] avg. time: 9.03053ms -> ~110.735Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.5708ms -> ~64.2227Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 9.16814ms -> ~109.073Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.8967ms -> ~62.9061Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 9.43532ms -> ~105.985Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.1344ms -> ~61.9794Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 9.58751ms -> ~104.302Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.3888ms -> ~61.0173Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 9.55902ms -> ~104.613Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.7984ms -> ~59.5296Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 9.76517ms -> ~102.405Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.0502ms -> ~58.6505Hz
¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤
Is something wrong with TurboJpegRgbPacketProcessor linking to bridge?

wiedemeyer, 2015-10-21:
The code for the depth processor is in libfreenect2 and the registration code is in the kinect2_registration package. Both were originally written by me, but maybe there are some changes in the code of libfreenect2 that explains why they are behaving differently.
The biggest difference is the include of the opencl headers. Maybe it helps if you replace [line 21 -25 in kinect2_registration/src/depth_registration_opencl.cpp](https://github.com/code-iai/iai_kinect2/blob/master/kinect2_registration/src/depth_registration_opencl.cpp#L21-L25) with:
```
#ifdef __APPLE__
#include <OpenCL/cl.hpp>
#else
#define CL_USE_DEPRECATED_OPENCL_1_1_APIS
#include <CL/cl.h>
#undef CL_VERSION_1_2
#include <CL/cl.hpp>
#endif
```

shamoorti, 2015-10-21:
Unfortunately it didn't help. I added the code and recompiled it again but no success.

shamoorti, 2015-10-21:
Fortunately we debugged the error. As you can read in our opencl test we get this:
./cl-demo 1000000 10
Choose platform:
[0] Intel
[1] Intel
[2] Intel(R) Corporation
Enter choice: 2
Choose device:
[0] Intel(R) Core(TM) i7-5557U CPU @ 3.10GHz
Hence the platfrom number is 2.Go in the kinect2_registration/src/depth_registration_opencl.cpp line 83 and change the code like this:

const cl::Platform &platform = platforms[i]; to const cl::Platform &platform = platforms[Your platform number] in my case 2.
Thanks a lot to wiedemeyer.

wiedemeyer, 2015-10-21:
That will choose the CPU, not the GPU. But at least that works for you.
But I also changed the code to check for errors like in libfreenect2, it's in the devel branch if you want to test it.

shamoorti, 2015-10-21:
Ok then so why Libfreenect2 can run on GPU and bridge can not?

shamoorti, 2015-10-21:
See, when I run the bridge GPU is chosen:
Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl]   1: Intel(R) Core(TM) i7-5557U CPU @ 3.10GHz (CPU)[Intel(R) Corporation]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]

wiedemeyer, 2015-10-21:
That's what I am trying to find out. Did you checked out the devel branch?

shamoorti, 2015-10-21:
In devel I have lib and share and and some shell and bash. In the share folder there are kinect_bridge
kinect_view, kinect_calibration and kinect_ registration which all of them have make files. Which one should I make to test?

wiedemeyer, 2015-10-21:
`git checkout devel` in the iai_kinect2 package

shamoorti, 2015-10-21:
I get this :+1: 
buttler@buttler-desktop:~/catkin_ws/src/iai_kinect2$ git checkout devel
M	kinect2_registration/src/depth_registration_opencl.cpp
Branch devel set up to track remote branch devel from origin.
Switched to a new branch 'devel'

wiedemeyer, 2015-10-21:
Maybe you want to get rid of your changes before testing: `git stash`. And to get them back `git stash pop`

shamoorti, 2015-10-21:
I get this now:
Branch devel set up to track remote branch devel from origin.
Switched to a new branch 'devel'

wiedemeyer, 2015-10-22:
Did it changed something? Is it working now?

shamoorti, 2015-10-22:
No, still the same error!!!!

shamoorti, 2015-10-23:
Any luck finding out the problem?

wiedemeyer, 2015-10-23:
I am not sure. But I pushed some changes to the devel branch. Could you pull the recent changes, compile them, run it and past the output here?

shamoorti, 2015-10-23:
Do I only need to run 'git checkout devel' or recompile everything? I mean when I see iai_kinect2
 page the last changes is for 11 days ago. I'm sorry but I don't know how it works. :)

wiedemeyer, 2015-10-23:
```
cd <catkin_ws>/src/iai_kinect2
git checkout devel
git stash
git pull
cd <catkin_ws>
rm -rf devel build
catkin_make
```

shamoorti, 2015-10-23:
Perfect Wiedmeyer, it works like a charm. Here is the log:
[ INFO] Connected to master at [localhost:11311]
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/buttler/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl]   1: Intel(R) Core(TM) i7-5557U CPU @ 3.10GHz (CPU)[Intel(R) Corporation]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 025613444747
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 025613444747 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 025613444747
[ INFO] [Kinect2Bridge::initDevice] device firmware: 4.3.3913.0.7
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...
[Info] [Freenect2DeviceImpl] stopped
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [DepthRegistrationOpenCL::init] devices:
[ INFO] [DepthRegistrationOpenCL::init]   0: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [DepthRegistrationOpenCL::init]   1: Intel(R) Core(TM) i7-5557U CPU @ 3.10GHz (CPU)[Intel(R) Corporation]
[ INFO] [DepthRegistrationOpenCL::init] selected device: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [DepthRegistrationOpenCL::init] devices:
[ INFO] [DepthRegistrationOpenCL::init]   0: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [DepthRegistrationOpenCL::init]   1: Intel(R) Core(TM) i7-5557U CPU @ 3.10GHz (CPU)[Intel(R) Corporation]
[ INFO] [DepthRegistrationOpenCL::init] selected device: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [Kinect2Bridge::main] waiting for clients to connect
[ INFO] [Kinect2Bridge::callbackStatus] client connected. starting device...
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Info] [DepthPacketStreamParser] -1 of 30 packets were lost
[ INFO] [Kinect2Bridge::main] depth processing: ~124.19Hz (8.05218ms) publishing rate: ~26.9223Hz
[ INFO] [Kinect2Bridge::main] color processing: ~61.2779Hz (16.3191ms) publishin rate: 
¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤4
Amazing ~124.19Hz depth Hertz. Thanks again for your help. :)

wiedemeyer, 2015-10-23:
your welcome.

closed 2015-10-23 by wiedemeyer