code-iai/iai_kinect2 #171: Why is the color calibration be in the gray scale?

shamoorti, 2015-10-27:
Hi, I am calibrating Kinect and and when I run this line rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 record color, I get the gray scale image for calibration. Should it be like this?
Thanks in advance.

wiedemeyer, 2015-10-27:
Because calibration does not need color information.
What should a RGB-D topic be? A four channel image with R,G,B,D where R,G,B are 8-bit and D is 16 bit? I think a RGB image and a separate depth image is the better way.

shamoorti, 2015-10-27:
Ok I understood the calibration method. Yeah exacly I meant 8 bit R G B and
16 bit death. It makes it really easy for the matlab user to segment the
rigion of interest with clouding this topic. Otherwise you have to use
cartesian cordination and make the proper matrix for the camera intrinsics
For v1 matlab has this toolbox. Or maybe it's true to say it like this. For
people who use ROS to Matlab for kinect v2 bridge, a cloud topic would be
really helpfull for furture image proccessing. Look at it this way. Now I
have to subscribe to time the same topics to extract the color and depth to
make a cloud which will  drain quiet a lot my CPU. But with one cloud topic
I get both at once. What is your idea Thiemo?

wiedemeyer, 2015-10-27:
For a point cloud, you could simply launch the kinect2.launch. That will give a point cloud topic.

shamoorti, 2015-10-27:
Oh really good thanks. I did not know that.

shamoorti, 2015-10-27:
Oh really good thanks. I did not know that. But what kind of command is that? As a rosrun? Like:
rosrun kinect2.launch,   after running roscore?

wiedemeyer, 2015-10-28:
roslaunch kinect2_bridge kinect2.launch

shamoorti, 2015-10-28:
I get tis error:
buttler@buttler-desktop:~/catkin_ws$ roslaunch kinect2_bridge kinect2.launch
[kinect2.launch] is neither a launch file in package [kinect2_bridge] nor is [kinect2_bridge] a launch file name
The traceback for the exception was written to the log file

wiedemeyer, 2015-10-28:
Maybe you should get familiar with ROS first. Take a look at the getting started and tutorials on http://wiki.ros.org/.

shamoorti, 2015-10-28:
Ok, sorry I am not the responsible for the ROS environment. I will ask the ROS guy to fix this and will come back to you soon. Thanks again Thiemo.

vishu2287, 2015-10-28:
@shamoorti 

ros launches only if launch file exist. In kinect2_bridge/launch if you are able to find kinect2.launch file, then ros executes. it seems you dont have kinect2.launch file.

wiedemeyer, 2015-10-28:
I would guess the workspace is not sourced...

vishu2287, 2015-10-28:
@shamoorti  are you able to launch kinect2_bridge.launch?
roslaunch kinect2_bridge kienct2_bridge.launch

shamoorti, 2015-10-28:
Actually the command is roslaunch kinect2_bridge kinect2_bridge.launch. I think it was a typo from Thiemo.  roslaunch kinect2_bridge kinect2.launch, there is now launch file as kinect2.launch. Thank you guys now the Points topic is working. :)

shamoorti, 2015-10-28:
Wow, I can't even subscribe this in matlab. I get this error:
test = rossubscriber('/kinect2/sd/points');
Oct 28, 2015 11:31:06 AM org.jboss.netty.channel.DefaultChannelPipeline
WARNING: An exception was thrown by a user handler while handling an exception event ([id: 0x09edcacb, /127.0.0.1:52842 :> buttler-desktop/127.0.1.1:45131] EXCEPTION: java.lang.OutOfMemoryError: Java heap space)
org.ros.exception.RosRuntimeException: java.lang.OutOfMemoryError: Java heap space
	at org.ros.internal.transport.ConnectionTrackingHandler.exceptionCaught(ConnectionTrackingHandler.java:94)
	at org.jboss.netty.channel.Channels.fireExceptionCaught(Channels.java:533)
	at org.jboss.netty.channel.AbstractChannelSink.exceptionCaught(AbstractChannelSink.java:49)
	at org.jboss.netty.channel.Channels.fireMessageReceived(Channels.java:296)
	at org.jboss.netty.handler.codec.frame.FrameDecoder.unfoldAndFireMessageReceived(FrameDecoder.java:458)
	at org.jboss.netty.handler.codec.frame.FrameDecoder.callDecode(FrameDecoder.java:439)
	at org.jboss.netty.handler.codec.frame.FrameDecoder.messageReceived(FrameDecoder.java:311)
	at org.jboss.netty.channel.Channels.fireMessageReceived(Channels.java:268)
	at org.jboss.netty.channel.Channels.fireMessageReceived(Channels.java:255)
	at org.jboss.netty.channel.socket.nio.NioWorker.read(NioWorker.java:91)
	at org.jboss.netty.channel.socket.nio.AbstractNioWorker.processSelectedKeys(AbstractNioWorker.java:373)
	at org.jboss.netty.channel.socket.nio.AbstractNioWorker.run(AbstractNioWorker.java:247)
	at org.jboss.netty.channel.socket.nio.NioWorker.run(NioWorker.java:35)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
Caused by: java.lang.OutOfMemoryError: Java heap space
	at org.jboss.netty.buffer.LittleEndianHeapChannelBuffer.copy(LittleEndianHeapChannelBuffer.java:125)
	at org.jboss.netty.buffer.AbstractChannelBuffer.copy(AbstractChannelBuffer.java:522)
	at org.ros.internal.transport.queue.MessageReceiver.messageReceived(MessageReceiver.java:62)
	... 13 more

>> 
Which it seems to be the heap memory overflow!!!!

wiedemeyer, 2015-10-29:
Maybe your heap size is too small? Probably is fixable, [here](http://lmgtfy.com/?q=java.lang.OutOfMemoryError%3A+Java+heap+space).

shamoorti, 2015-10-29:
Thanks a lot Thiemo for your good answers. I have another question. Non of the topics they have the actual size of kinect depth which is 512*424 and I know why you have done that since you wanted to have the depth be synced with the RGB image. But what if I want to just subscribe hd depth and plot it. I did that and I'm not getting the actual depth of course. Any Idea how to fix that? Should I make my own topic for actual depth size?

wiedemeyer, 2015-10-29:
I don't really understand the question. The topics in `sd` will have the resolution of the ir sensor (512*424). What do you mean with you are not getting the actual depth?

shamoorti, 2015-10-29:
Ok sorry for making you confused, look at my cloud. You think this is the calibration problem?
[image]

GaiTech-Robotics, 2015-10-29:
@shamoorti 
are you using it in addition to any model? if so you need to specify position and orientation of sensor in your model

wiedemeyer, 2015-10-29:
Looks like a normal point cloud. The far away points are probably just noise. Just set max_depth to something smaller.

shamoorti, 2015-10-29:
It works like a charm Thiemo :). Thanks a lot.

shamoorti, 2015-10-29:
@GaiTech-Robotics 
No it's stand alone software for image processing.

closed 2015-10-31 by wiedemeyer