code-iai/iai_kinect2 #177: Understand frame rates of kinect2_bridge

MarcoFierimonte, 2015-11-03:
Hi, 

I don't understand how works the frame rate of ROS kinect2_bridge topics. Is it possible to set this rate? With the param _fps_limit I set the frame acquisition of Kinect2 but the ROS publishers have a rate of more or less 30 Hz.
I'd like to reduce it to give less work to my CPU/memory.

Thanks :)

wiedemeyer, 2015-11-03:
Hi,
the `fps_limit` will only limit the frames processed and published by the kinect2_bride node. It will not limit the frames received by libfreenect from the sensor. There is no feature for limiting those. For example with a `fps_limit` of 3, only every 10th frame received from libfreenect2 will get processed and published.

MarcoFierimonte, 2015-11-03:
Thank you for your reply Thiemo,

so if _fps_limit limit the frames processed and published by the kinect2_bride node, why with _fps_limit:=1 I have kinect2 topics with publishing rate of about 30 Hz?
I did this:
1) start the kinect2_bride node.
2) rostopi echo /kinect2/sd/points
3) rostopic hz -w 100 /kinect2/sd/points

and I have:
[image]

[image]

Is it normal? why?

Thanks.
Marco

wiedemeyer, 2015-11-03:
if you are using the launch file you don't need the leading `_` for the parameter. With rosrun `_fps_limit:=1`, with roslaunch `fps_limit:=1`.

MarcoFierimonte, 2015-11-03:
Ops.....you are right !!! Sorry for this stupid problem. Anyway I understand better how kinect2_bridge works....thanks !!! 

Marco

closed 2015-11-03 by wiedemeyer