code-iai/iai_kinect2 #178: How to make depth stream video

shakith, 2015-11-04:
I'm using the command rosrun image_view video_recorder image:=/kinect2/sd/image_depth _encoding:=16UC1 to make depth stream video. However, it crashes with the following error.

Starting to record MJPG video at [512 x 424]@15fps. Press Ctrl+C to stop recording. OpenCV Error: Assertion failed (image->depth == 8) in writeFrame, file /build/buildd/opencv-2.4.8+dfsg1/modules/highgui/src/cap_ffmpeg.cpp, line 238 terminate called after throwing an instance of 'cv::Exception' what(): /build/buildd/opencv-2.4.8+dfsg1/modules/highgui/src/cap_ffmpeg.cpp:238: error: (-215) image->depth == 8 in function writeFrame

Aborted (core dumped)

It doesnt work either with _encoding:=32FC1, 32SC1. If _encoding:=mono16 is used, it says "unable to convert 16UC1 to mono16"

Is it problem with converting from ROS encodingto OpenCV encoding? Or is the video-writer only limited to color videos? This would be strange.

Thanks.

wiedemeyer, 2015-11-04:
From the error message I would say that it is only supports 8 Bit channel depth. So RGB or grayscale. I think limiting a video codec to color images is not really strange, it is kind of strange to use them for something else than that. But maybe a different codec might work.
http://wiki.ros.org/image_view#image_view.2BAC8-diamondback.video_recorder

shakith, 2015-11-09:
Okay. Thanks

closed 2015-11-09 by shakith