code-iai/iai_kinect2 #179: Calibration process broken?

rahulraw, 2015-11-09:
I am trying to calibrate my Kinect2 using 
```
rosrun kinect2_calibration kinect2_calibration chess7x8x0.108 record color 
```
I don't understand the process after this. If I press space on the pop up color window, should it save a file? At the current moment it does not do anything.

How does one actually calibrate a kinect2 using this package?

wiedemeyer, 2015-11-10:
It only saves a file if the calibration pattern was detected. If it got detected the image shows some colored drawings on the board. For the instructions look here:
https://github.com/code-iai/iai_kinect2/blob/master/kinect2_calibration/README.md

rahulraw, 2015-11-10:
I see it was not detecting the calibration sheet. Thank you

closed 2015-11-10 by rahulraw