code-iai/iai_kinect2 #180: Should I rotate the cloud after subscribing the ROS topic?

MarcoFierimonte, 2015-11-10:
Hi Thiemo,

I'd like to create a personal ROS TF  tree but I don't understand very well how works coordinates systems between iai_kinect2 and PCL.
For istance:
1) if I subscribe to "kinect2/sd/points" topic, my ROS callback will receive a  sensor_msgs::PointCloud2 message. 
2) After that I use pcl::fromROSMsg() to convert the message in a pcl::PointCloud<pcl::PointXYZRGB> object.
3) I do some processing on the cloud
4) Finally I convert the cloud in a sensor_msgs::PointCloud2 using pcl::toROSMsg() and publish it on a ROS topic.

Why after point 2) if I use a pcl viewer to show the cloud is inverted? Have I to rotate the cloud using something like pcl::transformPointCloud () ?	

Thanks in advance :)

wiedemeyer, 2015-11-13:
What do you mean with inverted? How does it look in rviz?

MarcoFierimonte, 2015-11-13:
In rviz I look the same of the pcl viewer, the point cloud is upside down. Insteed if I use your iai_kinect2 viewer is correct. Also with the ./Protonect viewer is correct. Maybe it's because the coordinates systems between the kinect V2 and ROS are different? 

Thanks for your help.

wiedemeyer, 2015-11-16:
Hi,
I looked at rviz and it looked totally fine, nothing is inverted. If you mean, why the point cloud in rviz is projected along the `z` axis, in other words "projected into the sky". This is because no transform for given for `kinect2_link`. In rviz `z` is pointing up, therefore the point cloud is projected that way.

MarcoFierimonte, 2015-11-17:
Hi,

I don't understand if your kinect2_viewer do some rotation of the point cloud. Because using it I have: 
[image]

instead using my simple viewer I obtain the point cloud upside down:
[image]

finally with RVIZ: 
[image]

Is this result correct? Should I rotate the cloud obtained from "/kinect2/sd/points"?

wiedemeyer, 2015-11-17:
Looks just right. z is the depth, x is from left to right and y is from top to bottom. That is like the coordinates are defined for images. The problem is, that the default camera rotation is the pcl viewer and Rviz is different. You just have to change the camera rotation by rotation it around z by 180 degree. That is what I did in the kinect2_viewer.

MarcoFierimonte, 2015-11-18:
Thank you Thiemo, now I understand and works :)

closed 2015-11-18 by MarcoFierimonte

dandan19793, 2016-08-18:
hi,

how do i rotate the axes of the camera? also, when i look at the axes, i think they are not aligned
 correctly as in marco's case, see pic attached (i am somehow lying on my left side).
[image]

thanks