code-iai/iai_kinect2 #181: [ERROR] [DepthCalibration::compareDists] no distance sample data!

sshonig, 2015-11-12:
Hi,

Whenever we try to calibrate the depth measurement, we get the following error:
[ERROR] [DepthCalibration::compareDists] no distance sample data!

What does it mean and how can we fix it?

Thanks,
S.

wiedemeyer, 2015-11-13:
Hi,
it means that no data is present to do a calibration. You have to record data first, this is done by recording ir and sync.

hyderjake, 2015-11-17:
I have done all of the calibration steps and I still receive this error, can you help me understand why

sshonig, 2015-11-22:
Thanks. The problem wasn't that we didn't have data (we had recorded data prior to receiving this error), but that we did not record enough data points. Once we added an additional ~100 images at various distances the problem resolved.

closed 2015-11-23 by wiedemeyer

dragonlong, 2016-05-27:
What does it mean by adding an additional ~100 images at various distances??
I meet the same problem. Before that error occured, I saved 100 images for color,ir, and sync respectively but at the same distances. So should I change distance while saving the 100 images?

Solrac3589, 2017-08-23:
Hello.

I have the same problem explained here. 

I did around 170 images of colour, ir and sync and the proble is not solved.

I was wondering is it is the way i use to do the recording.

i have the calibration pattern in a table and while i go approaching the kinect, i go doing photos like hell.

I do this procedure 3 times, one for each kind of record.

PoRouS22, 2018-04-18:
Hey, I had the same problem today, but I found a solution for it. My problem was that I was swapping the number of chess squares, when I was recording depth. So instead of "rosrun kinect2_calibration kinect2_calibration chess9x10x0.0801 record depth" I typed "rosrun kinect2_calibration kinect2_calibration chess10x9x0.0801 calibrate color". That's why it didn't find any data when calibrating the recorded depth. To solve this problem I swapped the numbers for calibrating, then it worked.

mo5h, 2020-09-05:
I found that you've got to have the calibration board tilted, which I suppose makes sense since if it's upright there's no difference between the true depth data between the squares