code-iai/iai_kinect2 #182: clEnqueueWriteBuffer(-5) reproducible

karlpauwels, 2015-11-13:
I managed to reproduce the opencl error mentioned in #163 and #90 by making a small change to the extrinsic calibration file:
https://github.com/code-iai/iai_kinect2/blob/master/kinect2_bridge/data/196605135147/calib_pose.yaml#L16
If I change the z-translation from 8.8806735554907584e-04 to -8.8806735554907584e-01 and launch
`roslaunch kinect2_bridge kinect2_bridge.launch` the following errors occur when I subscribe to `/kinect2/qhd/image_depth_rect`:

```
[ERROR] [1447409963.943430111]: [DepthRegistrationOpenCL::registerDepth] [depth_registration_opencl.cpp](305) enqueueWriteBuffer failed: -5
[ERROR] [1447409963.979667709]: [DepthRegistrationOpenCL::registerDepth] [depth_registration_opencl.cpp](305) enqueueWriteBuffer failed: -5
[ERROR] [1447409964.008715427]: [DepthRegistrationOpenCL::registerDepth] [depth_registration_opencl.cpp](305) enqueueWriteBuffer failed: -5
[ERROR] [1447409964.046510702]: [DepthRegistrationOpenCL::registerDepth] [depth_registration_opencl.cpp](305) enqueueWriteBuffer failed: -5
```

This error also occurs with more reasonable calibration files.
There are no errors when I use the cpu for registration
`roslaunch kinect2_bridge kinect2_bridge.launch depth_method:=cpu reg_method:=cpu` or when I use opencl with the original calibration file.

I'm using the latest version of libfreenect2 and iai_kinect2.

wiedemeyer, 2015-11-13:
Hi Karl,
I was able to reproduce the error on my machine, thanks! I found the source of the problem and pushed a fix for it. It solved it on my machine, hopefully it fixes it for you too.

karlpauwels, 2015-11-13:
That solved it for me as well. Thanks so much for the quick fix!

closed 2015-11-13 by karlpauwels