code-iai/iai_kinect2 #183: Recording video

hyderjake, 2015-11-17:
Is there a way to record video streams with this software?

jeppewalther, 2015-11-17:
Take a look at ROS Bag http://wiki.ros.org/rosbag

tlind, 2015-11-22:
I also have a few questions related to recording data into rosbags. Maybe somebody with more experience can clear this up:

1. Which topics should I record in order to be able to reconstruct point clouds and registered/rectified images from them? If possible, I would like to record compressed data to reduce disk usage and required write bandwidth. Are the following topics sufficient?

        /kinect2/sd/image_depth/compressed
        /kinect2/sd/camera_info
        /kinect2/hd/image_color/compressed
        /kinect2/hd/camera_info
        /tf

2. Which nodelets do I have to run during playback to regenerate point clouds and registered/rectified images? I see in the `kinect2_bridge.launch` file that the point clouds are generated by `point_cloud_xyzrgb` nodelets, but what about the registration? As it seems to be done by the   `kinect2_bridge_nodelet`, is it even possible to do this in an offline fashion, from a bagfile?
3. Would this then allow me to use a different calibration at the time of playback, if I am unhappy with the initial calibration that was used during recording?
4. Why is there no compressedDepth image transport, as with the old Kinect / Asus sensors? 
5. Which compression method is faster, png or tiff? Is there a difference in quality?

Thanks in advance!

wiedemeyer, 2015-11-23:
Hi,
1. For the point cloud you will need `camera_info`, `tf` and `image_depth_rect`, and if you want a colored cloud also `image_color_rect`. It is no problem to store the compressed topics. That very helpful to keep the bags small. Maybe you need to republish the uncompressed images with this http://wiki.ros.org/image_transport#republish
2. You need one of the `point_cloud_xyzrgb` nodelets. All images in one of the groups `sd`, `qhd` and `hd` are registered to each other. No, offline fashion is not possible, but you can record all the topics you later need.
3. No not possible. (Well you could record the not rectified color and depth image and register them yourself on playback.)
4. Because it is only needed to compress 32 Bit float depth images. Just makes it more complicated.
5. tiff has a better speed / compression ratio. png could lead to higher compression but much higher CPU load.

closed 2015-12-09 by wiedemeyer

dandan19793, 2016-05-23:
hi,

i wanted to record a video (i need both color and depth) and later use the viewer for debugging.
i launch: roslaunch kinect2_bridge kinect2_bridge.launch in the one terminal and 
rosbag record /kinect2/hd/image_color_rect /kinect2/hd/image_depth_rect /kinect2/hd/camera_info /kinect2/hd/tf 
 in another and then try to play the bag file with rosrun kinect2_viewer kinect2_viewer kinect2 hd both
i can see the color image but not the depth. the fps is around 25 and not 30 as when i am not recording. its also quite difficult for my pc to do that although its an i7 with 16GB Ram and nvidia quadro k2000.

when i recorded compressed images with: rosbag record /kinect2/hd/image_color_rect/compressed /kinect2/hd/image_depth_rect/compressed /kinect2/hd/camera_info /kinect2/hd/tf

there was not even a window opening to display the images. 

what am i doing wrong here? does my approach make any sense?

thanks

wiedemeyer, 2016-05-24:
First of all, is the depth image showing up if you run the viewer with the kinect2_bridge directly?

Maybe you computer is busy writing all the data to the disk and therefore only processes 25 fps. Color images at 30 Hz are ~190 MByte/s and depth images are ~120 MBytes/s, so you would have to write data with 310 MByte/s to your disk. For recording it helps to use compressed images.

If you want to use the compressed images with the viewer you have to give it the parameter `compressed`. And there shouldn't be any topic called `/kinect2/hd/tf`.

dandan19793, 2016-05-25:
yes, it does. i just noticed that when running the bridge and viewer i get 30 fps (stated by the bridge in the shell) but when recording with rosbag record -a it drops down to approx 10 fps. i have a ssd with r/w of more than 500mb/s so i dont understand it. 

to make a long story short, which topics do i need to record in order to later get the color and the cloud with the image viewer (like i get from the bridge)? best will be if i get both in the color/hd frame (and not the ir/depth frame)

after reading the above i tried:
rosbag record 
/kinect2/sd/image_depth/compressed
/kinect2/sd/camera_info
/kinect2/hd/image_color/compressed
/kinect2/hd/camera_info
/tf
/kinect2/sd/image_depth_rect
/kinect2/hd/image_color_rect

and 
rosrun kinect2_viewer kinect2_viewer kinect2 sd both compressed
or
rosrun kinect2_viewer kinect2_viewer kinect2 hd both compressed

with no success. no window open to show any image.

vchu, 2016-11-21:
Quick comment in case anyone else stumbles across this like me and struggled to get rosbag working with `iai_kinect2`.

Wrote a very fast [tutorial](https://github.com/si-machines/data_logger_bag/wiki/Kinect2%20Playback) on how to record/playback.

dhiegomaga, 2019-02-06:
I was not able to get the pointcloud from the RGB and DEPTH images (using rosbag play).
Tried using `depth_image_proc` but it doesn't publish anything to the `/points` topic...

The RGBD images themselves look ok. Just can't use this nodelet!