code-iai/iai_kinect2 #184: New issue with Libfree latest release!

shamoorti, 2015-11-18:
Hey Thiemo,
Guys just updated Libfreenect2 and now I'm having a problem with making the bridge. The error while making is: 
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_calibration/src/kinect2_calibration.cpp: In member function ‘void Recorder::display()’:
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_calibration/src/kinect2_calibration.cpp:338:42: error: ‘CV_GRAY2BGR’ was not declared in this scope
           cv::cvtColor(color, colorDisp, CV_GRAY2BGR);
                                          ^
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_calibration/src/kinect2_calibration.cpp:345:40: error: ‘CV_GRAY2BGR’ was not declared in this scope
           cv::cvtColor(irGrey, irDisp, CV_GRAY2BGR);
                                        ^
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_calibration/src/kinect2_calibration.cpp: In member function ‘void CameraCalibration::calibrateExtrinsics()’:
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_calibration/src/kinect2_calibration.cpp:720:56: error: cannot convert ‘const cv::TermCriteria’ to ‘int’ for argument ‘13’ to ‘double cv::stereoCalibrate(cv::InputArrayOfArrays, cv::InputArrayOfArrays, cv::InputArrayOfArrays, cv::InputOutputArray, cv::InputOutputArray, cv::InputOutputArray, cv::InputOutputArray, cv::Size, cv::OutputArray, cv::OutputArray, cv::OutputArray, cv::OutputArray, int, cv::TermCriteria)’
                                 cv::CALIB_FIX_INTRINSIC);
                                                        ^
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_calibration/src/kinect2_calibration.cpp: In member function ‘void DepthCalibration::computeROI(const cv::Mat&, const std::vector<cv::Point_<float> >&, cv::Mat&, cv::Rect&) const’:
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_calibration/src/kinect2_calibration.cpp:1085:56: error: ‘CV_RGB’ was not declared in this scope
     cv::fillConvexPoly(mask, hull, CV_RGB(255, 255, 255));
                                                        ^
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘void Kinect2Bridge::initCompression(int32_t, int32_t, bool)’:
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:405:28: error: ‘CV_IMWRITE_JPEG_QUALITY’ was not declared in this scope
     compressionParams[0] = CV_IMWRITE_JPEG_QUALITY;
                            ^
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:407:28: error: ‘CV_IMWRITE_PNG_COMPRESSION’ was not declared in this scope
     compressionParams[2] = CV_IMWRITE_PNG_COMPRESSION;
                            ^
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:409:28: error: ‘CV_IMWRITE_PNG_STRATEGY’ was not declared in this scope
     compressionParams[4] = CV_IMWRITE_PNG_STRATEGY;
                            ^
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:410:28: error: ‘CV_IMWRITE_PNG_STRATEGY_RLE’ was not declared in this scope
     compressionParams[5] = CV_IMWRITE_PNG_STRATEGY_RLE;
                            ^
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘void Kinect2Bridge::initCompression(int32_t, int32_t, bool)’:
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:405:28: error: ‘CV_IMWRITE_JPEG_QUALITY’ was not declared in this scope
     compressionParams[0] = CV_IMWRITE_JPEG_QUALITY;
                            ^
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:407:28: error: ‘CV_IMWRITE_PNG_COMPRESSION’ was not declared in this scope
     compressionParams[2] = CV_IMWRITE_PNG_COMPRESSION;
                            ^
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:409:28: error: ‘CV_IMWRITE_PNG_STRATEGY’ was not declared in this scope
     compressionParams[4] = CV_IMWRITE_PNG_STRATEGY;
                            ^
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:410:28: error: ‘CV_IMWRITE_PNG_STRATEGY_RLE’ was not declared in this scope
     compressionParams[5] = CV_IMWRITE_PNG_STRATEGY_RLE;
                            ^
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘void Kinect2Bridge::processIrDepth(const cv::Mat&, const cv::Mat&, std::vector<cv::Mat>&, const std::vector<Kinect2Bridge::Status>&, libfreenect2::Frame*)’:
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:980:48: error: ‘CV_BGRA2BGR’ was not declared in this scope
       cv::cvtColor(tmp, images[COLOR_SD_RECT], CV_BGRA2BGR);
                                                ^
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘void Kinect2Bridge::processColor(const cv::Mat&, std::vector<cv::Mat>&, const std::vector<Kinect2Bridge::Status>&, libfreenect2::Frame*)’:
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1042:43: error: ‘CV_BGRA2BGR’ was not declared in this scope
       cv::cvtColor(tmp, images[COLOR_HD], CV_BGRA2BGR);
                                           ^
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1060:55: error: ‘CV_BGR2GRAY’ was not declared in this scope
       cv::cvtColor(images[COLOR_HD], images[MONO_HD], CV_BGR2GRAY);
                                                       ^
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1064:65: error: ‘CV_BGR2GRAY’ was not declared in this scope
       cv::cvtColor(images[COLOR_HD_RECT], images[MONO_HD_RECT], CV_BGR2GRAY);
                                                                 ^
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1068:57: error: ‘CV_BGR2GRAY’ was not declared in this scope
       cv::cvtColor(images[COLOR_QHD], images[MONO_QHD], CV_BGR2GRAY);
                                                         ^
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1072:67: error: ‘CV_BGR2GRAY’ was not declared in this scope
       cv::cvtColor(images[COLOR_QHD_RECT], images[MONO_QHD_RECT], CV_BGR2GRAY);
                                                                   ^
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘void Kinect2Bridge::processIrDepth(const cv::Mat&, const cv::Mat&, std::vector<cv::Mat>&, const std::vector<Kinect2Bridge::Status>&, libfreenect2::Frame*)’:
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:980:48: error: ‘CV_BGRA2BGR’ was not declared in this scope
       cv::cvtColor(tmp, images[COLOR_SD_RECT], CV_BGRA2BGR);
                                                ^
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘void Kinect2Bridge::processColor(const cv::Mat&, std::vector<cv::Mat>&, const std::vector<Kinect2Bridge::Status>&, libfreenect2::Frame*)’:
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1042:43: error: ‘CV_BGRA2BGR’ was not declared in this scope
       cv::cvtColor(tmp, images[COLOR_HD], CV_BGRA2BGR);
[... 19 lines cut ...]
make[1]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/all] Error 2
In file included from /usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:52:0,
                 from /usr/include/pcl-1.7/pcl/visualization/cloud_viewer.h:39,
                 from /home/mobin/catkin_ws/src/iai_kinect2/kinect2_viewer/src/viewer.cpp:32:
/usr/include/pcl-1.7/pcl/visualization/area_picking_event.h: In member function ‘bool pcl::visualization::AreaPickingEvent::getPointsIndices(std::vector<int>&) const’:
/usr/include/pcl-1.7/pcl/visualization/area_picking_event.h:64:30: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
           if (nb_points_ == -1)
                              ^
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_viewer/src/viewer.cpp: In member function ‘void Receiver::callback(sensor_msgs::Image_<std::allocator<void> >::ConstPtr, sensor_msgs::Image_<std::allocator<void> >::ConstPtr, sensor_msgs::CameraInfo_<std::allocator<void> >::ConstPtr, sensor_msgs::CameraInfo_<std::allocator<void> >::ConstPtr)’:
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_viewer/src/viewer.cpp:225:32: error: ‘CV_GRAY2BGR’ was not declared in this scope
       cv::cvtColor(tmp, color, CV_GRAY2BGR);
                                ^
[... cut ...]
/home/mobin/catkin_ws/src/iai_kinect2/kinect2_viewer/src/viewer.cpp:279:84: error: ‘CV_AA’ was not declared in this scope
         cv::putText(combined, oss.str(), pos, font, sizeText, colorText, lineText, CV_AA);
                                                                                    ^
make[2]: *** [iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/src/viewer.cpp.o] Error 1
make[1]: *** [iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
*************************************************
Can you check this out?

wiedemeyer, 2015-11-18:
I will check it. But it looks like something with OpenCV is wrong. Did you updated / changed your OpenCV installation recently? What version of OpenCV are you using?

shamoorti, 2015-11-18:
I have both 3 and 2. There would be a problem if I have 2 versions?

wiedemeyer, 2015-11-18:
I would guess it is finding and using version 3, which is not working with kinect2_bridge at the moment. So you have to make sure that it finds and uses OpenCV 2.4.x.

shamoorti, 2015-11-18:
Ok but how can I refrence it since it's reading from system file and I
can't include library like in IDE.

wiedemeyer, 2015-11-18:
Well, you could uninstall version 3, or install it somewhere that is not one of the default paths. Or change the find_package in the CMakeLists.txt from iai_kinect2 packages.

shamoorti, 2015-11-19:
Hey again thiemo, 
Finally I reset everything and now everything is working except calibration and viewer in the bridge. for both I get the error Segmentation fault (core dumped). But the bridge is working fine.
Also I wonder has Depth publishing rate decreased? I used to have 100~HZ with opencl but now seems half! Am I wrong? Look at this:
[ INFO] [DepthRegistrationOpenCL::init] selected device: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [Kinect2Bridge::main] waiting for clients to connect
[ INFO] [Kinect2Bridge::callbackStatus] client connected. starting device...
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
[Info] [DepthPacketStreamParser] -9 packets were lost
[Info] [DepthPacketStreamParser] 8 packets were lost
[ INFO] [Kinect2Bridge::main] depth processing: ~29.4169ms (~33.9941Hz) publishing rate: ~4.65806Hz
[ INFO] [Kinect2Bridge::main] color processing: ~17.3419ms (~57.6638Hz) publishing rate: ~3.99263Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 10.0158ms -> ~99.8421Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.7993ms -> ~59.5263Hz
¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤
Also I installed latest version of Libfreenect and the image is reversed horizontally and not calibrated anymore:
[image]

wiedemeyer, 2015-11-20:
`[Info] [OpenCLDepthPacketProcessor] avg. time: 10.0158ms -> ~99.8421Hz` it seems to run with 100hz.

shamoorti, 2015-11-23:
Ok I formated Ubuntu and reinstalled everything. I am experiencing this problem now ROS can't subscribe the depth field anymore. Sometimes it does and sometimes not. But most of the times not. I use this code in Matlab which before the latest update it was working ok:
CLoudIm = rossubscriber('/kinect2/hd/image_depth_rect');
CLoud= CLoudIm.LatestMessage;
CLoud = receive(CLoudIm);
CLoud = CLoud.readImage;
But now my readimage is always filled with zeros. !!! By the way color map frame works fine. Also I ahve not installed opencv or pcl individually but both Calibration and Viwer are working fine .
Thanks in advance :)

wiedemeyer, 2015-11-23:
What is the output of kinect2_bridge? Sounds like so USB issue. Also check protonect.

shamoorti, 2015-11-23:
mdh-butler@mdhbutler-desktop:~/catkin_ws$ /home/mdh-butler/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge _depth_method:=opengl _fps_limit:=5
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: 5
       calib_path: /home/mdh-butler/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 025613444747
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 025613444747 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 025613444747
[ INFO] [Kinect2Bridge::initDevice] device firmware: 4.3.3913.0.7
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...
[Info] [Freenect2DeviceImpl] stopped
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [DepthRegistrationOpenCL::init] devices:
[ INFO] [DepthRegistrationOpenCL::init]   0: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [DepthRegistrationOpenCL::init] selected device: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [DepthRegistrationOpenCL::init] devices:
[ INFO] [DepthRegistrationOpenCL::init]   0: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [DepthRegistrationOpenCL::init] selected device: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [Kinect2Bridge::main] waiting for clients to connect
[ INFO] [Kinect2Bridge::callbackStatus] client connected. starting device...
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
[Info] [DepthPacketStreamParser] -10 packets were lost
[Info] [DepthPacketStreamParser] 8 packets were lost
[ INFO] [Kinect2Bridge::main] depth processing: ~35.6281ms (~28.0677Hz) publishing rate: ~4.66335Hz
[ INFO] [Kinect2Bridge::main] color processing: ~27.8368ms (~35.9236Hz) publishing rate: ~3.99716Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 10.0436ms -> ~99.5662Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 18.6737ms -> ~53.5512Hz
[ INFO] [Kinect2Bridge::main] depth processing: ~32.2767ms (~30.9821Hz) publishing rate: ~4.98731Hz
[ INFO] [Kinect2Bridge::main] color processing: ~24.5503ms (~40.7327Hz) publishing rate: ~4.98731Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 9.6475ms -> ~103.654Hz
^C[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...
[... 34 lines cut ...]
[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 025613444747
device firmware: 4.3.3913.0.7
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] not all subsequences received 896
[Info] [OpenCLDepthPacketProcessor] avg. time: 9.00638ms -> ~111.032Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.1272ms -> ~58.3868Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 9.16253ms -> ~109.14Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.9865ms -> ~58.8703Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 8.8163ms -> ~113.426Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.017ms -> ~58.7648Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 9.01111ms -> ~110.974Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.8046ms -> ~59.5075Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 9.02249ms -> ~110.834Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.9596ms -> ~58.9638Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 9.1673ms -> ~109.083Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.8553ms -> ~59.3284Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 9.05797ms -> ~110.4Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.9534ms -> ~58.9854Hz

wiedemeyer, 2015-11-23:
I think it OpenGL. Maybe it does not work with your Intel GPU. For protonect you used OpenCL, but for kinect2_bridge you used OpenGL. Try to use OpenCL instead `_depth_method:=opencl`. Does `protonect gl` work?

shamoorti, 2015-11-23:
Protonect works with gl ok. This is the bridge with depth CL:
mdh-butler@mdhbutler-desktop:~/catkin_ws$ /home/mdh-butler/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge _depth_method:=opencl _fps_limit:=5
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: 5
       calib_path: /home/mdh-butler/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Warning] [OpenCLDepthPacketProcessorImpl] Your ocl-icd has deadlock bugs. Update to 2.2.4+ is recommended.
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 025613444747
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 025613444747 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 025613444747
[ INFO] [Kinect2Bridge::initDevice] device firmware: 4.3.3913.0.7
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...
[Info] [Freenect2DeviceImpl] stopped
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [DepthRegistrationOpenCL::init] devices:
[ INFO] [DepthRegistrationOpenCL::init]   0: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [DepthRegistrationOpenCL::init] selected device: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [DepthRegistrationOpenCL::init] devices:
[ INFO] [DepthRegistrationOpenCL::init]   0: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [DepthRegistrationOpenCL::init] selected device: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [Kinect2Bridge::main] waiting for clients to connect
[ INFO] [Kinect2Bridge::callbackStatus] client connected. starting device...
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
[Info] [DepthPacketStreamParser] -9 packets were lost
[Info] [DepthPacketStreamParser] 8 packets were lost
[ INFO] [Kinect2Bridge::main] depth processing: ~31.398ms (~31.8492Hz) publishing rate: ~4.6665Hz
[ INFO] [Kinect2Bridge::main] color processing: ~16.8482ms (~59.3534Hz) publishing rate: ~3.99986Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 11.3037ms -> ~88.4669Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.456ms -> ~60.7679Hz
[ INFO] [Kinect2Bridge::main] depth processing: ~32.2361ms (~31.0211Hz) publishing rate: ~4.98545Hz
[ INFO] [Kinect2Bridge::main] color processing: ~19.9711ms (~50.0724Hz) publishing rate: ~4.98545Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 11.3866ms -> ~87.8227Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.726ms -> ~63.5889Hz
^C[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing

wiedemeyer, 2015-11-23:
are the images still all zeros?

shamoorti, 2015-11-23:
The depth is most of the times zero. I just subscribe and receive the depth and do not do any processing on it. Still the image is sometimes zero.

wiedemeyer, 2015-11-23:
How does it look, if you use the kinect2_viewer, for example `rosrun kinect2_viewer kinect2_viewer sd image`

shamoorti, 2015-11-23:
[image]

wiedemeyer, 2015-11-23:
Looks like the depth image is not zero.

shamoorti, 2015-11-23:
Yes exactly. But as I said it sometimes happens. Randomly or it starts to not subscribe at all.

shamoorti, 2015-11-24:
Look at this Thiemo, Although I'm putting 30 fps rate limit but the publishing rate is <10 HZ for both color and depth. Is it normal?
mdh-butler@mdhbutler-desktop:~$ /home/mdh-butler/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge _depth_method:=opencl _fps_limit:=30
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: 30
       calib_path: /home/mdh-butler/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Warning] [OpenCLDepthPacketProcessorImpl] Your ocl-icd has deadlock bugs. Update to 2.2.4+ is recommended.
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 025613444747
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 025613444747 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 025613444747
[ INFO] [Kinect2Bridge::initDevice] device firmware: 4.3.3913.0.7
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...
[Info] [Freenect2DeviceImpl] stopped
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [DepthRegistrationOpenCL::init] devices:
[ INFO] [DepthRegistrationOpenCL::init]   0: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [DepthRegistrationOpenCL::init] selected device: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [DepthRegistrationOpenCL::init] devices:
[ INFO] [DepthRegistrationOpenCL::init]   0: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [DepthRegistrationOpenCL::init] selected device: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [Kinect2Bridge::main] waiting for clients to connect
[ INFO] [Kinect2Bridge::callbackStatus] client connected. starting device...
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
[Info] [DepthPacketStreamParser] -10 packets were lost
[Info] [DepthPacketStreamParser] 7 packets were lost
[Info] [DepthPacketStreamParser] 2 packets were lost
[ INFO] [Kinect2Bridge::main] depth processing: ~86.1472ms (~11.608Hz) publishing rate: ~13.2256Hz
[ INFO] [Kinect2Bridge::main] color processing: ~175.812ms (~5.68789Hz) publishing rate: ~7.60474Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 12.2202ms -> ~81.8315Hz
[ INFO] [Kinect2Bridge::main] depth processing: ~194.77ms (~5.13427Hz) publishing rate: ~9.65219Hz
[ INFO] [Kinect2Bridge::main] color processing: ~225.878ms (~4.42716Hz) publishing rate: ~8.98653Hz
[Info] [DepthPacketStreamParser] 8 packets were lost
[Info] [TurboJpegRgbPacketProcessor] avg. time: 35.9044ms -> ~27.8518Hz
[Info] [DepthPacketStreamParser] 2 packets were lost
[Info] [OpenCLDepthPacketProcessor] avg. time: 16.7486ms -> ~59.7064Hz
[Info] [DepthPacketStreamParser] 6 packets were lost
[ INFO] [Kinect2Bridge::main] depth processing: ~191.856ms (~5.21224Hz) publishing rate: ~7.6518Hz
[ INFO] [Kinect2Bridge::main] color processing: ~337.279ms (~2.96491Hz) publishing rate: ~7.31911Hz
[Info] [DepthPacketStreamParser] 3 packets were lost
[Info] [DepthPacketStreamParser] 4 packets were lost
[Info] [DepthPacketStreamParser] 4 packets were lost
[Info] [OpenCLDepthPacketProcessor] avg. time: 14.5741ms -> ~68.6151Hz
[ INFO] [Kinect2Bridge::main] depth processing: ~159.629ms (~6.26451Hz) publishing rate: ~9.93995Hz
[ INFO] [Kinect2Bridge::main] color processing: ~177.76ms (~5.62555Hz) publishing rate: ~10.6026Hz

wiedemeyer, 2015-11-24:
Could it be that you CPU is under full load? The Broadwell U is just a dual core, which is limited 15 or 28 Watt TDP. Depending on which topics you listen too, the CPU load can be quite high, especially for the compressed topics. The kinect2_bridge can only process 30 fps, if the CPU is strong enough. I guess you CPU is the limiting factor here. Try to stop all other applications that are using the CPU a lot, and just run kinect2_bridge and `rostopic hz /kinect2/sd/camera_info`, then you can see the maximum frame rate your system can get.

wiedemeyer, 2015-11-24:
One question, which depth image is zero sometimes? Could you write a simple program that listens to a topic and just checks if the image contains just zeros? Try with `/kinect2/sd/image_depth`. If that image is always correct, try it with `/kinect2/sd/image_depth_rect`, `/kinect2/qhd/image_depth_rect` and `/kinect2/hd/image_depth_rect`.

shamoorti, 2015-11-24:
By the way is it necessary to have Opencv and PCL stand alone installation? In my last installation of bridge I used to ahve them but now with only the libraries in the bridge system seems to work. Do you think that is also an issue?
This is what I get : 
average rate: 4.739
	min: 0.079s max: 0.398s std dev: 0.13574s window: 4
average rate: 5.710
	min: 0.000s max: 0.449s std dev: 0.15815s window: 10
average rate: 6.446
	min: 0.000s max: 0.449s std dev: 0.14452s window: 19
average rate: 6.348
	min: 0.000s max: 0.449s std dev: 0.14229s window: 23
average rate: 6.654
	min: 0.000s max: 0.509s std dev: 0.16041s window: 31
average rate: 6.907
	min: 0.000s max: 0.509s std dev: 0.15862s window: 39
average rate: 6.413
	min: 0.000s max: 0.606s std dev: 0.17548s window: 43
average rate: 6.208
	min: 0.000s max: 0.678s std dev: 0.18802s window: 49
average rate: 6.547
	min: 0.000s max: 0.678s std dev: 0.17840s window: 58
average rate: 6.729
	min: 0.000s max: 0.678s std dev: 0.16874s window: 67
average rate: 6.591
	min: 0.000s max: 0.678s std dev: 0.16852s window: 72
average rate: 6.634
	min: 0.000s max: 0.678s std dev: 0.16808s window: 79
average rate: 6.426
	min: 0.000s max: 0.678s std dev: 0.16916s window: 82
average rate: 6.295
	min: 0.000s max: 0.757s std dev: 0.17904s window: 86
^Caverage rate: 6.438
	min: 0.000s max: 0.757s std dev: 0.17555s window: 96

shamoorti, 2015-11-24:
Kinda strange behaviour! Now with the same options in fps limit color is procession ~51654Hz!!!:
mdh-butler@mdhbutler-desktop:~$ /home/mdh-butler/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge _fps_limit:=5
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: 5
       calib_path: /home/mdh-butler/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Warning] [OpenCLDepthPacketProcessorImpl] Your ocl-icd has deadlock bugs. Update to 2.2.4+ is recommended.
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 025613444747
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 025613444747 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 025613444747
[ INFO] [Kinect2Bridge::initDevice] device firmware: 4.3.3913.0.7
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...
[Info] [Freenect2DeviceImpl] stopped
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [DepthRegistrationOpenCL::init] devices:
[ INFO] [DepthRegistrationOpenCL::init]   0: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [DepthRegistrationOpenCL::init] selected device: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [DepthRegistrationOpenCL::init] devices:
[ INFO] [DepthRegistrationOpenCL::init]   0: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [DepthRegistrationOpenCL::init] selected device: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [Kinect2Bridge::main] waiting for clients to connect
[ INFO] [Kinect2Bridge::callbackStatus] client connected. starting device...
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
[Info] [DepthPacketStreamParser] -9 packets were lost
[Info] [DepthPacketStreamParser] 7 packets were lost
[ INFO] [Kinect2Bridge::main] depth processing: ~28.004ms (~35.7091Hz) publishing rate: ~4.65497Hz
[ INFO] [Kinect2Bridge::main] color processing: ~0.0667969ms (~14970.7Hz) publishing rate: ~3.98997Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 10.0552ms -> ~99.4512Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.3386ms -> ~57.6749Hz
[ INFO] [Kinect2Bridge::main] depth processing: ~33.8631ms (~29.5306Hz) publishing rate: ~4.99598Hz
[ INFO] [Kinect2Bridge::main] color processing: ~0.0186125ms (~53727.2Hz) publishing rate: ~4.99598Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 10.4316ms -> ~95.863Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.5468ms -> ~60.4346Hz
[ INFO] [Kinect2Bridge::main] depth processing: ~32.0506ms (~31.2007Hz) publishing rate: ~5.32167Hz
[ INFO] [Kinect2Bridge::main] color processing: ~0.020355ms (~49128Hz) publishing rate: ~5.32167Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 10.8037ms -> ~92.5611Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.3249ms -> ~61.2561Hz
[ INFO] [Kinect2Bridge::main] depth processing: ~34.5921ms (~28.9083Hz) publishing rate: ~4.98428Hz
[ INFO] [Kinect2Bridge::main] color processing: ~0.0193596ms (~51654Hz) publishing rate: ~4.98428Hz

wiedemeyer, 2015-11-24:
That Hz is just a number calculated from the time spend processing the images in kinect2_bridge, since you did not listen to a color image, there is nothing to be processed.

shamoorti, 2015-11-24:
Yes, exactly. I forgot about that!

shamoorti, 2015-11-24:
I found the problem Thiemo, the depth is not zero but it contains some zero values in HD rect, Hence when I find a pixel in RGB and then if I want it to find it in depth so it may be zero in depth. And that's a problem.
[image]

wiedemeyer, 2015-11-24:
Yes, you need to check whether the depth value is valid or not.

closed 2015-12-09 by wiedemeyer