code-iai/iai_kinect2 #186: [kinect2_head_kinect2_bridge-2] process has died

sshonig, 2015-11-25:
Hi,

Whenever I try to run kinect2_bridge i receive the error below repeatedly. I am running kinect2_bridge on Ubuntu 14.04 (on a dual boot computer alongside Windows 10). 
While trying to get to the bottom of this, I noticed that kinect2_bridge is not receiving any depth data. However, when I run the kinect sdk depth test in Windows 10 on the same computer, it seems to detect depth data via my USB 3.0 fine.

Any ideas as to what might be the problem or what I might do to explore the problem further? I've been hacking at it for hours...

I am extremely new  at ros, so if I'm leaving out any relevant information please let me know and I'll send it over. 

Thanks!

SUMMARY
========

PARAMETERS
 * /kinect2_head_kinect2_bridge/sensor_id: 
 * /kinect2_head_kinect2_bridge/sensor_name: kinect2_head
 * /kinect2_head_points_xyzrgb_highres/queue_size: 2
 * /kinect2_head_points_xyzrgb_lowres/queue_size: 2
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /
    kinect2_head_kinect2_bridge (kinect2_bridge/kinect2_bridge)
    kinect2_head_points_xyzrgb_highres (nodelet/nodelet)
    kinect2_head_points_xyzrgb_lowres (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [2712]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 599687aa-938b-11e5-be87-5c514f70e57c
process[rosout-1]: started with pid [2725]
started core service [/rosout]
process[kinect2_head_kinect2_bridge-2]: started with pid [2728]
process[kinect2_head_points_xyzrgb_lowres-3]: started with pid [2729]
process[kinect2_head_points_xyzrgb_highres-4]: started with pid [2730]
[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistration::New] Using OpenCL registration method!
[OpenCLDepthPacketProcessor::init] ERROR: clGetPlatformIDs(-1001)
terminate called after throwing an instance of 'cl::Error'
  what():  clGetPlatformIDs

[kinect2_head_kinect2_bridge-2] process has died [pid 2728, exit code -6, cmd /home/shanee/workspace/ros/catkin/devel/lib/kinect2_bridge/kinect2_bridge __name:=kinect2_head_kinect2_bridge __log:=/home/shanee/.ros/log/599687aa-938b-11e5-be87-5c514f70e57c/kinect2_head_kinect2_bridge-2.log].
log file: /home/shanee/.ros/log/599687aa-938b-11e5-be87-5c514f70e57c/kinect2_head_kinect2_bridge-2*.log
[kinect2_head_kinect2_bridge-2] restarting process
process[kinect2_head_kinect2_bridge-2]: started with pid [2810]
[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistration::New] Using OpenCL registration method!
[OpenCLDepthPacketProcessor::init] ERROR: clGetPlatformIDs(-1001)
terminate called after throwing an instance of 'cl::Error'
  what():  clGetPlatformIDs

JimmyDaSilva, 2015-11-30:
Hello @sshonig ,

This problem is not a problem with kinect2_bridge but with freenect2, but anyway I'll try to help you here!
It seems that the problem is with OpenCL, so you need to read carefully the installation procedure about OpenCL with libfreenect2, here at paragraph 5:
https://github.com/OpenKinect/libfreenect2#debianubuntu-1404

Once you have done so, recompile and reinstall libfreenect2.
In ```libfreenect2/build``` do :
```make && sudo make install```

Then you can try running the Kinect2 by using Protonect.
In ```libfreenect2/build/bin``` try to run these commands:
``` 
./Protonect cpu
./Protonect cl
./Protonect gl
```
That way you will try to run the Kinect2 with the cpu, opencl and opengl.

If one of these fails, you can try to fix it by reading again the installation instructions, or you can post an issue to libfreenect2.
If you are able to run the kinect2 with Protonect but not with kinect2_bridge @wiedemeyer will try to help you here

Good luck !

wiedemeyer, 2015-12-14:
no updates, so I close the issue

closed 2015-12-14 by wiedemeyer

Abbyls, 2016-08-28:
@wiedemeyer @JimmyDaSilva @sshonig @mintar 
Hi, I just met the same problem with @sshonig , but i am using NVIDIA Jetson TX1, the OPENCL can't support it! what should i do, i feel very sad, anyone kown the solutions?