code-iai/iai_kinect2 #187: Problem with Kinect2_bridge package

djo90, 2015-11-27:
when I did rosrun kinect2_bridge kinect2_bridge I got this error message
[rospack] Error: package 'kinect2_bridge' not found

so I tried those command 
sudo -i
source /home/djo/catkin_ws/devel/setup.bash
rosrun kinect2_bridge kinect2_bridge

I got this error 
[Error] [Freenect2Impl] Your libusb does not support large iso buffer!

So I tried your command 
ldd ./Protonect | grep libusb

and I got this message 
libusb-1.0.so.0 => /home/djo/libfreenect2/depends/libusb/lib/libusb-1.0.so.0 (0x00007f0eaebee000)

what should I do in my case to can run kinect2_bridge?

shamoorti, 2015-12-01:
Did you get any error while installing Libfreenect2? Your bridge viewer works?

wiedemeyer, 2015-12-01:
Is `Protonect` working, or does it give the same error? Have you tried to source you work space and run it without sudo?

djo90, 2015-12-01:
as long I run Protonect from the bin in the depends folder it works perfect. I get 4 screens from my image, but once I try to run Protonect from the bin in the main libfreenect2 folder I get error in build. So when I want to run rosrun kinect2_bridge kinect2_bridge I guess that the bridge connect to the "not working" Protonect (the one in the libfreenect2) and for that reason there is no viewers.

djo90, 2015-12-01:
I fixed, the problem. Now Protonect works perfect when I run it alone. But when I do rosrun kinect2_bridge kinect2_bridge, I get this and no viewer appears. How can I fix that?

[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/djo/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel HD Graphics FamilyO (GPU)[Intelc]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel HD Graphics FamilyO (GPU)[Intelc]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
/tmp/file0qLB2O.cl:16803:54: warning: double precision constant requires cl_khr_fp64, casting to single precision
  float max_depth = phase_final * UNAMBIGIOUS_DIST * 2.0;
                                                     ^
/tmp/file0qLB2O.cl:16807:32: warning: double precision constant requires cl_khr_fp64, casting to single precision
  xmultiplier = (xmultiplier * 90.0) / (max_depth * max_depth * 8192.0);
                               ^
/tmp/file0qLB2O.cl:16807:65: warning: double precision constant requires cl_khr_fp64, casting to single precision
  xmultiplier = (xmultiplier * 90.0) / (max_depth * max_depth * 8192.0);
                                                                ^
3 warnings generated.
ASSERTION FAILED: TODO Boolean values cannot escape their definition basic block
  at file /build/buildd/beignet-0.3/backend/src/llvm/llvm_gen_backend.cpp, function void gbe::GenWriter::emitMovForPHI(llvm::BasicBlock*, llvm::BasicBlock*), line 1012
Stack dump:
0.	Running pass 'Function Pass Manager' on module '/tmp/fileF4wAPI.ll'.
1.	Running pass 'Gen Back-End' on function '@processPixelStage2'

shamoorti, 2015-12-02:
look at this thread, it's the same problem as yours. 
 Failed to open X display at launch kinect2_bridge.launch with machine tag #166

wiedemeyer, 2015-12-09:
BTW: kinect2_bridge will not open any viewers. For that purpose you have to run kinect2_viewer.

djo90, 2015-12-16:
Even when I use the command 
rosrun kinect2_viewer kinect2_viewer kinect2 sd cloud 

It shows those lines, and there is any viewer that appears.

[ INFO] [main] topic color: /kinect2/sd/image_color_rect
[ INFO] [main] topic depth: /kinect2/sd/image_depth_rect
[ INFO] [main] starting receiver...

wiedemeyer, 2015-12-16:
kinect2_bridge is the publisher, that publishes the images from the device through ROS topics. kinect2_viewer is a ROS node that receives images from ROS topics to display them. In order to get it displaying images you need to run kinect2_bridge as well.

closed 2016-01-04 by wiedemeyer