code-iai/iai_kinect2 #188: Strange values from depth map!

shamoorti, 2015-12-01:
Hello guys,
As you know I am working on a robotic project which I find and object in RGB image (x,y values) and send it to depth map to find the representative value in depth which is the distance to the object. But I get to much offset. In this picture you can see the cup being found in matlab and the red star in the middle of the cup is the point which I want to find the distance to it hence the distance to object. But as I send it to depth map it like depth = double(depth(x,y) or using impixel(depth, x ,y) I get wrong values. In the case of this picture the distance shows 1.65 meters which the actual value is  1 meter exact. I subscribe :
Kinect = rossubscriber('/kinect2/hd/image_mono');
Depth = rossubscriber('/kinect2/hd/image_depth_rect');  
Then I receive the them fine but the depth value is wrong.
[image]
Anyone has any idea what's wrong? By the way Rviz shows the perfect cloud but faced to the ground like this. Is this tf problem or just like this is ok?
[image]
?

wiedemeyer, 2015-12-01:
On default z is pointing up in RViz. You could use a static TF publisher to provide a transform for the TF link of the Kinect2.
I don't know what `impixel` or the other does. But the depth values are stored in mm, so it would be `depth(y, x) / 1000.0` to get the measurement in meters. On the other hand taking just one point is a bad idea, better take the average or median of a region around the point you are looking for.

wiedemeyer, 2015-12-01:
And your should also use `mono_rect` together with the `depth_rect`, else the image coordinates won't match between the images.

shamoorti, 2015-12-01:
Thanks a lot Thiemo for the good tip. Also look at this point cloud from rviz. It seems cup is so shifted back in depth. Still camera is not calibrated I think.
[image]

shamoorti, 2015-12-02:
Yeah there was definitely calibration problems which is fixed now. Also thank Thiemo for the good advice about the centre point. You can close this thread.

closed 2015-12-09 by wiedemeyer