code-iai/iai_kinect2 #195: general calibration questions?

TheodoreT, 2015-12-13:
Hi Thiemo, 

First before I begin many thanks for your efforts and your work here. At the moment I started working in a project where I will have to work with the new Kinect one sensor (here to note that I have no previous experience with neither the old kinect xbox nor with the new kinect one). Therefore I started playing around with it and I successfully managed to acquire images using the Kinect SDK v2.0 (my intention is, at some point, to switch to libfreenect2 library in order to be OS independent) and opencv. Now what I would like to do is to overlay the two views from the rgb sensor and the ir/depth sensor in order to create a rgb-d image as you are doing here with the registration in the calibration sub-project. 

For that reason I have searched around the internet in order to find out how I can do that, plus I tried to go through the code you have supplied here in order to see how you are doing it, but I have to admit that it was not that easy for me. From what I understood I need to calibrate the two sensors like a stereo vision system and the proper way is first to calibrate the two sensors individually and then apply any mapping between the two views. This procedure seems to be quite straight forward if the two cameras have the same resolution and aspect ratio. However, considering that the sensors in the Kinect one differ, i.e. the color camera has a full HD resolution (1920x1080 pixel) and the depth/infrared image is 512x424 pixel respectively in addition to the different aspect ratio, this procedure seems to become a bit tricky. Therefore, I have some questions which I would like, if you also can to help me, clear in my mind since it seems that you already have this experience.

For the calibration I am gonna be using the opencv library as well. The procedure seems to be the following: first individually calibrate the two sensors using the `calibrate()` function and then use the extracted cameraMatrix/intrinsic and distortion coefficients/distCoeffs values into the `stereoCalibrate()` function in order to obtain the rotation(R) and translation(T) matrices plus some others, e.g. R1, R2, P1, P2, Q,etc..., needed later for the mapping from `stereoRectify()`, is my thought correct? 

However, here also comes the question how to deal with the different resolution and aspect ratio of the two sensors. Should I resize one of the images to the resolution of the other before I do any of the previous procedures, I noticed that you are resizing the small resolution view to the high resolution view, can this be done on the other way around? is there any reason for going like that. Moreover, what about the aspect ratio, how do you deal with this since even if you resize the image there gonna be some misalignment due to the latter. 

Lastly would be easy to explain a bit how you apply the registration afterwards, when you have obtained the needed matrices. I noticed that you were doing something with distances and using the `rodrigues()`, `solvePnP()`, etc... functions. However, it was not clear to me as well what you are doing. Therefore, I would be grateful to you if you could describe me how you are doing the registration.

Please accept my apologies for the long message but I was struggling to find some information for quite some time now and I am a bit confused.

Thanks in advance for your time and I would appreciate any feedback.

wiedemeyer, 2015-12-14:
Hi,
I only use stereo calibration to compute the rotation and translation between the sensors. For the depth to color registration I do the following:
- Compute the 3D point for each depth pixel
- Rotate and translate the points (this results in points as seen from the position of the color sensor)
- Create a mapping between both sensors using the intrinsic parameters of both. With this mapping you can lookup where a pixel from one camera would be in the other, this takes into account aspect ratio, etc.
- map each transformed depth pixel to the color sensor intrinsic paramters.
There is some additional code in kinect2_registration, this is for up-scaling and interpolation.

There is also another registration, which is done by libfreenect2, it calculates for each depth pixel the corresponding color pixel, here you end up with a color image in the frame of the ir sensors.

TheodoreT, 2015-12-14:
Hi Thiemo, thank you for the response. Ok I got it about the calibration. However, regarding the mapping would your approach work if I wanted to do the opposite, meaning instead of scaling up the ir sensor view, scaling down the rgb sensor view?

wiedemeyer, 2015-12-14:
well then simply use the registration that comes with libfreenect2.

TheodoreT, 2015-12-14:
Ok, I will try to have a look on their code as well. Thanks again for your time :-).

closed 2015-12-15 by wiedemeyer