code-iai/iai_kinect2 #196: Kinect_bridge gets slower after some hours of use.

filipetrocadoferreira, 2015-12-17:
I'm using Kinect_bridge in a i5 without any GPU.  With the "depth_method" parameter set to default, it runs opengldetphregistration (it should not be opencl?)

When I fresh install libfreenect and this package and run Kinect_Bridge i can get in qhd 50 fps. But after some hours of use it easily gets 6fps to qhd and 15 to sd.

wiedemeyer, 2015-12-18:
It is only OpenCL if you compiled libfreenect2 with OpenCL support.
The Kinect2 is only able to deliver 30 fps, so 50 fps cannot be true. Maybe you are referring to the calculated maximum frame rate, that your system could handle.
What CPU are you using? What is the exact name?
The best performance should be achieved, when the number of threads is <= number of CPU cores.
That the frame rate gets lower over time might be a heat issue. Could it be, that the CPU is heating up and therefore lower the clocks? Please check if the temperatures could be the source.

filipetrocadoferreira, 2015-12-18:
Ok, I could had read wrong the initial frame_rate but it was visually quite faster. 

The CPU is i5-4460.

And I can only turn the system faster again after format the disc. So it's not related with CPU temperature. 

Can it be related to compiling other ROS packages and some shared compilation flags?

wiedemeyer, 2015-12-18:
I don't know. Maybe something with your system is wrong.

wiedemeyer, 2016-01-11:
any update on this?

closed 2016-01-15 by wiedemeyer

filipetrocadoferreira, 2016-02-22:
Not really. It seems like when CPU temperature increases the performance of iai_kinect bridge, mainly when subscribing to qhd/depth topics decreases a lot.