code-iai/iai_kinect2 #197: no Kinect2 devices found in odroid xu4(arm board and Mali GPU)

CrazyPopLin, 2015-12-28:
Hi, I tried to use iai_kinect2 in odroid xu4 which is arm based board  has Mali-T628 MP6(OpenGL ES 3.0/2.0/1.1 and OpenCL 1.1 Full profile).
According to Mali GPU not support OpenGL and I build libfreenect2 without any OpenGL dependency and build successfully. 

>odroid@odroid:~/Dependency/libfreenect2/build/bin$ ./Protonect cl -noviewer
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:6 with serial 501169643042
[Info] [Freenect2Impl] found 1 devices
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl]   1: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Warning] [OpenCLDepthPacketProcessorImpl] Your ocl-icd has deadlock bugs. Update to 2.2.4+ is recommended.
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Debug] [DepthPacketStreamParser] image data too short!
[Info] [Freenect2DeviceImpl] started
device serial: 501169643042
device firmware: 2.3.3912.0
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] not all subsequences received 497
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Info] [DepthPacketStreamParser] 17 packets were lost
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
...
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Info] [TurboJpegRgbPacketProcessor] avg. time: 63.3753ms -> ~15.779Hz
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
...
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [OpenCLDepthPacketProcessor] avg. time: 87.2436ms -> ~11.4622Hz
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
The viewer is turned off. Received 100 frames. Ctrl-C to stop.
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
>[Debug] [DepthPacketStreamParser] skipping depth packet

As @xlz said, If I got frame rate info it should be programmatically correct. 
And After I build iai_kinect2 successfully and `rosrun kinect2_bridge kinect2_bridge `:

>odroid@odroid:~$ rosrun kinect2_bridge kinect2_bridge
[Error] [Freenect2Impl] Your libusb does not support large iso buffer!
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/odroid/Skywalker/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl]   1: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Warning] [OpenCLDepthPacketProcessorImpl] Your ocl-icd has deadlock bugs. Update to 2.2.4+ is recommended.
[ERROR] [Kinect2Bridge::initDevice] no Kinect2 devices found!
>[ERROR] [Kinect2Bridge::start] Initialization failed!

Also I have rebuilt the libfreenect2 just incase and not works.
Any suggestions will be appreciated and thanks for your help.

xlz, 2015-12-28:
> [Error] [Freenect2Impl] Your libusb does not support large iso buffer!

Linking to the wrong libusb.

> Also I have rebuilt the libfreenect2 just incase and not works.

What does this sentence mean?

CrazyPopLin, 2015-12-28:
>Linking to the wrong libusb.

I am just confused, because for the libfreenect2, the libusb seems right. 
 >odroid@odroid:~/Dependency/libfreenect2/build/bin$ ldd ./Protonect | grep libusb
>	libusb-1.0.so.0 => /home/odroid/Dependency/libfreenect2/depends/libusb/lib/libusb-1.0.so.0 (0xb6d68000)

I need to install libsub in the iai_kinet2 path direction again like `install_libusb.sh` does?

Just a tutorial in main page of iai_kinect2 .https://github.com/code-iai/iai_kinect2
>Protonect works fine, but kinect2_bridge is still not working / crashing.
>
If that is the case, you have to make sure that Protonect uses the same version of libfreenect2 as kinect2_bridge does. To do so, run make and sudo make install in the build folder again. And try out kinect2_bridge again.
>
cd libfreenect2/build
>make & sudo make install

Thanks.

CrazyPopLin, 2015-12-29:
fixed it after reinstall libusb, thx.

closed 2015-12-29 by CrazyPopLin