code-iai/iai_kinect2 #198: Flat surfaces appear curved - is this normal?

mriem, 2015-12-28:
Hello, 

I started using the kinect one a week ago and found this great package in ROS which really eased the process for me. Thanks!

Anyway, I have done a calibration as described in the git hub page and I got back some results which seem to make sense, i.e., they are in the same order of magnitude as the examples in the kinect2_bridge/data.

The problem I have is that flat surfaces such as walls appear curved. Before this kinect one, I used and Asus Xtion and this effect was not so noticeable. I want to know if you people also have noticed this (or if I am doing something wrong).

To be honest, the curving effect is also visible when I run kinect2_bridge without the calibration (default values), but I thought the calibration should somehow try to compensate for this.

Attached are some images which show what I am talking about.

Thanks in advance for your help.

[image]
[image]
[image]
[image]

Miguel Oliveira

vishu2287, 2015-12-29:
@mriem , You need to calibrate kinect2 
https://github.com/code-iai/iai_kinect2/tree/master/kinect2_calibration#calibrating-the-kinect-one

miguelriemoliveira, 2015-12-29:
Hi Vishu,

Thanks for the feedback.

I had already done the calibration as described in the git hub page. I use around 150 RGB, IR and Sync images. Still, I get these curved surfaces. 

Any other ideas? Could it be that the calibration is wrong / not optimal?

Thanks, Miguel

wiedemeyer, 2015-12-29:
I think it's a general problem of ToF sensors and has nothing to do with the calibration. See issue #121.

mriem, 2015-12-29:
Hello,

Thanks for the repply. I saw issue #121 as you suggested.

There is just one thing that does not make sense to me:  if its a problem with the ToF, then why does the curved effect appear greater near the borders of the image? 

Take a look at this image. The camera is placed almost parallel to the right wall plane, at about 20 cm from it. Still, the curved effect appears to be greater in the ceiling (which is close to the borders of the image)

[image]

Anyway, I am closing this issue since its clearly not a kinect2_bridge problem (also occurs in the kinect2 sdk). 

Miguel

closed 2015-12-29 by mriem