code-iai/iai_kinect2 #200: kinect2_bridge crash shortly after launch

ahundt, 2015-12-31:
I'd appreciate some help with getting the bridge running. We have first verified that Protonect runs in cpu,opencl,and opengl mode successfully. We are using OpenCV3 and Ubuntu 14.04 with ROS Indigo. 

I'm trying to use the kinect2 bridge which I launch as follows:

```
roscore
rosrun kinect2_bridge kinect2_bridge_
```
(note we renamed the executable with an _ so it would work with eclipse debuggers, it isn't an important detail to the actual problem we're having)

It then runs as follows, ultimately crashing after about a second and producing a core dump:

```
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/cpaxton/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 16 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @7:6 with serial 127442134447
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 127442134447 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 127442134447
[ INFO] [Kinect2Bridge::initDevice] device firmware: 4.1.3912.0.7
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...
[Info] [Freenect2DeviceImpl] stopped
Segmentation fault (core dumped)
```

Here is the gdb session where we opened the core dump. At the bottom of this text is a backtrace indicating where the crash occurred: 
```
gdb '/home/cpaxton/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge_' core
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show c29 commits opying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/cpaxton/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge_...done.
[New LWP 4706]
[New LWP 4716]
[New LWP 4745]
[New LWP 4717]
[New LWP 4746]
[New LWP 4723]
[New LWP 4718]
[New LWP 4724]
[New LWP 4725]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

warning: the debug information found in "/lib64/ld-2.19.so" does not match "/lib64/ld-linux-x86-64.so.2" (CRC mismatch).

Core was generated by `/home/cpaxton/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge_'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007fb227e3c282 in cv::Mat::copyTo(cv::_OutputArray const&) const () from /usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4
Traceback (most recent call last):
  File "/usr/share/gdb/auto-load/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.19-gdb.py", line 63, in <module>
    from libstdcxx.v6.printers import register_libstdcxx_printers
ImportError: No module named 'libstdcxx'
(gdb) bt
#0  0x00007fb227e3c282 in cv::Mat::copyTo(cv::_OutputArray const&) const () from /usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4
#1  0x000000000044b9d1 in cv::Mat::clone (this=0x7fff3d249a38) at /usr/local/include/opencv2/core/mat.inl.hpp:628
#2  0x0000000000455510 in Kinect2Bridge::initDevice (this=0x7fff3d2497b0, sensor=...)
    at /home/cpaxton/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:556
#3  0x000000000044fe38 in Kinect2Bridge::initialize (this=0x7fff3d2497b0)
    at /home/cpaxton/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:284
#4  0x000000000044e252 in Kinect2Bridge::start (this=0x7fff3d2497b0)
    at /home/cpaxton/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:147
#5  0x00000000004496fd in main (argc=1, argv=0x7fff3d24a848)
    at /home/cpaxton/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1418
(gdb) Quit
(gdb) 
```
This issue is also of note to @fjonath1 and @cpaxton 

Thanks!

ahundt, 2015-12-31:
Seems to be failing on this line:
https://github.com/code-iai/iai_kinect2/blob/master/kinect2_bridge/src/kinect2_bridge.cpp#L556

The issue may be that both opencv2 and opencv3 are being linked, which means it is my fault and not yours, will confirm and close the issue shortly!

```
cpaxton@balmorra:~/catkin_ws/devel/lib/kinect2_bridge$ ldd kinect2_bridge_ 
	linux-vdso.so.1 =>  (0x00007ffff54f2000)
	libnodeletlib.so => /opt/ros/indigo/lib/libnodeletlib.so (0x00007f4a2d839000)
	libopencv_imgproc.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.2.4 (0x00007f4a2d37b000)
	libopencv_core.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4 (0x00007f4a2cf44000)
	libclass_loader.so => /opt/ros/indigo/lib/libclass_loader.so (0x00007f4a2cd25000)
	libtf.so => /opt/ros/indigo/lib/libtf.so (0x00007f4a2caf3000)
	libroscpp.so => /opt/ros/indigo/lib/libroscpp.so (0x00007f4a2c799000)
	libroscpp_serialization.so => /opt/ros/indigo/lib/libroscpp_serialization.so (0x00007f4a2c596000)
	librosconsole.so => /opt/ros/indigo/lib/librosconsole.so (0x00007f4a2c36c000)
	librostime.so => /opt/ros/indigo/lib/librostime.so (0x00007f4a2c142000)
	libboost_system.so.1.54.0 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.54.0 (0x00007f4a2bf3e000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f4a2bd1f000)
	libconsole_bridge.so.0.2 => /usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.2 (0x00007f4a2bb15000)
	libkinect2_registration.so => /home/cpaxton/catkin_ws/devel/lib/libkinect2_registration.so (0x00007f4a2b903000)
	libopencv_imgcodecs.so.3.0 => /usr/local/lib/libopencv_imgcodecs.so.3.0 (0x00007f4a2b66e000)
	libopencv_core.so.3.0 => /usr/local/lib/libopencv_core.so.3.0 (0x00007f4a2a9db000)
	libfreenect2.so => /usr/local/lib/libfreenect2.so (0x00007f4a2a798000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f4a2a493000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f4a2a18d000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f4a29f77000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f4a29bb1000)
	libbondcpp.so => /opt/ros/indigo/lib/libbondcpp.so (0x00007f4a2997f000)
	libtinyxml.so.2.6.2 => /usr/lib/x86_64-linux-gnu/libtinyxml.so.2.6.2 (0x00007f4a2976a000)
	libroslib.so => /opt/ros/indigo/lib/libroslib.so (0x00007f4a29558000)
	libboost_filesystem.so.1.54.0 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.54.0 (0x00007f4a29342000)
	libboost_thread.so.1.54.0 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.54.0 (0x00007f4a2912c000)
	libtbb.so.2 => /usr/lib/libtbb.so.2 (0x00007f4a28ef7000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f4a28cde000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f4a28ad5000)
	libGL.so.1 => /usr/lib/nvidia-304/libGL.so.1 (0x00007f4a287b8000)
	libPocoFoundation.so.9 => /usr/lib/libPocoFoundation.so.9 (0x00007f4a2846c000)
	libtf2_ros.so => /opt/ros/indigo/lib/libtf2_ros.so (0x00007f4a281cd000)
	libtf2.so => /opt/ros/indigo/lib/libtf2.so (0x00007f4a27f9d000)
	libxmlrpcpp.so => /opt/ros/indigo/lib/libxmlrpcpp.so (0x00007f4a27d7f000)
	libcpp_common.so => /opt/ros/indigo/lib/libcpp_common.so (0x00007f4a27b76000)
	librosconsole_log4cxx.so => /opt/ros/indigo/lib/librosconsole_log4cxx.so (0x00007f4a27962000)
	librosconsole_backend_interface.so => /opt/ros/indigo/lib/librosconsole_backend_interface.so (0x00007f4a2775f000)
	liblog4cxx.so.10 => /usr/lib/liblog4cxx.so.10 (0x00007f4a27377000)
	libboost_regex.so.1.54.0 => /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.54.0 (0x00007f4a27070000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f4a2da92000)
	libopencv_imgproc.so.3.0 => /usr/local/lib/libopencv_imgproc.so.3.0 (0x00007f4a260ed000)
	libgomp.so.1 => /usr/lib/x86_64-linux-gnu/libgomp.so.1 (0x00007f4a25ede000)
	libjpeg.so.8 => /usr/lib/x86_64-linux-gnu/libjpeg.so.8 (0x00007f4a25c89000)
	libpng12.so.0 => /lib/x86_64-linux-gnu/libpng12.so.0 (0x00007f4a25a62000)
	libtiff.so.5 => /usr/lib/x86_64-linux-gnu/libtiff.so.5 (0x00007f4a257f0000)
	libjasper.so.1 => /usr/lib/x86_64-linux-gnu/libjasper.so.1 (0x00007f4a25599000)
	libIlmImf.so.6 => /usr/lib/x86_64-linux-gnu/libIlmImf.so.6 (0x00007f4a252e9000)
	libHalf.so.6 => /usr/lib/x86_64-linux-gnu/libHalf.so.6 (0x00007f4a250a6000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f4a24ea2000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f4a24c89000)
	libturbojpeg.so.0 => /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0 (0x00007f4a24a2f000)
	libglfw.so.3 => /usr/lib/x86_64-linux-gnu/libglfw.so.3 (0x00007f4a2481c000)
	libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007f4a24616000)
	librospack.so => /opt/ros/indigo/lib/librospack.so (0x00007f4a243d2000)
	libnvidia-tls.so.304.131 => /usr/lib/nvidia-304/tls/libnvidia-tls.so.304.131 (0x00007f4a241ce000)
	libnvidia-glcore.so.304.131 => /usr/lib/nvidia-304/libnvidia-glcore.so.304.131 (0x00007f4a21de4000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f4a21aaf000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f4a2189c000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f4a2165e000)
	libactionlib.so => /opt/ros/indigo/lib/libactionlib.so (0x00007f4a21441000)
	libaprutil-1.so.0 => /usr/lib/x86_64-linux-gnu/libaprutil-1.so.0 (0x00007f4a21219000)
	libapr-1.so.0 => /usr/lib/x86_64-linux-gnu/libapr-1.so.0 (0x00007f4a20fe8000)
	libicuuc.so.52 => /usr/lib/x86_64-linux-gnu/libicuuc.so.52 (0x00007f4a20c6e000)
[... cut ...]
	libnih.so.1 => /lib/x86_64-linux-gnu/libnih.so.1 (0x00007f4a1cce7000)
	libnih-dbus.so.1 => /lib/x86_64-linux-gnu/libnih-dbus.so.1 (0x00007f4a1cadd000)
	libdbus-1.so.3 => /lib/x86_64-linux-gnu/libdbus-1.so.3 (0x00007f4a1c898000)
	libXrender.so.1 => /usr/lib/x86_64-linux-gnu/libXrender.so.1 (0x00007f4a1c68d000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f4a1c48a000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f4a1c285000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f4a1c07f000)
```

ahundt, 2015-12-31:
That was the problem, sorry for the spurious issue.

closed 2015-12-31 by ahundt

wiedemeyer, 2015-12-31:
Hi,
you are right, it will not work because cv_bridge needs OpenCV 2.4.x.