code-iai/iai_kinect2 #201: Dependency problems while installing iai_kinect2

rshanor, 2016-01-01:
I have been following the install guide and ran into issues building iai_kinect2. Protonect works fine on my computer. After cloning, calling "rosdep install -r --from-paths" doesn't seem to do anything and returns "rosdep: error: no packages or stacks specified". When I then try catkin_make, I am given a string of errors related to what looks like libfreenect2 dependencies. I am new to ROS. Any idea what I might be doing wrong? Thanks!  

/home/rshanor/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘void Kinect2Bridge::initConfig(bool, bool, double, double)’:
/home/rshanor/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:400:5: error: ‘Config’ is not a member of ‘libfreenect2::Freenect2Device’
     libfreenect2::Freenect2Device::Config config;
     ^
/home/rshanor/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:400:43: error: expected ‘;’ before ‘config’
     libfreenect2::Freenect2Device::Config config;
                                           ^
/home/rshanor/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:401:5: error: ‘config’ was not declared in this scope
     config.EnableBilateralFilter = bilateral_filter;
     ^
/home/rshanor/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:405:13: error: ‘class libfreenect2::Freenect2Device’ has no member named ‘setConfiguration’
     device->setConfiguration(config);
             ^
/home/rshanor/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘void Kinect2Bridge::initConfig(bool, bool, double, double)’:
/home/rshanor/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:400:5: error: ‘Config’ is not a member of ‘libfreenect2::Freenect2Device’
     libfreenect2::Freenect2Device::Config config;
     ^
/home/rshanor/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:400:43: error: expected ‘;’ before ‘config’
     libfreenect2::Freenect2Device::Config config;
                                           ^
/home/rshanor/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:401:5: error: ‘config’ was not declared in this scope
     config.EnableBilateralFilter = bilateral_filter;
     ^
/home/rshanor/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:405:13: error: ‘class libfreenect2::Freenect2Device’ has no member named ‘setConfiguration’
     device->setConfiguration(config);

wiedemeyer, 2016-01-04:
Hi, please make sure to use the most recent version of libfreenect2.

rshanor, 2016-01-05:
Yup, that fixed it. Thanks!

closed 2016-01-05 by rshanor