code-iai/iai_kinect2 #202: Depth offset in calibration?

soulslicer, 2016-01-04:
I notice the calibration package generates a calib_depth.yaml. What exactly does this do?

wiedemeyer, 2016-01-04:
It is an offset for the depth values.

soulslicer, 2016-01-04:
May I know how this is arrived at/required? As I understand, calib_ir and calib_color are the instrinsic/projection matrices of each camera, and calib_pose is the extrinsic transform between both cameras. I am doing the calibration step using an external software, so I am trying to figure out how the depth offset is calculated.

wiedemeyer, 2016-01-04:
I think it is explained here: https://github.com/code-iai/iai_kinect2/tree/master/kinect2_calibration#calibration-of-the-depth-measurements
Basically the distance to the pattern board is calculated and compared to the measured distance. The offset is the average difference between the calculated and the measured distances.

closed 2016-01-04 by soulslicer