code-iai/iai_kinect2 #203: bulk transfer failed: LIBUSB_ERROR_TIMEOUT Operation timed out

schelian, 2016-01-05:
Hi all, I got iai_kinect2 working fine on one computer but I'm having issues with a second computer.  I've followed the installation steps carefully and below is the error message I get when I run rosrun kinect2_bridge kinect2_bridge.  I get a similar error with Protonect.  

I am running Ubuntu 14.04 LTS (3.13.0-74-generic) with ROS Indigo (1.11.16).  lsusb shows the Kinect is plugged in.

Any help would be appreciated!

Error message:
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/toyota/tim/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: GeForce GTX 570 (GPU)[NVIDIA Corporation]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: GeForce GTX 570 (GPU)[NVIDIA Corporation]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Warning] [OpenCLDepthPacketProcessorImpl] Your ocl-icd has deadlock bugs. Update to 2.2.4+ is recommended.
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 17 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @6:3 with serial 087110440847
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 087110440847 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_TIMEOUT Operation timed out
*** Error in `/home/toyota/tim/devel/lib/kinect2_bridge/kinect2_bridge': double free or corruption (top): 0x0000000001eea740 ***
Aborted (core dumped)

wiedemeyer, 2016-01-05:
Hi,
what error does Protonect report? USB problems could have many reasons. Wrong libusb, unsupported chipset, old kernel, etc.
Please check if the right libusb is linked `ldd /home/toyota/tim/devel/lib/kinect2_bridge/kinect2_bridge | grep libusb`.
If this is correct, try to install a newer kernel, like `linux-generic-lts-utopic` (3.16), `linux-generic-lts-vivid` (3.19) or `linux-generic-lts-wily` (4.2).

schelian, 2016-01-05:
Hi Thiemo, originally the executeables were linked to the wrong library so I fixed that by including libfreenect2/depends/libusb/lib/ in LD_LIBRARY_PATH, but I still get the same error for Protonect shown below...  I will check if I can upgrade the kernel on this machine.

Output from Protonect:
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 17 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @6:3 with serial 087110440847
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_TIMEOUT Operation timed out
Segmentation fault (core dumped)

Output of ldd ./Protonect | grep usb:
	libusb-1.0.so.0 => /home/toyota/tim/src/libfreenect2/depends/libusb/lib/libusb-1.0.so.0 (0x00007f68635e6000)

schelian, 2016-01-06:
Hello, I've upgraded the kernel to 3.19 and libusb looks OK but I still get the same error (LIBUSB_ERROR_TIMEOUT...)

wiedemeyer, 2016-01-06:
Which USB controller are you using? The Intel and Renesas/Nec should work. You can use `lspci` for that.
Just to make sure, the OS is not inside a virtual machine, and you rebooted the system after upgrading the kernel, right. Check which kernel is running with `uname -a`.
I am not an expert on this, maybe someone on libfreenect2 knows more, you might want to open an issue there. You can also try out the 4.2 kernel, just to see if it helps.

closed 2016-01-15 by wiedemeyer

LiShuaixin, 2017-03-05:
I'm trying to connect Kinect2 to my laptop.I am running Ubuntu 14.04 (4.4.0-64-generic) in VMware with ROS Indigo. lsusb shows the Kinect is plugged in. When I run sudo ./Protonect it reports that Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:4 with serial 006381364947
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_TIMEOUT Operation timed out
[Info] [Freenect2DeviceImpl] closing...
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_TIMEOUT Operation timed out
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_TIMEOUT Operation timed out
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed

It seems that something is wrong with my VMware ubuntu system. Can it be fixed? Wish your reply.

inckagan, 2017-12-01:
hi, I am facing exactly same issue with @LiShuaixin Have you got any solution for that ?