code-iai/iai_kinect2 #205: Simultaneously subscribe of rgb and depth topic slowed rgb topic

Mosa-, 2016-01-08:
Hi I have a problem.
I subscribed the rgb and the depth topic in the same ROS node. This action slowed the message rate of the RGB topic...
  
         camImage = getNodeHandle().subscribe(kinectTopic, 100, &LipRec::imageCallback, this);
         camImageDepth = getNodeHandle().subscribe("/kinect2/qhd/image_depth_rect", 100, &LipRec::imageDepthCallback, this);

Is there a possibility to reduce only the rate (fps) of the depth topic and keep the rgb fps?

wiedemeyer, 2016-01-11:
No it is not possible to lower the frame rate of color and depth independently. Normally subscribing to both should not lower any frame rate at all. Processing of Kinect2 data is pretty demanding, I guess that as soon as you subscribe to both topics your CPU load raises a lot and your CPU is not fast enough to process all frames and therefore starts to skip frames. Especially the decoding of the JPEG color images is using a lot CPU resources.

Mosa-, 2016-01-11:
Ok,

thanks for you answer,
Do you know how to get both topics without skipping color frames?

wiedemeyer, 2016-01-11:
There are some general things: Do not run any CPU hungry applications in background. Use the OpenCL / OpenGL for depth computations. Use nodelets with zero memory copy together with the kinect2_bridge nodelet, if used on the same computer. Limit the frame rate.
Another solution is a faster CPU, like a quad core i5/i7.

closed 2016-01-26 by wiedemeyer