code-iai/iai_kinect2 #206: Build problem related to opencl on OSX 10.11, libfreenect2 installed fine with C+11 and OpenCL

fjonath1, 2016-01-08:
I tried building on El Capitan, and it fails at kinect2_registration.
At first, it failed because it cannot find cl.hpp file in #include <OpenCL/cl.hpp>.
I fixed it with cl.hpp from https://www.khronos.org/registry/cl/ version 1.2, which is the same cl.hpp header libfreenect2 use.
Running the build again and I run thru this problem.

```
Felixs-MacBook-Pro:catkin_ws FelixJonathan$ catkin build kinect2_registration
--------------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /Users/FelixJonathan/ros_catkin_ws/install_isolated
Workspace:                   /Users/FelixJonathan/catkin_ws
Source Space:       [exists] /Users/FelixJonathan/catkin_ws/src
Build Space:        [exists] /Users/FelixJonathan/catkin_ws/build
Devel Space:        [exists] /Users/FelixJonathan/catkin_ws/devel
Install Space:     [missing] /Users/FelixJonathan/catkin_ws/install
DESTDIR:                     None
--------------------------------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
--------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
--------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------------------------------- 
Found '50' packages in 0.1 seconds. 
Starting ==> kinect2_registration                                                                                                             
                                                                                                                                              
[kinect2_registration] ==> '/Users/FelixJonathan/catkin_ws/build/kinect2_registration/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j' in '/Users/FelixJonathan/catkin_ws/build/kinect2_registration'
Scanning dependencies of target kinect2_registration
[ 50%] Building CXX object CMakeFiles/kinect2_registration.dir/src/kinect2_registration.cpp.o
[ 50%] Building CXX object CMakeFiles/kinect2_registration.dir/src/depth_registration_opencl.cpp.o
warning: unknown warning option '-Wunused-but-set-variable'; did you mean '-Wunused-const-variable'? [-Wunknown-warning-option]
warning: unknown warning option '-Wunused-but-set-parameter'; did you mean '-Wunused-parameter'? [-Wunknown-warning-option]
warning: unknown warning option '-Wunused-but-set-variable'; did you mean '-Wunused-const-variable'? [-Wunknown-warning-option]
warning: unknown warning option '-Wunused-but-set-parameter'; did you mean '-Wunused-parameter'? [-Wunknown-warning-option]
/Users/FelixJonathan/catkin_ws/src/iai_kinect2/kinect2_registration/src/depth_registration_opencl.cpp:165:3: error: expected expression
  CHECK_CL_ERROR(err, "cl::Platform::get");
  ^
/Users/FelixJonathan/catkin_ws/src/iai_kinect2/kinect2_registration/src/depth_registration_opencl.cpp:38:62: note: expanded from macro 'CHECK_CL_ERROR'
#define CHECK_CL_ERROR(err, str) do { if(err != CL_SUCCESS) {OUT_ERROR(FG_BLUE "[" << CL_FILENAME << "]" FG_CYAN "(" << __LINE__ << ") " FG_YELLOW << str << FG_RED " failed: " << err); return false; } } while(0)
                                                             ^
/Users/FelixJonathan/catkin_ws/src/iai_kinect2/kinect2_registration/include/kinect2_registration/kinect2_console.h:51:24: note: expanded from macro 'OUT_ERROR'
#define OUT_ERROR(msg) OUT_AUX(FG_RED, FG_RED, ROS_ERROR_STREAM, msg)
                       ^
/Users/FelixJonathan/catkin_ws/src/iai_kinect2/kinect2_registration/include/kinect2_registration/kinect2_console.h:46:78: note: expanded from macro 'OUT_AUX'
#define OUT_AUX(FUNC_COLOR, MSG_COLOR, STREAM, MSG) STREAM(FUNC_COLOR "[" << OUT_FUNCTION(__PRETTY_FUNCTION__) << "] " MSG_COLOR << MSG << NO_COLOR)
                                                                             ^
note: (skipping 3 expansions in backtrace; use -fmacro-backtrace-limit=0 to see all)
/Users/FelixJonathan/ros_catkin_ws/install_isolated/include/ros/console.h:394:43: note: expanded from macro 'ROS_LOG_STREAM_COND'
      ROSCONSOLE_PRINT_STREAM_AT_LOCATION(args); \
                                          ^
/Users/FelixJonathan/ros_catkin_ws/install_isolated/include/ros/console.h:357:56: note: expanded from macro 'ROSCONSOLE_PRINT_STREAM_AT_LOCATION'
    ROSCONSOLE_PRINT_STREAM_AT_LOCATION_WITH_FILTER(0, args)
                                                       ^
/Users/FelixJonathan/ros_catkin_ws/install_isolated/include/ros/console.h:352:64: note: expanded from macro 'ROSCONSOLE_PRINT_STREAM_AT_LOCATION_WITH_FILTER'
    __rosconsole_print_stream_at_location_with_filter__ss__ << args; \
                                                               ^
/Users/FelixJonathan/catkin_ws/src/iai_kinect2/kinect2_registration/src/depth_registration_opencl.cpp:169:5: error: expected expression
    OUT_ERROR("no opencl platforms found.");
    ^
/Users/FelixJonathan/catkin_ws/src/iai_kinect2/kinect2_registration/include/kinect2_registration/kinect2_console.h:51:24: note: expanded from macro 'OUT_ERROR'
#define OUT_ERROR(msg) OUT_AUX(FG_RED, FG_RED, ROS_ERROR_STREAM, msg)
                       ^
[... 46 lines cut ...]
make[1]: *** [CMakeFiles/kinect2_registration.dir/all] Error 2
make: *** [all] Error 2
[kinect2_registration] <== '/Users/FelixJonathan/catkin_ws/build/kinect2_registration/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j' failed with return code '2'
                                                                                                                                               
Failed   <== kinect2_registration [ 3.9 seconds ]                                                                                             
[build] There were '1' errors:                                                                                                                

Failed to build package 'kinect2_registration' because the following command:

# Command to reproduce:
cd /Users/FelixJonathan/catkin_ws/build/kinect2_registration && /Users/FelixJonathan/catkin_ws/build/kinect2_registration/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j; cd -

# Path to log:
cat /Users/FelixJonathan/catkin_ws/build/build_logs/kinect2_registration.log

Exited with return code: 2 
                                                                                                                                              
[build] Runtime: 4.1 seconds 

```

Note: I truncated the error log, but basically it repeating the same error related to opencl call in depth_registration_opencl.cpp.

I can run protonect just fine. Any suggestion would be great.
Thanks

wiedemeyer, 2016-01-11:
Hi,
libfreenect2 includes their on OpenCL headers. Maybe that is why it is working for libfreenect2 but not for the kinect2_bridge. You can try to copy the CL folder to `kinect2_registration/include/` and change the include in [depth_registration_opencl.cpp](https://github.com/code-iai/iai_kinect2/blob/master/kinect2_registration/src/depth_registration_opencl.cpp#L22-L29) like they are in [opencl_depth_packet_processor.cpp](https://github.com/OpenKinect/libfreenect2/blob/master/src/opencl_depth_packet_processor.cpp#L39-L50). Maybe that helps.

fjonath1, 2016-01-26:
Hi, just changing include file from CL folder in libfreenect2 does not fix the building problem.
(It still has the same exact problem).

wiedemeyer, 2016-01-27:
What exactly did you change? How does you include statements now look like?

fjonath1, 2016-01-27:
Actually the openCL might not be the problem anymore after replacing it with the libfreenect opencl.
Here is the include statement I have. I don't modify anything in it.
(Note: I had add cl.hpp from libfreenect openCL to the standard location of OpenCL/cl.hpp in OSX. That's the only modification I made.) 

```

#include <fstream>

#include <kinect2_registration/kinect2_console.h>

#ifdef __APPLE__
#include <OpenCL/cl.hpp>
#else
#define CL_USE_DEPRECATED_OPENCL_1_1_APIS
#include <CL/cl.h>
#undef CL_VERSION_1_2
#include <CL/cl.hpp>
#endif

#ifndef REG_OPENCL_FILE
#define REG_OPENCL_FILE ""
#endif

#include "depth_registration_opencl.h"
```

The problem is still about `error: expected expression` on `CHECK_CL_ERROR(err, "cl::Platform::get"); ` when calling this macro
```
#define CHECK_CL_ERROR(err, str) do { if(err != CL_SUCCESS) {OUT_ERROR(FG_BLUE "[" << CL_FILENAME << "]" FG_CYAN "(" << __LINE__ << ") " FG_YELLOW << str << FG_RED " failed: " << err); return false; } } while(0)
```

wiedemeyer, 2016-01-28:
Maybe something in the macro is not supported by your compiler. Try to change it to:
```
#define CHECK_CL_ERROR(err, str) do { if(err != CL_SUCCESS) {OUT_ERROR(str << " failed: " << err); return false; } } while(0)
```

fjonath1, 2016-01-28:
Still no luck, it expanded to the other macro in kinect2_console.
I agree that it might be compiler problem, since I can build this in linux without any problem.
The compiler I use is the standard clang from the osx. `clang --version` result:
```
Apple LLVM version 7.0.2 (clang-700.1.81)
Target: x86_64-apple-darwin15.2.0
Thread model: posix
```
The error from building
```
  CHECK_CL_ERROR(err, "cl::Platform::get");
  ^
/Users/FelixJonathan/catkin_ws/src/iai_kinect2/kinect2_registration/src/depth_registration_opencl.cpp:38:62: note: expanded from macro 'CHECK_CL_ERROR'
#define CHECK_CL_ERROR(err, str) do { if(err != CL_SUCCESS) {OUT_ERROR(FG_BLUE "[" << CL_FILENAME << "]" FG_CYAN "(" << __LINE__ << ") " FG_YELLOW << str << FG_RED " failed: " << err); return false; } } while(0)
                                                             ^
/Users/FelixJonathan/catkin_ws/src/iai_kinect2/kinect2_registration/include/kinect2_registration/kinect2_console.h:51:24: note: expanded from macro 'OUT_ERROR'
#define OUT_ERROR(msg) OUT_AUX(FG_RED, FG_RED, ROS_ERROR_STREAM, msg)
                       ^
/Users/FelixJonathan/catkin_ws/src/iai_kinect2/kinect2_registration/include/kinect2_registration/kinect2_console.h:46:78: note: expanded from macro 'OUT_AUX'
#define OUT_AUX(FUNC_COLOR, MSG_COLOR, STREAM, MSG) STREAM(FUNC_COLOR "[" << OUT_FUNCTION(__PRETTY_FUNCTION__) << "] " MSG_COLOR << MSG << NO_COLOR)
                                                                             ^
/Users/FelixJonathan/catkin_ws/src/iai_kinect2/kinect2_registration/include/kinect2_registration/kinect2_console.h:39:29: note: expanded from macro 'OUT_FUNCTION'
#define OUT_FUNCTION(NAME) ([](const std::string &name)\
                            ^
note: (skipping 2 expansions in backtrace; use -fmacro-backtrace-limit=0 to see all)

```

wiedemeyer, 2016-01-29:
so clang is having a problem with the lambda function. Maybe there are some flags that you have to set when using clang, or maybe the function has to be written differently.

Or if nothing works, replace the [line 39 - 46](https://github.com/code-iai/iai_kinect2/blob/master/kinect2_registration/include/kinect2_registration/kinect2_console.h#L39-L46) with this:
```
const std::string getFunctionName(const std::string &name);
#define OUT_AUX(FUNC_COLOR, MSG_COLOR, STREAM, MSG) STREAM(FUNC_COLOR "[" << getFunctionName(__PRETTY_FUNCTION__) << "] " MSG_COLOR << MSG << NO_COLOR)
```

and paste this to the end of kinect2_registration.cpp:
```
const std::string getFunctionName(const std::string &name)
{
  size_t end = name.rfind('(');
  if(end == std::string::npos)
  {
    end = name.size();
  }
  size_t begin = 1 + name.rfind(' ', end);
  return name.substr(begin, end - begin);
}
```

fjonath1, 2016-02-02:
I got it compiling and submitted pull request #219 .

fjonath1, 2016-02-02:
Note: It runs fine too.
Thanks @wiedemeyer for your help inspecting the problem. The last tips really help me pass the kinect2_registration macro related problem. I just need to fix some minor stuff after that to make it build.

closed 2016-02-02 by fjonath1