code-iai/iai_kinect2 #207: [Freenect2DeviceImpl] stopped in kinect2_bridge

kylop, 2016-01-11:
When I run either
  ```rosrun kinect2_bridge kinect2_bridge _depth_method:=opencl _reg_method:=opencl```
or
  ```  roslaunch kinect2_bridge kinect2_bridge.launch```
I get the following error:
```
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/kilop/ros/ARM_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: cpu
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: true
     base_name_tf: kinect2
   worker_threads: 4
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) HD Graphics Haswell GT2 Mobile (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) HD Graphics Haswell GT2 Mobile (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Warning] [OpenCLDepthPacketProcessorImpl] Your ocl-icd has deadlock bugs. Update to 2.2.4+ is recommended.
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 16 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:13 with serial 500374240242
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 500374240242 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 500374240242
[ INFO] [Kinect2Bridge::initDevice] device firmware: 1.1.3323.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...

[Info] [Freenect2DeviceImpl] stopped
Segmentation fault (core dumped)
```
and 
```
started roslaunch server http://kilop-M3800U:36565/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/kilop/ros/A...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
[... 85 lines cut ...]
[FATAL] [1452519375.941336528]: Service call failed!
[FATAL] [1452519375.943085784]: Service call failed!
[kinect2-1] process has died [pid 2174, exit code -11, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=kinect2 __log:=/home/kilop/.ros/log/f191db9c-b864-11e5-aa93-7c7a9127e43f/kinect2-1.log].
log file: /home/kilop/.ros/log/f191db9c-b864-11e5-aa93-7c7a9127e43f/kinect2-1*.log
[kinect2_bridge-2] process has died [pid 2175, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/kilop/.ros/log/f191db9c-b864-11e5-aa93-7c7a9127e43f/kinect2_bridge-2.log].
log file: /home/kilop/.ros/log/f191db9c-b864-11e5-aa93-7c7a9127e43f/kinect2_bridge-2*.log
[kinect2_bridge-2] restarting process
process[kinect2_bridge-2]: started with pid [2353]
[ INFO] [1452519376.246144665]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1452519376.248711258]: waitForService: Service [/kinect2/load_nodelet] could not connect to host [kilop-M3800U:50246], waiting...
[kinect2_points_xyzrgb_sd-3] process has died [pid 2176, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/sd/camera_info rgb/image_rect_color:=kinect2/sd/image_color_rect depth_registered/image_rect:=kinect2/sd/image_depth_rect depth_registered/points:=kinect2/sd/points __name:=kinect2_points_xyzrgb_sd __log:=/home/kilop/.ros/log/f191db9c-b864-11e5-aa93-7c7a9127e43f/kinect2_points_xyzrgb_sd-3.log].
log file: /home/kilop/.ros/log/f191db9c-b864-11e5-aa93-7c7a9127e43f/kinect2_points_xyzrgb_sd-3*.log
[kinect2_points_xyzrgb_qhd-4] process has died [pid 2177, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/qhd/camera_info rgb/image_rect_color:=kinect2/qhd/image_color_rect depth_registered/image_rect:=kinect2/qhd/image_depth_rect depth_registered/points:=kinect2/qhd/points __name:=kinect2_points_xyzrgb_qhd __log:=/home/kilop/.ros/log/f191db9c-b864-11e5-aa93-7c7a9127e43f/kinect2_points_xyzrgb_qhd-4.log].
log file: /home/kilop/.ros/log/f191db9c-b864-11e5-aa93-7c7a9127e43f/kinect2_points_xyzrgb_qhd-4*.log
[kinect2_points_xyzrgb_hd-5] process has died [pid 2180, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/hd/camera_info rgb/image_rect_color:=kinect2/hd/image_color_rect depth_registered/image_rect:=kinect2/hd/image_depth_rect depth_registered/points:=kinect2/hd/points __name:=kinect2_points_xyzrgb_hd __log:=/home/kilop/.ros/log/f191db9c-b864-11e5-aa93-7c7a9127e43f/kinect2_points_xyzrgb_hd-5.log].
log file: /home/kilop/.ros/log/f191db9c-b864-11e5-aa93-7c7a9127e43f/kinect2_points_xyzrgb_hd-5*.log
[kinect2_points_xyzrgb_sd-3] restarting process
process[kinect2_points_xyzrgb_sd-3]: started with pid [2370]
[kinect2_points_xyzrgb_qhd-4] restarting process
process[kinect2_points_xyzrgb_qhd-4]: started with pid [2371]
[kinect2_points_xyzrgb_hd-5] restarting process
process[kinect2_points_xyzrgb_hd-5]: started with pid [2372]
```
respectively.
Protonect works flawlessly with opengl, opencl and cpu. Any idea how to make the kinect2_bridge working?

wiedemeyer, 2016-01-12:
Hi,
first of all, are you using OpenCV 3.0? That won't work. There is a description in the FAQ.
If you are not using OpenCV 3.0, compile it in DEBUG and run it with gdb, do a stack trace and see where it is crashing.

kylop, 2016-01-12:
Hello,
thanks for you quick reply. I have indeed tried to use OpenCV 3.0 after following this post [ "OpenCV3 support added #199" ](https://github.com/code-iai/iai_kinect2/pull/199/files?diff=split). Now I am forcing (trying to force) ROS indigo to use a standalone OpenCV version modifying the CMakeLists.txt as follow

```
#find_package(OpenCV REQUIRED) 
find_package(OpenCV 2.4.8 REQUIRED)
include_directories(/usr/local/include /usr/local/include/opencv2 /usr/local/include/opencv)
link_directories(/usr/local/lib)
...

target_link_libraries(kinect2_bridge_nodelet
  ${catkin_LIBRARIES}
  #${OpenCV_LIBRARIES} 
  opencv_core
  opencv_highgui
  opencv_imgproc 
...

target_link_libraries(kinect2_bridge
  ${catkin_LIBRARIES}
  #${OpenCV_LIBRARIES} 
  opencv_core
  opencv_highgui
  opencv_imgproc 
...
```
but kinect2_bridge still crashes. Here the DEBUG prompted messages
```
 gdb kinect2_bridge
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from kinect2_bridge...done.
(gdb) run
Starting program: /home/kilop/ros/ARM_ws/devel/lib/kinect2_bridge/kinect2_bridge 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Traceback (most recent call last):
  File "/usr/share/gdb/auto-load/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.19-gdb.py", line 63, in <module>
    from libstdcxx.v6.printers import register_libstdcxx_printers
ImportError: No module named 'libstdcxx'
[New Thread 0x7fffe7241700 (LWP 31139)]
[New Thread 0x7fffe6a40700 (LWP 31140)]
[New Thread 0x7fffe623f700 (LWP 31141)]
[New Thread 0x7fffe5a3e700 (LWP 31146)]
[New Thread 0x7fffe523d700 (LWP 31147)]
[New Thread 0x7fffe4a3c700 (LWP 31148)]
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/kilop/ros/ARM_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: true
     base_name_tf: kinect2
   worker_threads: 4
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) HD Graphics (GPU)[Intel(R) Corporation]
[Info] [OpenCLDepthPacketProcessorImpl]   1: Intel(R) HD Graphics Haswell GT2 Mobile (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) HD Graphics (GPU)[Intel(R) Corporation]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[New Thread 0x7fffcd2af700 (LWP 31170)]
[New Thread 0x7fffccaae700 (LWP 31171)]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 17 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:7 with serial 500374240242
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 500374240242 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 500374240242
[ INFO] [Kinect2Bridge::initDevice] device firmware: 1.1.3323.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...
[Info] [Freenect2DeviceImpl] stopped

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7324b51 in cv::setIdentity(cv::_OutputArray const&, cv::Scalar_<double> const&) () from /usr/local/lib/libopencv_core.so.2.4
(gdb) bt
#0  0x00007ffff7324b51 in cv::setIdentity(cv::_OutputArray const&, cv::Scalar_<double> const&) ()
   from /usr/local/lib/libopencv_core.so.2.4
#1  0x00007ffff7404476 in cv::MatOp_Initializer::assign(cv::MatExpr const&, cv::Mat&, int) const ()
   from /usr/local/lib/libopencv_core.so.2.4
#2  0x000000000044bca5 in cv::Mat::operator= (this=0x7fffffffc898, e=...) at /opt/ros/indigo/include/opencv2/core/mat.inl.hpp:3017
#3  0x0000000000454b12 in Kinect2Bridge::initDevice (this=0x7fffffffc730, sensor=...)
    at /home/kilop/ros/ARM_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:532
#4  0x000000000044fed6 in Kinect2Bridge::initialize (this=0x7fffffffc730)
    at /home/kilop/ros/ARM_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:284
#5  0x000000000044e2c8 in Kinect2Bridge::start (this=0x7fffffffc730)
    at /home/kilop/ros/ARM_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:147
#6  0x000000000044953d in main (argc=1, argv=0x7fffffffd7d8)
    at /home/kilop/ros/ARM_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1418
(gdb) quit
```
I believe you are right and it is some OpenCV incompatibility issue. Any idea how to solve it?
Many thanks.

wiedemeyer, 2016-01-15:
Hi,
if you installed OpenCV 3.0 with apt-get, just use apt-get to uninstall it, or if you installed it from source you might want to look into the OpenCV documentationfor uninstall instructions. You can use ldd to check which libraries are used.

wiedemeyer, 2016-01-27:
I will close this now. Feel free to reopen it if needed.

closed 2016-01-27 by wiedemeyer