code-iai/iai_kinect2 #211: Use calibration with a different RGBD camera

RodBelaFarin, 2016-01-15:
I tried to use your calibration tool with a different RGBD camera, as it is in my point of view actually the best way to calibrate intrinsic and extrinsic parameters with ROS. I remapped all topics to fit, like shown in the rqt_graph attached.
Unfortunately the CV Images do not show up. After a little debugging, I found out, that the callback function of the Subscriber is not called anyway. Any idea why not?

[image]

wiedemeyer, 2016-01-15:
Hi,
the calibration tool is listening to the compressed images, maybe you are not proving them. But you can change the calibration tool to use the raw images. Go to [line 154](https://github.com/code-iai/iai_kinect2/blob/master/kinect2_calibration/src/kinect2_calibration.cpp#L154) and change `compressed` to `raw` and check if it works.

RodBelaFarin, 2016-01-15:
I am already one step further :D
All compressed images are provided and I checked them all through rviz. Therefore I did change line 154 already. That's the reason why it rqt_graph is showing actually the link from the raw images to the calibration tool. Never the less I added a terminal output in the first line of the callback function, but it is never reached.

wiedemeyer, 2016-01-15:
Then it could be the synchronizer. Right now `ExactTime` is used, but if the timestamps of depth and color are not identical, it won't work. Try to change it to `ApproximateTime`. It is in [line 88](https://github.com/code-iai/iai_kinect2/blob/master/kinect2_calibration/src/kinect2_calibration.cpp#L88).

RodBelaFarin, 2016-01-15:
After republishing the Images with a new Timestamp this works for me, thx! Now the IR image is doubled and therefore shrinked to half in width in the CV image viewer, but that seems to be some encoding problems.

closed 2016-01-21 by wiedemeyer