code-iai/iai_kinect2 #212: Close the kinect when kinect is not connected

zfxw0206, 2016-01-15:
Here is the problem:

When we don't connect kinect to PC and run "roslaunch kinect2_bridge kinect2_bridge.launch", we will get:

[Freenect2Impl] enumerating devices...
[Freenect2Impl] 16 usb devices connected
[Freenect2Impl] found 0 devices
Error: no Kinect2 devices found!
Initialization failed!

Then the program will stop here, even though we plug in kinect again, it won't run. We have to "Ctrl +C" and run "roslaunch" again. Now I am thinking can we close it automatically when no kinect connected.

Thank you!

wiedemeyer, 2016-01-18:
Hi,
The program is not waiting until a device is connected, it will just stop and you have to start is again.
That the launch file is not terminating is a problem related to the nodelet. I found no way to tell the nodelet, that the initialization failed. Therefor the nodelet would just continue running.
I added two commits (002f6d03dfe6adec8741ac470f3487457ba35c1a, 087cabe6352df2ad28d2b46af0f2be488f91b1ce) to the devel branch, which should solve the issue.
There are now a nodelet and a node version of kinect2_bridge in the launch file. The nodelet version has a bit less overhead and is used by default, while respawning seems to only work with the node version.
For the nodelet version run `roslaunch kinect2_bridge kinect2_bridge.launch respawn:=false`, it should now terminate if no sensor is found.
For the node version `roslaunch kinect2_bridge kinect2_bridge.launch use_nodelet:=false`. This should restart the kinect2_bridge until a sensor is found.

zfxw0206, 2016-01-20:
Got it, thank you!

closed 2016-01-21 by wiedemeyer