code-iai/iai_kinect2 #213: Failed to load nodelet [/kinect2_bridge] of type [kinect2_bridge/kinect2_bridge_nodelet]

17Skye17, 2016-01-22:
I tried:
roslaunch kinect2_bridge kinect2_bridge.launch

but I came aross an error：
[ERROR] [1453460840.163354866]: Failed to load nodelet [/kinect2_bridge] of type [kinect2_bridge/kinect2_bridge_nodelet]: Could not find library corresponding to plugin kinect2_bridge/kinect2_bridge_nodelet. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[FATAL] [1453460840.163649816]: Service call failed!

I followed the steps on github,but still failed to open kinect camera .Does anyone know why this happens？
Thanks in advance

wiedemeyer, 2016-01-26:
Make sure that your workspace is compiled and that it is sourced.

closed 2016-02-04 by wiedemeyer

Abbyls, 2016-08-28:
@17Skye17 
Hi, I met the same problem! Have you solved it? Could you share the solutions? Thank you very much!

Abbyls, 2016-08-28:
I solved it! sudo apt-get install ros-jade-image-pipeline is ok! But i met another problem, I want to install OPENCL but it can't support Jetson!