code-iai/iai_kinect2 #214: Build error with freenect2 not found

srikanths001, 2016-01-23:
I am newbie. Installed libfreenect2 with all its dependencies. The kinect2 works fine with Protonect. It works with all cpu, gl and cl options.
But when I try to build iai_kinect2, I get cmake error with finding package for freenect2. I have installed libfreenect2 and it works with Protonect2.
I was wondering how to set the path for the freenect2 in the iai_kinect2 CMakeLists file.

Would highly appreciate your help regarding the same.

[error_log.txt](https://github.com/code-iai/iai_kinect2/files/101779/error_log.txt)

CMake Error at iai_kinect2/kinect2_bridge/CMakeLists.txt:22 (find_package):
  By not providing "Findfreenect2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "freenect2", but CMake did not find one.

JimmyDaSilva, 2016-01-23:
You probably forgot a step during libfreenect2's installation.
Go back to ```libfreenect2/build``` and run ```make && sudo make install```

srikanths001, 2016-01-25:
Built libfreenect2 again. It works well now. Thanks.

Btw, on another note, I was calibrating the kinect. I get re-projection errors for color camera to be around 0.2, for ir camera around 0.15.
But for the sync (combination of ir and color extrinsics), the re-projection error I get is around 1.1 something.
What is considered as a good re-projection error. Is there any way to get the best results. Because, whenever I try the calibration, the errors generated seem to be random.

Thanks again. Would appreciate your opinion regarding the same.

JimmyDaSilva, 2016-01-25:
Glad to here that @srikanths001 !

A good reprojection error is below 0.2 so you should be fine. If you want better results you need to follow carefully the calibration instructions (take one image at a time, have a really flat checkerboard, make sure nothing in moving when you capture the image, make sure to move everywhere in the image, add skew to the checkerboard...). But clearly 0.2 should be fine.
The best way to check your calibration results is to run :
```rosrun kinect2_viewer kinect2_viewer qhd both``` and see how it looks.

As the problem is solved here, please close this issue. You can open another one for another problem if necessary.

closed 2016-01-25 by srikanths001

balloch, 2016-12-04:
I know that this is a closed issue, but I wanted to also add that commenting out the `HINTS "$ENV{HOME}/freenect2") ` line got rid of this issue for me.