code-iai/iai_kinect2 #216: Supressing Topics

OAkyildiz, 2016-01-27:
Hello,

Is modifying the launch file enough to supress topics (and possibly nodes) such as hd and qhd?

Best,

JimmyDaSilva, 2016-01-27:
If you open the launch file you will see
```
  <!-- sd point cloud (512 x 424) -->
  <node pkg="nodelet" type="nodelet" name="$(arg base_name)_points_xyzrgb_sd" machine="$(arg machine)"
        args="load depth_image_proc/point_cloud_xyzrgb $(arg nodelet_manager)" respawn="true">
    <remap from="rgb/camera_info"             to="$(arg base_name)/sd/camera_info" />
    <remap from="rgb/image_rect_color"        to="$(arg base_name)/sd/image_color_rect" />
    <remap from="depth_registered/image_rect" to="$(arg base_name)/sd/image_depth_rect" />
    <remap from="depth_registered/points"     to="$(arg base_name)/sd/points" />
    <param name="queue_size" type="int" value="$(arg queue_size)" />
  </node>

  <!-- qhd point cloud (960 x 540) -->
  <node pkg="nodelet" type="nodelet" name="$(arg base_name)_points_xyzrgb_qhd" machine="$(arg machine)"
        args="load depth_image_proc/point_cloud_xyzrgb $(arg nodelet_manager)" respawn="true">
    <remap from="rgb/camera_info"             to="$(arg base_name)/qhd/camera_info" />
    <remap from="rgb/image_rect_color"        to="$(arg base_name)/qhd/image_color_rect" />
    <remap from="depth_registered/image_rect" to="$(arg base_name)/qhd/image_depth_rect" />
    <remap from="depth_registered/points"     to="$(arg base_name)/qhd/points" />
    <param name="queue_size" type="int" value="$(arg queue_size)" />
  </node>

  <!-- hd point cloud (1920 x 1080) -->
  <node pkg="nodelet" type="nodelet" name="$(arg base_name)_points_xyzrgb_hd" machine="$(arg machine)"
        args="load depth_image_proc/point_cloud_xyzrgb $(arg nodelet_manager)" respawn="true">
    <remap from="rgb/camera_info"             to="$(arg base_name)/hd/camera_info" />
    <remap from="rgb/image_rect_color"        to="$(arg base_name)/hd/image_color_rect" />
    <remap from="depth_registered/image_rect" to="$(arg base_name)/hd/image_depth_rect" />
    <remap from="depth_registered/points"     to="$(arg base_name)/hd/points" />
    <param name="queue_size" type="int" value="$(arg queue_size)" />
  </node>
```

So yes you can remove some nodelets.
Though I don't think there is a point in doing this. The kinect2_bridge nodelet are waiting for subscribers before computing data. So basically the hd/qhd topics aren't filled up with data until you start watching them.

wiedemeyer, 2016-01-27:
As @JimmyDaSilva said, you can remove the point cloud nodelets if you don't need them. Anything else is not possible (at least not without changing the code). The kinect2_bridge does not only wait for subscribers, it will also only compute the parts that are needed for the subscribed topics. Why do you need/want to suppress topics?

closed 2016-02-04 by wiedemeyer

OAkyildiz, 2016-02-14:
I wanted to get rid off extra topic names but I guess that is very cosmetic 
.