code-iai/iai_kinect2 #217: Step 6  Connect your sensor and run kinect2_bridge:

HeGaoYuan, 2016-01-28:
when I tried:
roslaunch kinect2_bridge kinect2_bridge.launch
but I got this error:
[kinect2_bridge.launch] is neither a launch file in package [kinect2_bridge] nor is [kinect2_bridge] a launch file name
The traceback for the exception was written to the log file

thank U so much!

wiedemeyer, 2016-01-28:
Did you compiled your workspace and sourced it?

HeGaoYuan, 2016-01-29:
thanks!!! My problem has been solved by your advice!!!

closed 2016-01-29 by wiedemeyer

dandan19793, 2016-04-18:
hallo thiemo,

i followed your tutorial carefully and received the same error message as HeGauYuan.
how do i compile my workspace and source it?

any help will be much appreciated

wiedemeyer, 2016-04-19:
http://wiki.ros.org/ROS/Tutorials/InstallingandConfiguringROSEnvironment

dandan19793, 2016-04-19:
hallo thiemo,

thanks for the quick reply. i already followed the link you had sent me so i didn't think i should do it again. 
anyway, after doing it again I am now stuck at: 
[ INFO] [1461072506.260517892]: [Kinect2Bridge::main] waiting for clients to connect
do you have any idea what might be wrong now?

here is the full output:

student@student:~/catkin_ws$ roslaunch kinect2_bridge kinect2_bridge.launch
... logging to /home/student/.ros/log/3c31a176-0632-11e6-923d-fcaa142568f8/roslaunch-student-17141.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://student:43180/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/student/cat...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_device: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[kinect2-1]: started with pid [17159]
process[kinect2_bridge-2]: started with pid [17160]
process[kinect2_points_xyzrgb_sd-3]: started with pid [17161]
process[kinect2_points_xyzrgb_qhd-4]: started with pid [17162]
process[kinect2_points_xyzrgb_hd-5]: started with pid [17163]
[ INFO] [1461072504.963747776]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1461072504.967466582]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
[ INFO] [1461072504.972209021]: Initializing nodelet with 8 worker threads.
[ INFO] [1461072504.988429160]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ INFO] [1461072505.113160394]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/student/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: default
     depth_device: -1
       reg_method: default
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [CudaDepthPacketProcessorImpl] device 0: Quadro 2000 @ 1251MHz Memory 1020MB
[Info] [CudaDepthPacketProcessorImpl] selected device 0
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 12 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:21 with serial 023571344747
[Info] [Freenect2Impl] found 1 devices
[ INFO] [1461072505.205632980]: [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [1461072505.205674170]: [Kinect2Bridge::initDevice]   0: 023571344747 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [1461072505.302098458]: [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [1461072505.911789470]: [Kinect2Bridge::initDevice] device serial: 023571344747
[ INFO] [1461072505.911825852]: [Kinect2Bridge::initDevice] device firmware: 4.0.3911.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[ WARN] [1461072506.130053428]: [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [1461072506.130098991]: [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [1461072506.130109731]: [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [1461072506.130119235]: [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [1461072506.167248008]: [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [1461072506.167300154]: [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [1461072506.173636184]: [DepthRegistrationOpenCL::init] devices:
[ INFO] [1461072506.173738251]: [DepthRegistrationOpenCL::init]   0: Quadro 2000 (GPU)[NVIDIA Corporation]
[ INFO] [1461072506.173806233]: [DepthRegistrationOpenCL::init] selected device: Quadro 2000 (GPU)[NVIDIA Corporation]
[ INFO] [1461072506.213921387]: [DepthRegistrationOpenCL::init] devices:
[ INFO] [1461072506.214022185]: [DepthRegistrationOpenCL::init]   0: Quadro 2000 (GPU)[NVIDIA Corporation]
[ INFO] [1461072506.214082417]: [DepthRegistrationOpenCL::init] selected device: Quadro 2000 (GPU)[NVIDIA Corporation]
[ INFO] [1461072506.260517892]: [Kinect2Bridge::main] waiting for clients to connect

nomi321, 2016-04-27:
m also facing the same issue. Kindly reply if anybody solved it ... thanks

dandan19793, 2016-04-27:
You need to run the source commands from the link about the ros environment
in every shell you open and every time you open a new shell

m also facing the same issue. Kindly reply if anybody solved it ... thanks

nomi321, 2016-04-27:
sorry bro i didnt get your answer. can you please elaborate this ? thankx

dandan19793, 2016-04-28:
hi,

every time you open a shell you need to run:

 source /opt/ros/<distro>/setup.bash
source devel/setup.bash

as stated in the ros tutorial. again, for every shell, every time you open one.

hope that helped

wiedemeyer, 2016-04-28:
Just form clarification: `[Kinect2Bridge::main] waiting for clients to connect` is no error / issue. kinect2_bridge is ready to go and is just waiting for clients to connect. It is only processing and transmitting frames if there is at least one node to send them to. For more detailed explenaitions please read the tutorials for ROS: http://wiki.ros.org/ROS/Tutorials

nomi321, 2016-04-28:
ok then what to do after this step as it is stucked here until i enter ctrl+c to terminate it?

JimmyDaSilva, 2016-04-28:
You can run RVIZ and look at the images and/or pointclouds. Things are only published once someone it asking for them via ROS.

nomi321, 2016-04-28:
bro What is RVIZ. I m totally new. Please elaborate it. thanks

nomi321, 2016-04-28:
how can i get information about the image/cloud if yes then how ? . I am using the sensor to identify objects. can anybody tell me that what steps should i follow to accomplish it.
i shall be thankful ...

dandan19793, 2016-04-28:
you can run: "rosrun kinect2_viewer kinect2_viewer kinect2 sd cloud" in another shell to watch the images. 
take a look at viewer.cpp at  kinect2_viewer. for example at the function createCloud and the depth and color matrices

nomi321, 2016-04-28:
when i run rosrun rviz rviz i get no output except just cursor keeps on blinking as
noaman@Rime:~/catkin_ws$ "rosrun rviz rviz
> 
when i run 
rosrun image_view image_view image:=/camera/rgb/image_color
then i get this output with a blank window
[ INFO] [1461838542.636481161]: Using transport "raw"

i dont know how to see point cloud. i was following the tutorial 
http://wiki.ros.org/openni_launch/Tutorials/QuickStart

will this tutorial work for kinect v2 or i m going in right direction ? pllzzzzzzzzz helppp

wiedemeyer, 2016-04-28:
Just follow the tutorials for ROS: http://wiki.ros.org/ROS/Tutorials
After you did that a lot of things should be self explanatory.

nomi321, 2016-04-28:
bro there are different tutorials. can u suggest me some related to my problem/requirments? thanks

nomi321, 2016-04-28:
just tell me that how can i get point cloud data ? what are the steps involved ?

wiedemeyer, 2016-04-28:
start with the first, skip what you already know and go to the next until you know all the basics.

nomi321, 2016-04-28:
ok i m doing that. where can i find tutorials regarding kinect v2 Sensor ? I m using it to identify objects? can you help me to give me a direction to go ? thanks in advance

pallavbakshi, 2017-01-19:
Hi @HeGaoYuan  and @dandan19793 , I am stuck at the same problem like you. I have already sourced it as mentioned above but it didn't solve the problem. Any suggestions?

TrinhTUHH, 2017-08-05:
You should add this two line to your bashrc file.
First open bashrc file by sudo gedit ~/.bashrc, then add these two line to the end.
`source /opt/ros/indigo/setup.bash
 source ~/catkin_ws/devel/setup.bash`
Replace 'indigo' by your ros version, and 'catkin_ws' by the location of your catkin workspace. Then open a new Terminal and run `roslaunch kinect2_bridge kinect2_bridge.launch`. Now it should work.
If it still doesn't work, reopen the bashrc file and check if there are duplicate lines of the above 2 lines. Delete the duplicate line and retry.

LittleFishLove, 2018-02-01:
@wiedemeyer  
Hi, I fellow your steps:
 `source devel/setup.bash`
 `echo $ROS_PACKAGE_PATH`
and after I input `roslaunch kinect2_bridge kinect2_bridge.launch`, a  new problem occured:
------------------------------------------------------------------------------------
... logging to /home/gengruopeng/.ros/log/bb636d2e-071d-11e8-9985-f0761c69baf9/roslaunch-gengruopeng-12839.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://gengruopeng:44568/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/gengruopeng...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_device: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: kinetic
 * /rosversion: 1.12.12

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [12850]
ERROR: could not contact master [http://localhost:11311]
The traceback for the exception was written to the log file
[master] killing on exit
-------------------------------------------------------------------
Can you give me some advance? Thank you so much!