code-iai/iai_kinect2 #218: Problem about the kinect calibration

HeGaoYuan, 2016-01-30:
Hello,
When I finished the Step 6: Connect your sensor and run kinect2_bridge, I got the kinect2_bridge hangs and prints "waiting for clients to connect". But you said it is a normal behaviour.
Then I continued the Step 7: Calibrate your sensor using the kinect2_calibration.
I did the first step of the kinect2_calibration: rosrun kinect2_bridge kinect2_bridge _fps_limit:=2. Then I also got the kinect2_bridge hangs and prints "waiting for clients to connect".
You have said that it is because kinect2_bridge will only process data when clients are connected(ROS nodes listening to at least one of the topics). I am sorry that I do not familiar with the ROS, I do not know what the meaning of it is. 
Can you tell me what I should do in details?
This is the first time I play with the ROS, so if my question is too dumb to you, please forgive me.
Thank you so much!!!

HeGaoYuan, 2016-01-31:
Hi, 
I have solved my problem already. It is because I do not familiar with the ROS. The concept of node, topic, master, client, server in ROS are unfamiliar for me. I solved it by open new terminals.

But now, I have another question to ask.
My printer maybe not very good, so it printed the 5x7x0.03 pdf, then I got a paper with 0.033m not the 0.03m. You said the distance between intersections of black and white corners should be 3cm exactly, otherwise the calibration will not work.
I want to ask whether I can just change all the command about the size to 5x7x0.033? Is it OK?
Or is there other details that I should care about?

Thank you so much!!! Looking forward to your reply.

wiedemeyer, 2016-01-31:
Hi,
your guess is right, if the distance is 0.033, then you could just change the parameter and it should work. But make sure that your measurement is as exact as possible. To reduce measurement errors, measure the cumulative distance over for example 3 or 5 field.

HeGaoYuan, 2016-01-31:
OK, thank you so much.
And there is another doubt about calibration.
When I was doing the calibration, the colour lines overlayed on the calibration pattern. But sometimes the colour line could change it's colour. I mean,  the top line points are red this time, then they will change to purple, then again red, then purple......
How about this situation? Is it fine?

wiedemeyer, 2016-02-02:
That shouldn't be a problem.

closed 2016-02-04 by wiedemeyer