code-iai/iai_kinect2 #220: DepthRegistrationOpenCL::init, Kinect2Bridge::start, Service call failed

schelian, 2016-02-03:
I am running Ubuntu 14.04 LTS (3.19.0-49-generic) with ROS Indigo (1.11.16). lsusb shows the Kinect is plugged in.  Protonect seems to work OK except for CL; that error message is below.

When I run roslaunch kinect2_bridge kinect2_bridge.launch I get the message below; trying it with cpu or opengl options doesn't help.  The kinect2_viewer just hangs without displaying anything.  Any help would be appreciated!  You guys, esp. Thiemo, have been so helpful in the past!

I also noticed a thread about libfreenect2 incompatibility issues (https://github.com/code-iai/iai_kinect2/pull/189) but that seems to be resolved.  Should I try rolling back to a previous version of libfreenect2?

**Protonect CL message:**
[Error] [OpenCLDepthPacketProcessor] could not initialize OpenCLDepthPacketProcessor

**kinect2_bridge message:**
... logging to /home/toyota/.ros/log/cd663c90-ca16-11e5-9227-a0d3c11af24c/roslaunch-bnabbe-HP-Z620-Workstation-2841.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://bnabbe-HP-Z620-Workstation:44558/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/toyota/tim/...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_devive: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[kinect2-1]: started with pid [2859]
process[kinect2_bridge-2]: started with pid [2860]
process[kinect2_points_xyzrgb_sd-3]: started with pid [2861]
process[kinect2_points_xyzrgb_qhd-4]: started with pid [2862]
process[kinect2_points_xyzrgb_hd-5]: started with pid [2863]
[ INFO] [1454463550.689232283]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1454463550.695292110]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
[ INFO] [1454463550.843720481]: Initializing nodelet with 4 worker threads.
[ INFO] [1454463550.897651749]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ INFO] [1454463553.198849231]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/toyota/tim/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: default
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Error] [OpenCLDepthPacketProcessorImpl] cl::Platform::get failed: -1001
[Warning] [OpenCLDepthPacketProcessorImpl] Your ocl-icd has deadlock bugs. Update to 2.2.4+ is recommended.
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 087110440847
[Info] [Freenect2Impl] found 1 devices
[ INFO] [1454463553.234005868]: [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [1454463553.234091762]: [Kinect2Bridge::initDevice]   0: 087110440847 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[ INFO] [1454463553.431955528]: [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [1454463554.124707129]: [Kinect2Bridge::initDevice] device serial: 087110440847
[ INFO] [1454463554.124800507]: [Kinect2Bridge::initDevice] device firmware: 1.1.3323.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...
[Error] [OpenCLDepthPacketProcessor] could not initialize OpenCLDepthPacketProcessor
[Info] [Freenect2DeviceImpl] stopped
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Error] [OpenCLDepthPacketProcessorImpl] cl::Program failed: -34
[ WARN] [1454463554.373095169]: [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [1454463554.373166731]: [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [1454463554.373204298]: [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [1454463554.373239613]: [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [1454463554.493337727]: [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [1454463554.493439864]: [DepthRegistration::New] Using OpenCL registration method!
[ERROR] [1454463554.533975711]: [DepthRegistrationOpenCL::init] [depth_registration_opencl.cpp](165) cl::Platform::get failed: -1001
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[ERROR] [1454463554.536379094]: [Kinect2Bridge::start] Initialization failed!
terminate called after throwing an instance of 'std::system_error'
  what():  Invalid argument
[FATAL] [1454463554.987070568]: Service call failed!
[FATAL] [1454463554.987102897]: Service call failed!
[kinect2-1] process has died [pid 2859, exit code -6, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=kinect2 __log:=/home/toyota/.ros/log/cd663c90-ca16-11e5-9227-a0d3c11af24c/kinect2-1.log].
log file: /home/toyota/.ros/log/cd663c90-ca16-11e5-9227-a0d3c11af24c/kinect2-1*.log
[kinect2_bridge-2] process has died [pid 2860, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/toyota/.ros/log/cd663c90-ca16-11e5-9227-a0d3c11af24c/kinect2_bridge-2.log].
log file: /home/toyota/.ros/log/cd663c90-ca16-11e5-9227-a0d3c11af24c/kinect2_bridge-2*.log
[kinect2_points_xyzrgb_hd-5] process has died [pid 2863, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/hd/camera_info rgb/image_rect_color:=kinect2/hd/image_color_rect depth_registered/image_rect:=kinect2/hd/image_depth_rect depth_registered/points:=kinect2/hd/points __name:=kinect2_points_xyzrgb_hd __log:=/home/toyota/.ros/log/cd663c90-ca16-11e5-9227-a0d3c11af24c/kinect2_points_xyzrgb_hd-5.log].
log file: /home/toyota/.ros/log/cd663c90-ca16-11e5-9227-a0d3c11af24c/kinect2_points_xyzrgb_hd-5*.log
[kinect2_bridge-2] restarting process
process[kinect2_bridge-2]: started with pid [3065]
[kinect2_points_xyzrgb_hd-5] restarting process
process[kinect2_points_xyzrgb_hd-5]: started with pid [3066]
[ INFO] [1454463555.299099247]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1454463555.305702366]: waitForService: Service [/kinect2/load_nodelet] could not connect to host [bnabbe-HP-Z620-Workstation:34423], waiting...

wiedemeyer, 2016-02-03:
There is obviously something wrong with your OpenCL setup. Did you installed Beignet and nvidia drivers? Also check that `/etc/OpenCL/vendors/*.icd` is giving the correct library. You could also check [google](https://www.google.com/search?q=cl::Platform::get+%22-1001%22).

CPU or OpenGL should work fine with the kinect2_bridge:
`roslaunch kinect2_bridge kinect2_bridge.launch depth_method:=opengl reg_method:=cpu` or
`rosrun kinect2_bridge kinect2_bridge _depth_method:=opengl _reg_method:=cpu`

And the kinect2_viewer is not hanging, it's just waiting for images. If it is not receiving images, then there is nothing it could display.

schelian, 2016-02-09:
Please close this issue.  I reinstalled everything from NVIDIA and things work now.  Not sure why that worked since this package shouldn't need NVIDIA but I'm happy now. :-P

closed 2016-02-09 by schelian