code-iai/iai_kinect2 #221: Publishing framerate slows down after about 1000 frames(30s)

liruihao, 2016-02-03:
I use it successful at the beginning. And the image frame rate is about 30 Hz using OpenCL. But after I receive 1000 frames, the publish speed slow down to 2 Hz. Could you tell me what should I do? 

Thank you very much for your help.

wiedemeyer, 2016-02-04:
What is the output of kinect2_bridge? You should look at your CPU / RAM usage, is there something changing after 30 seconds? Also check the GPU and CPU frequencies and temperature, maybe there is some heat related throttling. Does this also affect Protonect?

liruihao, 2016-02-04:
Thank you very much! I think it must be the problem of the temperature. I let the laptop rest for one night. And I try it this morning. All look fine. The framerate can last for 30Hz for a long time.

Thank you for your help.

liruihao, 2016-02-04:
Sorry for the comments above. when I run it for the second time, the same thing happened. I just use the "image_viewer" to display it.  The output of the kinect2_bridge is shown below:

[ INFO] [1454586916.745322548]: [Kinect2Bridge::main] depth processing: ~6.98436ms (~143.177Hz) publishing rate: ~29.9823Hz
[ INFO] [1454586916.745399388]: [Kinect2Bridge::main] color processing: ~14.3286ms (~69.7904Hz) publishing rate: ~29.9823Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 0ms -> ~infHz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 0ms -> ~infHz
[ INFO] [1454586919.747006174]: [Kinect2Bridge::main] depth processing: ~6.65308ms (~150.306Hz) publishing rate: ~29.9833Hz
[ INFO] [1454586919.747078574]: [Kinect2Bridge::main] color processing: ~12.9609ms (~77.155Hz) publishing rate: ~29.9833Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 0ms -> ~infHz
[Info] [DepthPacketStreamParser] 9 packets were lost
[ INFO] [1454586922.749774182]: [Kinect2Bridge::main] depth processing: ~32.5078ms (~30.7618Hz) publishing rate: ~25.9814Hz
[ INFO] [1454586922.750866598]: [Kinect2Bridge::main] color processing: ~116.42ms (~8.58961Hz) publishing rate: ~4.99642Hz
[Info] [DepthPacketStreamParser] 3 packets were lost
[Info] [DepthPacketStreamParser] 3 packets were lost
[Info] [DepthPacketStreamParser] 3 packets were lost
[Info] [OpenCLDepthPacketProcessor] avg. time: 0ms -> ~infHz
[ INFO] [1454586925.763053853]: [Kinect2Bridge::main] depth processing: ~36.2748ms (~27.5673Hz) publishing rate: ~26.5471Hz
[ INFO] [1454586925.763853165]: [Kinect2Bridge::main] color processing: ~197.943ms (~5.05196Hz) publishing rate: ~2.98655Hz
[Info] [DepthPacketStreamParser] 4 packets were lost
[Info] [DepthPacketStreamParser] 2 packets were lost
[Info] [DepthPacketStreamParser] 5 packets were lost
[ INFO] [1454586928.769535313]: [Kinect2Bridge::main] depth processing: ~36.2473ms (~27.5883Hz) publishing rate: ~26.2765Hz
[ INFO] [1454586928.770211831]: [Kinect2Bridge::main] color processing: ~160.147ms (~6.24427Hz) publishing rate: ~2.99353Hz
[Info] [DepthPacketStreamParser] 4 packets were lost
[Info] [OpenCLDepthPacketProcessor] avg. time: 0ms -> ~infHz
[Info] [DepthPacketStreamParser] 3 packets were lost
[Info] [DepthPacketStreamParser] 3 packets were lost
[ INFO] [1454586931.773508492]: [Kinect2Bridge::main] depth processing: ~40.0735ms (~24.9542Hz) publishing rate: ~26.6308Hz
[ INFO] [1454586931.774190253]: [Kinect2Bridge::main] color processing: ~200.292ms (~4.99272Hz) publishing rate: ~2.99596Hz
[Info] [DepthPacketStreamParser] 3 packets were lost
[Info] [OpenCLDepthPacketProcessor] avg. time: 0ms -> ~infHz
[Info] [DepthPacketStreamParser] 4 packets were lost
^C[ INFO] [1454586934.783427791]: [Kinect2Bridge::main] depth processing: ~42.9849ms (~23.264Hz) publishing rate: ~25.2504Hz
[Info] [DepthPacketStreamParser] 12 packets were lost
[ INFO] [1454586983.223891475]: [Kinect2Bridge::main] color processing: ~196.373ms (~5.09234Hz) publishing rate: ~3.32243Hz

We can see it turn to 2Hz suddenly. I do not know how to solve  the problem. Could you tell me how to do with it? Thank you .

liruihao, 2016-02-04:
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Warning] [OpenCLDepthPacketProcessorImpl] Your ocl-icd has deadlock bugs. Update to 2.2.4+ is recommended.
[Info] [Freenect2Impl] enumerating devices...

It told me that my ocl-icd has deadlock bugs, is it the reason for the problem?

liruihao, 2016-02-04:
I finally know what wrong with it. It is the problem of the laptop power. When I keep it in charge, all is fine. But when I use the battery without charging it will slow down after 30s. Maybe  GPU needs stable and sufficient voltage and current to work normally. And my battery is not good enough.

Thank you very much for your help.

wiedemeyer, 2016-02-04:
I guess when you laptop is on battery the cpu/gpu are set from a max performance mode to a min power mode.

liruihao, 2016-02-04:
Yeah, you must be right. I am wondering how can I change the settings in ubuntu. 

Thank you very much.

liruihao, 2016-02-04:
I have set the CPU and GPU to maxmum performance. When i run without charging, the speed increase from 2Hz to 10Hz. And when I plug the power supply in, it immediately increase to 30Hz. I think you can close this issue.

Thanks again for your help.

closed 2016-02-04 by wiedemeyer