code-iai/iai_kinect2 #222: OpenCL registration is not available!

rshanor, 2016-02-05:
Hi,

I have been using iai_kinect2 on two different computers for a while now. Yesterday, I think due to some linux update, the bridge stopped working using CL. It still works with CPU. I am getting the same error on both machines. However, protonect still runs fine with the cl flag. And advice? 

Thanks,
Rick

[ERROR] [1454613959.698271901]: [Kinect2Bridge::initRegistration] OpenCL registration is not available!
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[ERROR] [1454613959.699242088]: [Kinect2Bridge::start] Initialization failed!
terminate called after throwing an instance of 'std::system_error'
  what():  Invalid argument
[FATAL] [1454613959.849973610]: Service call failed!
[FATAL] [1454613959.850120716]: Service call failed!
[kinect2-2] process has died [pid 4454, exit code -6, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=kinect2 __log:=/home/rshanor/.ros/log/1ddf8910-cb75-11e5-bd88-104a7d7fca01/kinect2-2.log].
log file: /home/rshanor/.ros/log/1ddf8910-cb75-11e5-bd88-104a7d7fca01/kinect2-2*.log
[kinect2_bridge-3] process has died [pid 4461, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/rshanor/.ros/log/1ddf8910-cb75-11e5-bd88-104a7d7fca01/kinect2_bridge-3.log].
log file: /home/rshanor/.ros/log/1ddf8910-cb75-11e5-bd88-104a7d7fca01/kinect2_bridge-3*.log
[kinect2_points_xyzrgb_sd-4] process has died [pid 4462, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/sd/camera_info rgb/image_rect_color:=kinect2/sd/image_color_rect depth_registered/image_rect:=kinect2/sd/image_depth_rect depth_registered/points:=kinect2/sd/points __name:=kinect2_points_xyzrgb_sd __log:=/home/rshanor/.ros/log/1ddf8910-cb75-11e5-bd88-104a7d7fca01/kinect2_points_xyzrgb_sd-4.log].
log file: /home/rshanor/.ros/log/1ddf8910-cb75-11e5-bd88-104a7d7fca01/kinect2_points_xyzrgb_sd-4*.log

wiedemeyer, 2016-02-15:
Hi,
try to reinstall your graphics driver and check again. Make sure that the OpenCL parts of the driver are installed too.

QingkaiLu, 2016-02-21:
@rshanor Did that fix the problem? I am facing the same issue.

rshanor, 2016-02-21:
I have still not been able to fix this issue. I have tried a complete system reinstall and still run into this issue. Pretty confused to what the problem could be... After reinstall, I installed nvidia-352 with nvidia-libopencl and nvidia-opencl-icd. Any other ideas?

QingkaiLu, 2016-02-21:
This is what we did for the re-installation of opencl:
sudo apt-get purge nvidia*
sudo apt-get install nvidia-libopencl1-352
sudo apt-get install nvidia-opencl-icd-352
sudo apt-get install nvidia-352

After doing this, the libfreenect2 works fine with the opencl (./Protonect cl). However ,the iai_kinect2 still doesn't work with opencl.  The cpu version of iai_kinect2 works fine. But if I change the "reg_method" to "opencl", it has the error: "[Kinect2Bridge::initRegistration] OpenCL registration is not available!".

@wiedemeyer Any suggestions about how to solve this issue? Do I need to re-install the nvidia driver again? Thank you a lot for any help!

wiedemeyer, 2016-02-22:
`[Kinect2Bridge::initRegistration] OpenCL registration is not available!` means that it was not compiled with OpenCL support. You have to recompile iai_kinect2 after you installed OpenCL. Remove `devel` and `build` folders and run `catkin_make` again.
I also added some OpenCL compatibility changes, that where added to libfreenect2 also to the kinect2_registration. Please checkout the devel branch and check if it fixes your problem.

QingkaiLu, 2016-02-22:
Remove devel and build folders and run catkin_make again solved my problem, now iai_kinect2 works well with opencl. Previously, I did catkin_make after the reinstall of the driver and opencl, but I didn't delete devel and build folders.

Thank you so much for the help! I appreciate all the effort you guys put in the iai_kinect2, which makes it a lot easier to use kinect in ROS.

rshanor, 2016-02-25:
The new devel branch helped, thanks!

closed 2016-02-25 by wiedemeyer