code-iai/iai_kinect2 #223: kinect2_bridge, 0Hz Publishing rate

haxan7, 2016-02-10:
`Protonect gl`, `Protonect cl` and `Protonect cpu` all work perfectly but `roslaunch kinect2_bridge kinect2_bridge.launch` fails to publish anything.

```
[ INFO] [1455104650.956339683]: [Kinect2Bridge::main] depth processing: ~-nanms (~-nanHz) publishing rate: ~0Hz
[ INFO] [1455104650.956483375]: [Kinect2Bridge::main] color processing: ~-nanms (~-nanHz) publishing rate: ~0Hz
[ INFO] [1455104653.956905909]: [Kinect2Bridge::main] depth processing: ~-nanms (~-nanHz) publishing rate: ~0Hz
[ INFO] [1455104653.957052427]: [Kinect2Bridge::main] color processing: ~-nanms (~-nanHz) publishing rate: ~0Hz
```

```
$ ldd Protonect | grep libfree
	libfreenect2.so.0.1 => /home/hassan/ext_src/libfreenect2/build/lib/libfreenect2.so.0.1 (0x00007fddb0dba000)
```

```
$ ldd ~/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge | grep libfree
	libfreenect2.so.0.1 => /usr/local/lib/libfreenect2.so.0.1 (0x00007f9ba7d0c000)
	libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f9b9593d000)
```

xArkanEx, 2016-02-11:
I do have the same problem here.
Protonect works fine on `gl, cl and cpu` modes but impossible to get any data from kinect2_bridge as it fails to publish any data.

Same Info message as above : 
```
[ INFO] [1455184486.640053452]: [Kinect2Bridge::main] depth processing: ~-nanms (~-nanHz) publishing rate: ~0Hz
[ INFO] [1455184486.640273220]: [Kinect2Bridge::main] color processing: ~-nanms (~-nanHz) publishing rate: ~0Hz
```
Thank you for your help.

haxan7, 2016-02-11:
@xArkanEx I have got it working by rolling back libfreenect2 to [9f39b744a8fe89d4d86e38bb2c494303ab41ae86](https://github.com/OpenKinect/libfreenect2/commit/9f39b744a8fe89d4d86e38bb2c494303ab41ae86) and re-installing.

xArkanEx, 2016-02-11:
Thank you @haxan7  I will try it.
By investigating a bit I found that the  `waitForNewFrame` method of the  `libfreenect2::SyncMultiFrameListener` in the `receiveFrames` function of `kinect2_bridge` always returned false.
That's when i though the problem had more chance to come from `libfreenect2` than from `iai_kinect2`.

wiedemeyer, 2016-02-11:
I just tried out the current master from libfreenect2 and I can confirm the problem. I will look into it later, for now please use an older commit from libfreenect2.

xlz, 2016-02-11:
I believe this issue is fixed in latest libfreenect2 master.

closed 2016-02-12 by wiedemeyer