code-iai/iai_kinect2 #226: Service Call failed, then enter run loop

hyderjake, 2016-02-14:
First loop detects device but I get service call failed then process died, following loops cannot find devices. cannot find similar errors in issues pages
Protonect gl, cl, cpu work, gdb has no backtrace error to post

Ubuntu 14.04, kernel 3.19.0-49, nvidia-361

wiedemeyer, 2016-02-15:
Hi,
I need some more information. Which service call failed? What is the terminal output? Did you used `roslaunch` or `rosrun`? Does `rosrun kinect2_bridge kinect2_bridge` work?

hyderjake, 2016-02-15:
jacob@Ghost:~/catkin_ws$ roslaunch kinect2_bridge kinect2_bridge.launch
... logging to /home/jacob/.ros/log/901646ee-d409-11e5-8512-5ce0c5d7dfde/roslaunch-Ghost-2717.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Ghost:39919/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/jacob/catki...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_devive: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [2729]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 901646ee-d409-11e5-8512-5ce0c5d7dfde
process[rosout-1]: started with pid [2742]
started core service [/rosout]
process[kinect2-2]: started with pid [2748]
process[kinect2_bridge-3]: started with pid [2756]
process[kinect2_points_xyzrgb_sd-4]: started with pid [2761]
process[kinect2_points_xyzrgb_qhd-5]: started with pid [2762]
process[kinect2_points_xyzrgb_hd-6]: started with pid [2764]
[ INFO] [1455557324.642302175]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1455557324.647576242]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
[ INFO] [1455557324.652979968]: Initializing nodelet with 8 worker threads.
[ INFO] [1455557324.669232604]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ERROR] [1455557324.853380250]: Failed to load nodelet [/kinect2_bridge] of type [kinect2_bridge/kinect2_bridge_nodelet] even after refreshing the cache: Could not find library corresponding to plugin kinect2_bridge/kinect2_bridge_nodelet. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.                                      
[ERROR] [1455557324.853413950]: The error before refreshing the cache was: Could not find library corresponding to plugin kinect2_bridge/kinect2_bridge_nodelet. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.                                                                                                                     
[FATAL] [1455557324.853538564]: Service call failed!
[kinect2_bridge-3] process has died [pid 2756, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/jacob/.ros/log/901646ee-d409-11e5-8512-5ce0c5d7dfde/kinect2_bridge-3.log].                                                                                                                                  
log file: /home/jacob/.ros/log/901646ee-d409-11e5-8512-5ce0c5d7dfde/kinect2_bridge-3*.log                                          
[kinect2_bridge-3] restarting process
process[kinect2_bridge-3]: started with pid [2907]
[ INFO] [1455557325.172608947]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ERROR] [1455557325.261043218]: Failed to load nodelet [/kinect2_bridge] of type [kinect2_bridge/kinect2_bridge_nodelet] even after refreshing the cache: Could not find library corresponding to plugin kinect2_bridge/kinect2_bridge_nodelet. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.                                      
[ERROR] [1455557325.261077033]: The error before refreshing the cache was: Could not find library corresponding to plugin kinect2_bridge/kinect2_bridge_nodelet. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.                                                                                                                     
[FATAL] [1455557325.261202398]: Service call failed!
[kinect2_bridge-3] process has died [pid 2907, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/jacob/.ros/log/901646ee-d409-11e5-8512-5ce0c5d7dfde/kinect2_bridge-3.log].                                                                                                                                  
log file: /home/jacob/.ros/log/901646ee-d409-11e5-8512-5ce0c5d7dfde/kinect2_bridge-3*.log                                          
[kinect2_bridge-3] restarting process
[... cut ...]
[kinect2_bridge-3] process has died [pid 2928, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/jacob/.ros/log/901646ee-d409-11e5-8512-5ce0c5d7dfde/kinect2_bridge-3.log].                                                                                                                                  
log file: /home/jacob/.ros/log/901646ee-d409-11e5-8512-5ce0c5d7dfde/kinect2_bridge-3*.log

Same output if run with the dedicated graphics
^^cannot find that log file in that location either

hyderjake, 2016-02-15:
I think this is a little different than what I described before

hyderjake, 2016-02-16:
reinstalling ROS(again) it seems i need to do that on every reboot, I cannot understand why yet but here is the error that I posted about

jacob@Ghost:~/catkin_ws$ roslaunch kinect2_bridge kinect2_bridge.launch 
... logging to /home/jacob/.ros/log/038824cc-d454-11e5-adc0-5ce0c5d7dfde/roslaunch-Ghost-5491.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Ghost:60173/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/jacob/catki...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_devive: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [5503]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 038824cc-d454-11e5-adc0-5ce0c5d7dfde
process[rosout-1]: started with pid [5516]
started core service [/rosout]
process[kinect2-2]: started with pid [5524]
process[kinect2_bridge-3]: started with pid [5534]
process[kinect2_points_xyzrgb_sd-4]: started with pid [5535]
process[kinect2_points_xyzrgb_qhd-5]: started with pid [5536]
[ INFO] [1455589301.089899749]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
process[kinect2_points_xyzrgb_hd-6]: started with pid [5542]
[ INFO] [1455589301.091675793]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
[ INFO] [1455589301.101743213]: Initializing nodelet with 8 worker threads.
[ INFO] [1455589301.113534950]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ INFO] [1455589301.275195181]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/jacob/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
[... 60 lines cut ...]
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) HD Graphics BroadWell Mobile GT2 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) HD Graphics BroadWell Mobile GT2 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found 0 devices
[ERROR] [1455589303.113510726]: [Kinect2Bridge::initDevice] no Kinect2 devices found!
[ERROR] [1455589303.119193630]: [Kinect2Bridge::start] Initialization failed!
[ERROR] [1455589303.119263695]: [Kinect2Bridge::stop] kinect2_bridge is not running!
[FATAL] [1455589303.204327818]: Service call failed!
[kinect2_bridge-3] process has died [pid 5724, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/jacob/.ros/log/038824cc-d454-11e5-adc0-5ce0c5d7dfde/kinect2_bridge-3.log].                                                                                                                                  
log file: /home/jacob/.ros/log/038824cc-d454-11e5-adc0-5ce0c5d7dfde/kinect2_bridge-3*.log

wiedemeyer, 2016-02-16:
Your first problem looks like you didn't sourced the workspace correctly, or didn't compiled it.
For the latest problem, normaly the device is started to read out the parameters, and then the device is stopped again, but strangely libfreenect2 reports that is couldn't stop the device correctly.
Try to unplug the Kinect2, plug it back in, maybe even on a different USB 3 port and run `rosrun kinect2_bridge kinect2_bridge` (not the roslaunch command).
By the way, which version of libfreenect2 are you using? Tag v0.1.1 or master?

hyderjake, 2016-02-16:
I'm using the protonect master

hyderjake, 2016-02-16:
Using rosrun, i have could not stop device, initialization failed

wiedemeyer, 2016-02-17:
So, the problem is, that libfreenect2 `device::stop()` could not stop the device correctly. When you look at [libfreenect2.cpp](https://github.com/OpenKinect/libfreenect2/blob/master/src/libfreenect2.cpp#L771-L812) you will see, that it should print `stopped` in the terminal if everything worked, but in your log the last message is `canceling depth transfers...`, therefore one of the commands below did not work.
Does Protonect print out `stopped` when shutting it down? If not, you may want to open an issue on libfreenect2s github.
Until that is fixed, you could checkout the v0.1.1 tag of libfreenect2 and see if that one works for you.

hyderjake, 2016-02-18:
I have reverted the Protonect and ran rosrun kinect2_bridge, it is working, however kinect2_viewer doesn't work, is it still in the build list or did the command change?

hyderjake, 2016-02-18:
Sorry not sure what I did, kinect viewer is running now

closed 2016-02-19 by wiedemeyer