code-iai/iai_kinect2 #227: About the depth image

HeGaoYuan, 2016-02-17:
I did the rosrun kinect2_viewer kinect2_viewer hd cloud, 
then I knocked the space key to save the image.
As you know, I got the 0000_depth.png and 0000_color.png. The resolution of these images are 1920*1080, the same of the RGB camera of the kinect2. 
So, I want to ask if the value of the pixel of the depth.png is the depth value of the corresponding pixel in the color.png. But I got the value is very big, the biggest is 11233. I want know the unit of the value. What does the 11233 mean?
Thank U very much!!!

wiedemeyer, 2016-02-18:
It should be depth in mm, so it means 11.233 meters.

HeGaoYuan, 2016-02-18:
Thank you!!!
So this depth is just the measured depth? 
I think the measured depth is measured by the kinect2's depth camera, the computed depth is computed by your algorithm. Is it right?

wiedemeyer, 2016-02-18:
Yes, it is the measured depth. Computed depth is only for calibration and it is only computed there.

closed 2016-02-19 by wiedemeyer