code-iai/iai_kinect2 #228: IAI_Kinect2 optimization

shamoorti, 2016-02-18:
Dear Thiemo,
I hope you're doing great. Last semester I developed a vision system with the help of your great bridge between Kinect and ROS with Matlab. Final results were acceptable but some issues were sighted during the implementation. Although your code is optimized with OpenCL, still CPU usage is exorbitant.   Running the bridge at 5HZ on Intel i7 Haswell with 4 cores, leads to 58 percent CPU usage. I know hardware is an issue but do you have any solution about further optimization that I can do a research on? I read the threads about Jetson and Cuda and they were interesting. Is there any other solutions like optimizing buffers or more on OpenCL? And also can I ask why you chose OpenCL instead of OpenMP? Did you have any special criteria?

wiedemeyer, 2016-02-18:
Most demanding is compression. Try to use the uncompressed topics if possible. Also you can update libfreenect2. They recently implemented a VA-API based JPEG decoding which uses the Intel GPU to decode the JPEGs, on my system (i7 Haswell) the CPU load is around 2 to 3 percent, when libfreenect2 is processing the frames with VA-API and CUDA/OpenCL.

shamoorti, 2016-02-19:
Thanks a lot Thiemo. I will look into this.

closed 2016-02-19 by wiedemeyer