code-iai/iai_kinect2 #229: About the 3D reconstruction

HeGaoYuan, 2016-02-18:
Dear wiedemeyer,
    Firstly, thank you so much for your huge contribution to this project and your reply for my questions before.
    I have been trying to use the kinect2 to do 3D reconstruction. I want to obtain the 3D coordinates of interesting pixels in the image view. I have been working on this for many days but  I am stuck right now. I had some questions and would really appreciate your help. 

   I studied the book <Learning opencv>, it teaches how to use double cameras to calculate 3d coordinates. The method is firstly to calculate the Binocular Disparity, then use the Q matrix([image]) to get the results. The results X/W, Y/W, Z/W are the 3d coordinates of the point in the left camera's coordinate system. And there is another way to get the 3d coordinates form the Longuet-Higgins' paper.(https://en.wikipedia.org/wiki/Essential_matrix). But it seems that you don't use these two methods. I studied your code to calculate the depth. You get two depths, one is computed depth, another is measured depth. I think the method you use to get the computed depth is https://github.com/code-iai/iai_kinect2/blob/master/kinect2_calibration/src/kinect2_calibration.cpp#L951-L957 and the three functions computePointDists, computeDistance, getPlane. But I don' t understand how the code works. How can you get the depth through these codes? Can you give me a link about the theory of the code?
And another question is, I think the computed and measured depth are all related to the ir camera(i.e. the depth camera). Is it also the depth related to the RGB camera? In my opinion, it is.

     Then, if we get the depth(i.e. the Z value), we can easily get the 3d coordinates. As you said previously, "you could look into the code of the kinct2_viewer::createCloud and kinct2_viewer::createLookup. That code calculates the real world coordinates from the depth image." .
So, I want to ask: Are itP->z, itP->x, itP->y (3d coordinates) related to the RGB camera's coordinate system?
     And I studied your code in the kinect2_registration/src.Firstly, I want to know what is the relationship between these files? For example, what is the relationship between the depth_registration.cl and the depth_registrationopencl.cpp? 
    And the code https://github.com/code-iai/iai_kinect2/blob/master/kinect2_registraion/src/depth_registration.cl#L25-L106. I saw you get the 3d point through `const float4 point = (float4)((xR - cxR) * fxRInv * z, (yR - cyR) * fyRInv  * z, z, 1.0f);` Is the 3d point related to the R camera coordinate system? Is the R camera the RGB camera or the IR camera?
And I want to ask why you then compute projected image coordinates? Is it just used for the next step?(to check if depth is valid and projection is inside the image)
Actually, I don't totally understand the use of the files in the folder 'kinect2_registration/src'. What are they used for?
     
      These are a lot of questions to trouble you. Thank you very much!!! And looking forward to your reply.

wiedemeyer, 2016-02-18:
Hi,
that was a long one. I hope I caught all the questions.
First the short answer to the general question:
All images in the same topic group (`hd`, `qhd`, `sd`) are in the same frame. If you want the 3D point for a certain pixel, just take the depth value of that point and the intrinsic parameters and calculate the 3D point, as done in `kinct2_viewer::createCloud`. `hd` and `qhd` are in the frame of the color sensor and `sd` is in the frame of the IR sensor. There are also complete point clouds available when using the launch file, but they are demanding more CPU resources.
The computed and measured depth is just for calibration. The computed one is computed using the plane the chessboard defines and the measured one is from the device.
the `.cl` is the OpenCL code that is executed on the GPU, the `.cpp` file takes care of loading the `.cl` file to the GPU and to execute it.
The depth_registration registers a depth image from one camera frame to another one. It translates and rotates the depth image from IR frame to fit the color frame and then projects the 3D points back to the image plane. The values with `R` belong to the other frame, so in this example the color frame. Besides simply rotating and translating the depth, it also transforms the depth image, so that it fits the intrinsic parameters of the color camera (as if it was seen from the color camera) and interpolates intermediate values.

HeGaoYuan, 2016-02-21:
Hi, thank you for your awesome reply. I am writing the reply, please do not close it.

wiedemeyer, 2016-02-22:
You can do the same for the RGB view. For example `/kinect2/hd/image_color_rect` as `color` and `/kinect2/hd/image_depth_rect` as `depth`, then `color(200, 300)` corresponds to `depth(200, 300)`.
No, that is correct, because first `remapDepth` is called, this creates a bigger interpolated depth image with has the intrinsic parameters of the color image, therefore it is correct to use the color intrinsic parameters there.
Yes, it stores the undistorted images.

wiedemeyer, 2016-03-07:
If you have no further questions you can close this issue.

HeGaoYuan, 2016-03-08:
OK, I am so sorry for too many days not reply. 
Thank you very much!!!
You helped me a lot!!!

closed 2016-03-08 by HeGaoYuan